robotology/gazebo-yarp-plugins #382: Network Client for IMU

mhubii, 2018-09-09:
Is there a way to access the interface of an IMU in gazebo remotely but by reading the ports? Something like this:

```
// Create interface to device.
yarp::os::Property opt;

opt.put("device", "???"); // Which device could one use here?
opt.put("local", "/client/interial");
opt.put("remote", "/server/interial");

yarp::dev::PolyDriver* pd = new yarp::dev::PolyDriver(opt);

yarp::dev::IGenericSensor* gs;
pd->view(gs);
```

Thanks in advance, Martin

mhubii, 2018-09-09:
Guess not. But for someone with a simular problem, one can create a port that stores the last received value like so:

```
#include <yarp/os/all.h>

// A port that stores the value last received.
class IMUPort : public yarp::os::Port
{
public:

	IMUPort() {

		setCallbackLock(&mutex_);
		setReader(last_);
	}

	yarp::sig::Vector LastRead() {

		lockCallback();
		yarp::sig::Vector tmp = last_;
		unlockCallback();
		
		return tmp;		
	}

private:

	yarp::sig::Vector last_;
	yarp::os::Mutex mutex_;
};
```

Then in your main or wherever do something like this:

```
#include <stdio.h>

int main(int argc, char** argv)
{
	IMUPort imu;
	imu.open("/client/inertial");

	yarp.connect("/server/inertial", "/client/inertial");

        // Read the last received value.
        std::printf("IMU returns: %s\n", imu.LastRead().toString().c_str());

	imu.close();
}
```

closed 2018-09-09 by mhubii

traversaro, 2018-09-12:
Hi @mhubii ,
indeed, the IMUs in YARP have been traditionally been exposed using the `yarp::dev::IGenericSensor` C++ interface, that is exposed on the YARP network using the device `inertial` (implemented in the `yarp::dev::ServerInertial` class), that does not have a "client" corresponding device. We have implemented one in https://github.com/robotology/codyco-modules/blob/master/src/devices/genericSensorClient/GenericSensorClient.h , if you want to copy/use it. 

However, to overcome all this shortcomings, we implemented the new [Multiple Analog Sensor Interfaces](http://www.yarp.it/group__dev__iface__multiple__analog.html) that are meant to address all this issues. However, the devices in this repo still needs to be migrated to this new interfaces. I will open some new issues for that.