robotology/gazebo-yarp-plugins #392: YARP-indipendent plugin for setting initial joint position

traversaro, 2018-10-23:
This is based on discussion that we had some time ago with @nunoguedelha . 

## Problem 
We would like to specify initial joint positions for included models, something like:
~~~xml
<include>
    <uri>model://icub</uri>
    <pose>0.0 0 0.6 0 0 3.14</pose>
    <!-- Specify joint position- -> 
     ... 
</include>
~~~

## Current situation 
There is a Gazebo issue open on this, see https://bitbucket.org/osrf/gazebo/issues/1072/allow-to-set-initial-joint-pose-in-sdf , and the `gazebo_yarp_controlboard` plugin offers a similar feature using the `initialConfiguration` option https://github.com/robotology/gazebo-yarp-plugins/blob/01ad1d1252b0827033fd57058f066cc790b893c6/plugins/controlboard/src/ControlBoardDriver.cpp#L353 . However, there is no way at the moment to overwrite this parameter in an external model. 

## Possible solution 
I think it is possible to workaround this situation by exploiting the order in which the `Load` method of a ModelPlugin](http://gazebosim.org/tutorials?tut=plugins_model 

## Open problems 
* [ ] How to specify the joint configuration in SDF (I suggest **not** to assume any predefined serialization for the joint of the robot, and instead explicitly indicate the serialization used in someway, possible mapping names to the corresponding included model using  the `hasEnding` method: https://github.com/robotology/gazebo-yarp-plugins/blob/master/libraries/common/include/GazeboYarpPlugins/common.h#L52 .
* [ ] At the moment  with `gazebo-yarp-plugins` we tipically `include` the model, we should check how thing works with "nested" models, see http://gazebosim.org/tutorials?tut=nested_model&cat=build_robot

lrapetti, 2018-10-23:
Possible idea for the conf format:
~~~
 <model name="iCub_fixed">
<include>
    <uri>model://icub</uri>
    <pose>0.0 0 0.6 0 0 3.14</pose>
    <!-- Specify joint position - -> 
</include>
    <plugin name='initial_joints' filename='libgazebo_yarp_initialjoint.so'>
         <joint_initial_position suffix="sdaddas"> 0.0</joint>
         <joint_initial_position suffix="sdaddas"></joint>
         <joint_initial_position suffix="sdaddas"> 0.0</joint>
    </plugin>
</model>
~~~

barbalberto, 2018-10-23:
Maybe you already discuss about this, but personally I have a bad gut feeling about having a plugin whose only job is to initialize another one.

Can it be the other way around? 
The plugins can read the sdf tags and now we are using this feature only to read the configuration file name:
~~~
    <plugin name='controlboard_torso' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_torso.ini</yarpConfigurationFile>
    </plugin>
~~~

We could remove the initial configuration from the .ini file and write it directly in the sdf, like:

~~~
<plugin name='controlboard_torso' filename='libgazebo_yarp_controlboard.so'>
    <yarpConfigurationFile>model://icub/conf/gazebo_icub_torso.ini</yarpConfigurationFile>
    <joint_default_positions> 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 </joint>
</plugin>
~~~

Then, in case the sdf is imported into another one, maybe we can have the plugin to look for actual initialization values by some tag.
This would require to change a bit the plugin implementation, instead of writing a new one.
I think it is more scalable and has no concurrency issue, but I'm not 100% sure if it is feasible or not. It depends on what we can do with sdf information inside a plugin.

~~~
 <model name="iCub_fixed">
<include>
    <uri>model://icub</uri>
    <pose>0.0 0 0.6 0 0 3.14</pose>
    <!-- Specify joint position - -> 
         <torso_joint_initial_position joint="joint_ID"> 0.0 </joint>
         <torso_joint_initial_position joint="joint_ID"> 0.0 </joint>
         <torso_joint_initial_position joint="joint_ID"> 0.0 </joint>
    </plugin>
</model>
~~~

traversaro, 2018-10-23:
> Maybe you already discuss about this, but personally I have a bad gut feeling about having a plugin whose only job is to initialize another one.

Just to clarify, this plugin is just meant to easily initialize the state of the Gazebo model (a workaround for https://bitbucket.org/osrf/gazebo/issues/1072/allow-to-set-initial-joint-pose-in-sdf), and is not specific to the `gazebo_yarp_controlboard` , the only modification to `gazebo_yarp_controlboard` is to move part of the initialization to the first `Update`. This is useful in this case, but is also a useful improvement that can be helpful whenever there is anything else resetting the state of a simulation to a configuration different from the one specified in the YARP plugin (for example if the `gazebo_yarp_controlboard` is used inside a custom main that is using Gazebo API: https://bitbucket.org/osrf/gazebo/src/f4e51e8f64001beda70615ab8982a7d5a4e65b18/examples/stand_alone/custom_main/?at=default).

> We could remove the initial configuration from the .ini file and write it directly in the sdf, like:

That is already how it works, see https://github.com/robotology/icub-models/blob/5190cc50359459ee493cf65d297908741dcce340/iCub/robots/iCubGazeboV2_5_plus/model.urdf#L1929 . 

> Then, in case the sdf is imported into another one, maybe we can have the plugin to look for actual initialization values by some tag.
This would require to change a bit the plugin implementation, instead of writing a new one.
I think it is more scalable and has no concurrency issue, but I'm not 100% sure if it is feasible or not. It depends on what we can do with sdf information inside a plugin.

This is a good idea, even if in the end we decide for not using it for this use case, it can be used to specify arbitrary yarp parameters that overwrite the value given in the plugin's `.ini`, similar to how command line parameters overwrite parameters for regular YARP modules. [ModelPlugins](http://gazebosim.org/tutorials?tut=plugins_model&cat=write_plugin) get an [`sdf::ElementPtr` to their SDF element](http://osrf-distributions.s3.amazonaws.com/sdformat/api/6.0.0/classsdf_1_1Element.html), so using the `GetParent` method you can probably navigate parent tags. 
@lrapetti have you understood this? This could be used to solve both the initial configuration problem and the problem of overwriting the YARP parameters such as `gazeboYarpPluginsRobotName`.

traversaro, 2018-10-23:
Possible alternative idea for the conf file: 
~~~xml
 <model name="iCub_with_custom_position">
    <include>
        <uri>model://icub</uri>
        <pose>0.0 0 0.6 0 0 3.14</pose>
    </include>
    <yarpPluginConfigurationOverride plugin_name='controlboard_torso'>
         <initialConfiguration>0.0 0.0 0.0</joint>
         <yarpCommandLineOptions>--gazeboYarpPluginsRobotName icubCustomPrefix<yarpCommandLineOptions>
    </plugin>
</model>

lrapetti, 2018-10-29:
I was able to create a plugin (`ConfigurationOverride`) that is able to navigate through the `sdf` and change the `initialConfiguration` attribute of a given `controlboard`.
The only problem that I see at the moment is that the `Load()` of the plugin is executed after the `Load()` of the `controlboard` so the changes are not effective. I have to figure out what is the order in which the plugins are loaded, and how to change it.

### .sdf file
I am working with the following `sdf`:
- `Gazebo world` (I define a model taken from `icub_fixed_no_hands` to which I add the `configuration_override` plugin)
```xml
<world name="default">
...
    <!-- iCub -->
    <model name="myiCub">
      <include>
          <plugin name='configuration_override' filename='libgazebo_yarp_configurationoverride.so'>
          </plugin>
          <uri>model://icub_fixed_no_hands</uri>
      </include>
    </model>
  </world>
```
- `icub_fixed_no_hands` `sdf`:it is composed by the standard `icub` model to which the joint `fixed to ground` is added)
```xml
<model name="iCub_fixed_no_hands">
    <include>
      <uri>model://icub</uri>
      <pose>0.0 0 1.0 0 0 3.14</pose>
    </include>

    <joint name="fixed to ground" type="revolute">
      <parent>world</parent>
      <child>iCub::root_link</child>
       <axis>
        <limit>
          <lower>0</lower>
          <upper>0</upper>
        </limit>
        <xyz>0 0 1</xyz>
        </axis>
    </joint>
  </model>
```
- `icub` `sdf`: The `icub` `sdf` contains all the links and all the joints plus 7 plugins (6 `controlboard`, 1 `apply_external_wrench`).
```xml
<model name='iCub'>
    <pose>0 0 0.6 0 0 3.14</pose>
    <link name='root_link'>
        ....
    </link>
    <joint name='l_hip_pitch' type='revolute'>
       ....
    </joint>
      .....
    <plugin name='controlboard_torso' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_torso.ini</yarpConfigurationFile>
    </plugin>
    <plugin name='controlboard_head' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_head.ini</yarpConfigurationFile>
    </plugin>
    <plugin name='controlboard_left_arm_no_arm' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_left_arm_no_hand_for_no_hand_model.ini</yarpConfigurationFile>
      <initialConfiguration>-0.52 0.52 0 0.785 0 0 0.698</initialConfiguration>
    </plugin>
    <plugin name='controlboard_right_arm_no_arm' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_right_arm_no_hand_for_no_hand_model.ini</yarpConfigurationFile>
      <initialConfiguration>-0.52 0.52 0 0.785 0 0 0.698</initialConfiguration>
    </plugin>
    <plugin name='controlboard_left_leg' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_left_leg.ini</yarpConfigurationFile>
    </plugin>
    <plugin name='controlboard_right_leg' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_right_leg.ini</yarpConfigurationFile>
[... 28 lines cut ...]
            |
             -> <b>initialConfiguration</b>
       -> <b>plugin</b>  <i>controlboard_right_arm_no_arm</i>
            |
             -> <b>initialConfiguration</b>
       -> <b>plugin</b>  <i>controlboard_left_leg</i>
            |
             -> <b>initialConfiguration</b>
       -> <b>plugin</b>  <i>controlboard_right_leg</i>
            |
             -> <b>initialConfiguration</b>
       -> <b>plugin</b>  <i>apply_external_wrench</i>
       -> <b>joint</b>  <i>iCub_fixed_no_hands::fixed to ground</i>
       -> <b>plugin</b>  <i>configuration_override</i>
</pre>

The `configuration_override` plugin start from the **plugin**  _configuration_override_ leaf and then:
- `GetParent()` -> move to the **model**
- Iterate among **model** child using `GetFirstElement()` `GetNextElement()` until found a child with `child->HasAttribute("name")`==desired control board.
- from the given control_board element get the element `initialConfiguration` (`child->GetElement("initialConfiguration")`). At this point it is possible to access the initialConfiguration and change its values.

### Load order
As state above, the problem that remains open is the order in which the plugins load function are executed. The order presented in the tree above corresponds to the order obtained when iterating among the elements of the **model** and looks like it corresponds to the order of execution of the `Load()` functions.
From what I can see the order seems to start from the inner defined `sdf` ending with the `world sdf` (so `iCub` -> `icub_fixed_no_hands` -> `world` ), while inside the same `sdf` the order depends on the order in which the elements are defined.
If we could be able to execute the `Load()` of the `configuration_override` plugin first, we may be able to overwrite the `initialConfiguration` that is defined in the inner `sdf`

traversaro, 2018-10-29:
Have you tried to put the configuration override plugin **outside** the the `<include>` tag?

lrapetti, 2018-10-29:
> Have you tried to put the configuration override plugin **outside** the the `<include>` tag?

@traversaro 
Great thanks!
I tested it and it worked! I was able to modify the `initialConfiguration` in order to control the initial configuration of the `controlboard`.

traversaro, 2018-10-29:
Did you tried to navigate the SDF up from the SDF  model  to the parent model that include it, so that we could keep the logic inside one plugin as suggested by @barbalberto ?

lrapetti, 2018-10-29:
@traversaro yes, that's what I did

lrapetti, 2018-10-29:
Basically I moved up to the parent model with `sdf->GetParent`, and then I moved down by iterating among the child elements in order to find the `controlboard` plugins sdf to be modified.

lrapetti, 2018-10-30:
I am thinking on what should be the configuration format.

Something like what presented above by @traversaro would be great
```
    <yarpPluginConfigurationOverride plugin_name='controlboard_torso'>
         <initialConfiguration>0.0 0.0 0.0</initialConfiguration>
    </plugin>
```

But don't know if possible, and how to do it.  What I understood is that a gazebo plugin has to be defined in a precise way with two only attributes (`name` and `filename`), while other information has to be added as attributes. This observation would lead to something like:
```
    <plugin  name='configuration_override' filename='libgazebo_yarp_configurationoverride.so' >
     
  <plugin_name>controlboard_torso</plugin_name>
         <initialConfiguration>0.0 0.0 0.0</initialConfiguration>
    </plugin>
```

traversaro, 2018-10-30:
I think I did not expressed myself correctly. Are you coming to IIT today? We can have a short chat on this.

lrapetti, 2018-10-31:
After discussing with @traversaro I got the solution he proposed.

Instead of having a `configuration_override ` plugin, he propose to add an element to the `model sdf` (e.g. `<yarpPluginConfigurationOverride plugin_name='controlboard_torso'> </yarpPluginConfigurationOverride>`) and in the controlboard `Load()` search for the presence of this element in the parent model. I agree this solution looks better since it avoid adding a new plugin, and simplify the configuration file. 

I am trying to implement also this solution, but I got stuck because it seems I can not add any custom element to a `model sdf`. When I add the `yarpPluginConfigurationOverride ` element with the following configuration:
```xml
<model name="myiCub">
  <yarpPluginConfigurationOverride plugin_name='controlboard_torso'> </yarpPluginConfigurationOverride>
  <include>
    <uri>model://icub_fixed_no_hands</uri>
  </include>
</model>
```
I got the following `warning`:
```
Warning [parser.cc:950] XML Element[yarpPluginConfigurationOverride], child of element[model] not defined in SDF. Ignoring[yarpPluginConfigurationOverride]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
```
and if a search in the model elements I can not find the `yarpPluginConfigurationOverride` element (`model_sdf->HasElement("yarpPluginConfigurationOverride")`->`FALSE`).

I also tried to move the `yarpPluginConfigurationOverride` element in the `<include>`, in that case I was not getting the warning but still `HasElement()` is returning `FALSE`.

So my question is: **Are custom sdf element allowed?** (a part from the elements inside a plugin)

traversaro, 2018-10-31:
A related existing SDFormat issue: https://bitbucket.org/osrf/sdformat/issues/93/change-the-sdf-spec-to-make-less-gazebo .

lrapetti, 2018-11-05:
I have opened an issue at https://bitbucket.org/osrf/sdformat/issues/205/pass-parameter-to-plugin-from-parent-model, and commented https://bitbucket.org/osrf/sdformat/issues/93/change-the-sdf-spec-to-make-less-gazebo

lrapetti, 2018-11-06:
#### dummy plugin
I have implemented the solution using the dummy plugin in my repo [lrapetti/gazebo-yarp-plugins/override_from_controlboard](https://github.com/lrapetti/gazebo-yarp-plugins/tree/override_from_controlboard). The configuration file used is:
```xml
    <!-- iCub -->
    <model name="myiCub">
    <plugin name='configuration_override' filename='gazebo_yarp_dummy.so'>
      <yarpPluginConfigurationOverride plugin_name='controlboard_right_arm_no_arm'> </yarpPluginConfigurationOverride>
      <initialConfiguration>1.0 0.0 0.5 0.785 0 0 0.698</initialConfiguration>
    </plugin>
    <include>
      <uri>model://icub_fixed_no_hands</uri>
    </include>
    </model>
```

#### configuration override plugin
Alternatively, I have pushed in a branch of my repo also the solution including the controlboardOverride plugin [lrapetti/gazebo-yarp-plugins/configuration_override](https://github.com/lrapetti/gazebo-yarp-plugins/tree/configuration_override/plugins/configurationoverride) that uses the following configuration file:
```xml
    <!-- iCub -->
    <model name="myiCub">
    <plugin name='configuration_override' filename='libgazebo_yarp_configurationoverride.so'>
      <yarpPluginConfigurationOverride plugin_name='controlboard_right_arm_no_arm'> </yarpPluginConfigurationOverride>
      <initialConfiguration>1.0 0.0 0.5 0.785 0 0 0.698</initialConfiguration>
    </plugin>
    <include>
      <uri>model://icub_fixed_no_hands</uri>
    </include>
    </model>
```

At this point the two solutions use similar configuration files, but the `configuration override` solution has a clear filename that seems more convenient then the `gazebo_yarp_dummy`. The other advantage is that it doesn't require any change in other plugins (while for the other solution the search and the configuration overriding has to be done inside the `control_board` plugin or any other plugin that requires this feature).

@traversaro @barbalberto 
What do you think? Do you think it may be worth to do a pull request at this stage (in devel or in a branch), since any solution that involves changes in [SDF format](https://bitbucket.org/osrf/sdformat/src/default/) could take long.

lrapetti, 2018-11-07:
> > Comments that apply to both approaches: have you consider what do to in this case:
> > ```
> >     
> >     <model name="myiCub">
> >     <plugin name='configuration_override' filename='libgazebo_yarp_configurationoverride.so'>
> >       <yarpPluginConfigurationOverride plugin_name='controlboard_right_arm_no_arm'> </yarpPluginConfigurationOverride>
> >       <initialConfiguration>1.0 0.0 0.5 0.785 0 0 0.698</initialConfiguration>
> >     </plugin>
> >     <include>
> >       <uri>model://icub_fixed_no_hands</uri>
> >     </include>
> >     <include>
> >       <uri>model://icub_fixed_no_hands</uri>
> >     </include>
> >     </model>
> > ```
> 
> Using `<include>` the links from the included model are embedded directly into the parent model (see [Note on the include SDF element](http://gazebosim.org/tutorials?tut=nested_model&cat=build_robot)), which means that basically there will be just one single `<model>` containing all the element (links and plugins) of the `icub_fixed_no_hands` model repeated twice (I have to check what happens to the name of the plugins, if two plugins with the same attribute name can coexist, or if it is using a postfix like with the model e.g. `controlboard_right_arm_no_arm_0`).
> Depending on this, the element would be overriten in one of the two plugins (the one without `_0` postif), or to both of them.

I tested it, but actually it is not possible to `<include>` the model twice, and only one model is created in the scene (probably the reason is related to how the `<include>` works as explained above). So it seems the problem doesn't apply

lrapetti, 2018-11-07:
@traversaro 
> Possible downsides of the `libgazebo_yarp_configurationoverride`:
> 
> * We strictly depend on the order with which Gazebo loads the plugins. As far as I know, this is not documented, so it could eventually change.

I think this is the major drawback of this method.

> * We strictly depend on the fact that the plugin needs to be defined **before** the included model (not a big problem if properly document, and eventually if checked at runtime (i.e. an error is printed if the plugin is added after the included model.

This is not true, I checked and the plugin can also be after the `<include>` like follows:
```xml
    <!-- iCub -->
    <model name="myiCub">
    <include>
      <uri>model://icub_fixed_no_hands</uri>
    </include>
    <plugin name='configuration_override' filename='libgazebo_yarp_configurationoverride'>
      <yarpPluginConfigurationOverride plugin_name='controlboard_right_arm_no_arm'> </yarpPluginConfigurationOverride>
      <initialConfiguration>-2.0 0.52 0.0 0.785 0 0 0.698</initialConfiguration>
    </plugin>
    </model>
```
The problem I had in the past was when I put the plugin directly in the `<include>`.

> * I do not think it is an intended for a plugin to modify SDF values outside in memory, so again this is something that could change in the  future.

Agree on this.

lrapetti, 2018-11-07:
I also though about a drawback in defining the configuration to override directly in the `model` (the solution that we discussed previously and that is not currently feasible). What would happen if we want to change the same configuration to multiple plugins (e.g. change the initial configuration to two controlboard plugins):
```xml
<model name="myiCub">
  <yarpPluginConfigurationOverride plugin_name='controlboard_torso'> </yarpPluginConfigurationOverride>
  <initialConfiguration>1.0 1.0 0.0 </initialConfiguration>
  <yarpPluginConfigurationOverride plugin_name='controlboard_right_arm_no_arm'> </yarpPluginConfigurationOverride>
  <initialConfiguration>-2.0 0.52 0.0 0.785 0 0 0.698</initialConfiguration>
  <include>
    <uri>model://icub_fixed_no_hands</uri>
  </include>
</model>
```
In this case `yarpPluginConfigurationOverride` and `initialConfiguration` would not be packed together and the only possible correspondence during the sdf tree navigation would be the order in which they are defined. Probably packing them into a plugin could result in a more ordered structure.

traversaro, 2018-11-13:
@diegoferigo  as you sooner or later you will work with simulation scenarios expressed using SDFs, I think we may be interested on this.

lrapetti, 2018-11-16:
@traversaro 
For the time being, we could merge the `yarpPluginConfigurationOverride` plugin in the `devel`. I think that at the moment this solution is more general (able to change any configuration of any plugin), and doesn't require changes in other plugins structure since all the implementation is closed in the `yarpPluginConfigurationOverride` itself.
I think the other solution (the one involving `libgazebo_yarp_gazeboyarpdummy `) at the moment is not well defined since a nice implementation would require changes in Gazebo that may take long.

traversaro, 2018-11-16:
> For the time being, we could merge the yarpPluginConfigurationOverride plugin in the devel. I think that at the moment this solution is more general (able to change any configuration of any plugin), [...]

Ok, feel free to open a PR!

lrapetti, 2018-11-21:
The new plugin has been merged in `devel` with https://github.com/robotology/gazebo-yarp-plugins/pull/401.

@traversaro We may close this issue for the time beeing, and eventually evaluate alternative implementations when migrating to Gazebo 11.