robotology/gazebo-yarp-plugins #395: Investigate support for Gazebo 11 (Ignition Gazebo)

traversaro, 2018-11-10:
Citing [Gazebo October 2018 Newsletter](http://gazebosim.org/blog/newsletter_13):

>  Given the resources of Open Robotics, we have to make trade-offs in our development and maintenance efforts. Gazebo10, to be released in January 2019, will be the last major version of Gazebo based on the code in bitbucket.org/osrf/gazebo. A new version of Gazebo, based on the Ignition libraries, will be part of the Ignition Robotics release in February 2019. We will continue to support older versions of Gazebo including versions 7-10 through their planned EOL dates which reach out to 2023 for Gazebo9. 

The first (prerelease) version of Gazebo 11 will come out in February 2019, and it is currently under development in https://bitbucket.org/ignitionrobotics/ign-gazebo/src/default/, under the name of "Ignition Gazebo". Despite the simulator goes by "Gazebo" name, it is basically a complete rewrite of the Gazebo simulator, using a [Entity–component–system](https://en.wikipedia.org/wiki/Entity%E2%80%93component%E2%80%93system) pattern. As the Gazebo <= 10 plugin API is basically the complete Gazebo C++ API, the `gazebo-yarp-plugins` plugins will need to be adapted to run simulations with YARP-enabled models on the new Gazebo 11. I open this issue to start collecting ideas and resources related to this migrations. 

## Docs on Ignition Gazebo Architecture
* Original Confluence page on Ignition Gazebo (that used to be called Gazebo X, probably that page is slightly outdated but it is nevertheless a useful source of info): https://osrfoundation.atlassian.net/wiki/spaces/GAZ/pages/35740/Gazebo+X
* Tutorial  on Implementing a System Plugin in Ignition Gazebo (not to be confused with System plugins in Gazebo <= 10): https://bitbucket.org/ignitionrobotics/ign-gazebo/src/default/tutorials/create_system_plugins.md

## General Brainstorming
* As all big software  migrations, I fear this one will be long and for several years we will need to have plugins working for Gazebo <= 10. For this reason I think the repo should be able to compile plugins for both Gazebo <= 10 and Gazebo > 11 out of the same branch and out of the same build directory. 
Fortunately all different major versions of Ignition Libraries are  parallel installable, and also Gazebo 10 and Ignition Gazebo will be installable in parallel.  So this  should be doable.
* The amount of code  and logic contained in some plugins (for example the controlboard plugin) is huge and complex. I think the right strategy is try to share as much as possible that code between Gazebo <= 10 driver/plugins and Ignition Gazebo driver/plugins.

traversaro, 2018-12-15:
Relevant ign-gazebo design PR: https://bitbucket.org/ignitionrobotics/ign-gazebo/pull-requests/98/design-doc-beginning-of-plugin-interface/diff .

traversaro, 2019-03-19:
Renamed issue as apparently Gazebo 11 will still be the "old" gazebo, and the "new" Gazebo will be Ignition Gazebo (1|2|3|..).

traversaro, 2019-08-29:
Related issue: https://bitbucket.org/ignitionrobotics/ign-gazebo/issues/35/tutorial-on-how-to-transition-from-gazebo9 .

traversaro, 2020-04-21:
Nice tutorial on how to migrate a plugin from Gazebo Classic to Ignition Gazebo: https://github.com/ignitionrobotics/ign-gazebo/pull/80 .

diegoferigo, 2021-04-07:
https://github.com/ignitionrobotics/ign-gazebo/issues/325 collects upstream contributions providing high-level APIs that could help the migration

traversaro, 2022-05-18:
Changed the title of the issue now that "Ignition Gazebo"  is "Gazebo Sim", see https://community.gazebosim.org/t/a-new-era-for-gazebo/1356 .

traversaro, 2023-05-17:
This is also an interesting tutorial: https://gazebosim.org/api/sim/7/ardupilot.html . cc @GiacomoBisio @lucapa17

traversaro, 2024-02-29:
All work on this is ongoing at https://github.com/robotology/gz-sim-yarp-plugins, closing this issue as it does not make sense to track anything here. fyi @xela-95

closed 2024-02-29 by traversaro