robotology/gazebo-yarp-plugins #397: Compilation broken

pattacini, 2018-11-17:
`devel` branch is broken with the following error message:

```sh
Building CXX object plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/src/VideoTexture.cc.o
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc: In member function ‘virtual void gazebo::VideoTexture::Load(gazebo::rendering::VisualPtr, sdf::ElementPtr)’:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc:87:74: error: ‘_parent’ was not declared in this scope
     bool configuration_loaded = GazeboYarpPlugins::loadConfigModelPlugin(_parent, _sdf, m_parameters);
                                                                          ^~~~~~~
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc:87:74: note: suggested alternative: ‘parent’
     bool configuration_loaded = GazeboYarpPlugins::loadConfigModelPlugin(_parent, _sdf, m_parameters);
                                                                          ^~~~~~~
                                                                          parent
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc:87:83: error: ‘_sdf’ was not declared in this scope
     bool configuration_loaded = GazeboYarpPlugins::loadConfigModelPlugin(_parent, _sdf, m_parameters);
                                                                                   ^~~~
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc:87:83: note: suggested alternative: ‘sdf’
     bool configuration_loaded = GazeboYarpPlugins::loadConfigModelPlugin(_parent, _sdf, m_parameters);
                                                                                   ^~~~
                                                                                   sdf
plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/build.make:62: recipe for target 'plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/src/VideoTexture.cc.o' failed
make[2]: *** [plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/src/VideoTexture.cc.o] Error 1
CMakeFiles/Makefile2:1175: recipe for target 'plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/all' failed
make[1]: *** [plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

cc @lrapetti @traversaro

lrapetti, 2018-11-17:
Error in a copy-paste, I can fix it

lrapetti, 2018-11-17:
Fixed with https://github.com/robotology/gazebo-yarp-plugins/pull/398, I don't know why it was compiling in MacOS

pattacini, 2018-11-17:
Now I'm getting the following error:

```sh
Building CXX object plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/src/VideoTexture.cc.o
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc: In member function ‘virtual void gazebo::VideoTexture::Load(gazebo::rendering::VisualPtr, sdf::ElementPtr)’:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/videotexture/src/VideoTexture.cc:87:99: error: could not convert ‘parent’ from ‘gazebo::rendering::VisualPtr {aka std::shared_ptr<gazebo::rendering::Visual>}’ to ‘gazebo::physics::ModelPtr {aka boost::shared_ptr<gazebo::physics::Model>}’
     bool configuration_loaded = GazeboYarpPlugins::loadConfigModelPlugin(parent, sdf, m_parameters);
                                                                                                   ^
plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/build.make:62: recipe for target 'plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/src/VideoTexture.cc.o' failed
make[2]: *** [plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/src/VideoTexture.cc.o] Error 1
CMakeFiles/Makefile2:1175: recipe for target 'plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/all' failed
make[1]: *** [plugins/videotexture/CMakeFiles/gazebo_yarp_videotexture.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
Maybe it's better to give it a whirl first in Ubuntu then 😉

lrapetti, 2018-11-17:
I see, the point is that `VideoTexture` is attached to a [Visual](http://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1rendering_1_1Visual.html) instead of a [`Model`](http://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1physics_1_1Model.html), so it is not possible to use `loadConfigModelPlugin ` (nor `loadConfigSensorPlugin `). For the moment we can revert the changes on `VideoTexture` in https://github.com/robotology/gazebo-yarp-plugins/pull/394, and then eventually in the future create a `loadConfigVisualPlugin()` function in case we have multiple Visual Plugins.

pattacini, 2018-11-17:
Oki, I'm not involved in this, so you can proceed with your workaround.

traversaro, 2018-11-17:
I suspect you need to install/enable OpenCV to compile this plugin @lrapetti

traversaro, 2018-11-17:
I suspect this is way this was not catched by the CI, thanks @pattacini for reporting!

closed 2018-11-17 by pattacini