robotology/gazebo-yarp-plugins #402: World interface plugin: makeBox

valegagge, 2018-11-19:
I' m using the rpc port of world interface plugin to create a box with a particular name (for example myBox), but the plugin creates a box called myBoxN, where N is the current number of box in the world.

please see code here:
https://github.com/robotology/gazebo-yarp-plugins/blob/154f236e16a18bcc5ef9a9dd26d1a4c8792d8cac/plugins/worldinterface/src/worldproxy.cpp#L260

Could someone tell me if there is an motivation behind this behaviour? (This happens both on matser and on devel branch)
Thanks a lot

valegagge, 2018-11-19:
MakeSphere has the same behaviour, but I needed to changed it according to @randaz81 in order that makesphere creates a sphere with the given name.

please see https://github.com/robotology/gazebo-yarp-plugins/commit/de34ac26bcbd8565d1d55875d7d7c2ec34b93302

In my opinion, any makeObject function with ObjectName as parameter should create an object with the given name.

Since this issue is not blocking for my activities, I prefer to open an issue and ask if this behaviour is desidered before doing a pull request.

So, I wait their comments or comment of MakeBox users :)

lornat75, 2018-11-20:
I can't remember why it was implemented in this way, the behavior you describe makes sense to me.