robotology/gazebo-yarp-plugins #405: Possible bug in gazebo control board

barbalberto, 2018-12-04:
While looking at code I found following lines that looks like a bug to me:

https://github.com/robotology/gazebo-yarp-plugins/blob/154f236e16a18bcc5ef9a9dd26d1a4c8792d8cac/plugins/controlboard/src/ControlBoardDriver.cpp#L618

```
    for (size_t j = 0; j < m_numberOfJoints; ++j) 
    {
        double forceReference = 0;

      ... compute forceReference here ...

        if (check_joint_within_limits_override_torque(j, forceReference)==false) {}
        m_jointPointers[j]->SetForce(0, forceReference);
    }
```

First: the `if` is doing nothing in either cases, what it is supposed to do? 
Second: the function `check_joint_within_limits_override_torque(...)` is actually modifying the value inside, so it is not only a check, the name is quite misleading...
Third: why `setForce` is applied only to joint `0`? Shouldn't it be `setForce(j, value)`?

@traversaro @pattacini

francesco-romano, 2018-12-04:
I can answer for the last question: the index of the `setForce` is not the index of the joint. The joint is retrieved with `m_jointPointers[j]`.
The index is the index of the DoF associated with the joint (if I remember correctly). So for our 1-DoFs joints it is always 0.

traversaro, 2018-12-04:
> First: the `if` is doing nothing in either cases, what it is supposed to do? 

The original code : https://github.com/robotology/gazebo-yarp-plugins/commit/b5a79a8b08f9efca1a8a0318a3647a8452648a12#diff-a01901eeedeed61dae3fabb62c01e331L1526  seems to indicate that that if was used to guard a debug print, that was commented. I think it is safe to remove the if.

> Second: the function `check_joint_within_limits_override_torque(...)` is actually modifying the value inside, so it is not only a check, the name is quite misleading...

I totally agree, feel free to open a PR.

> Third: ...

See @francesco-romano reply.

barbalberto, 2018-12-04:
Ok, many thanks. I'll do a small cleanup commit later on.
Closing this issue since it is clear now it is not a bug

closed 2018-12-04 by barbalberto

prashanthr05, 2018-12-11:
@barbalberto   I think I can explain the first question. If I'm not wrong, what happens is that the reference torque to be set to the joint is first passed through this function to check if this torque is going to push the joint beyond its joint position limit. In case, it does, the signs of the control signal are inverted and the joint is pulled back into its joint limits.