robotology/gazebo-yarp-plugins #413: externalwrench and Gazebo step

triccyx, 2019-02-12:
The time to apply force doesn't take into account of the Gazebo step time.
For example, if the step time in Gazebo is 0.1 msec and real-time factor is 0.25 the force will be applied for 1/4 of the request time.

traversaro, 2019-04-21:
I think this should have been fixed in https://github.com/robotology/gazebo-yarp-plugins/pull/418 ,  can you confirm @Yeshasvitvs ?

Yeshasvitvs, 2019-04-21:
@traversaro this commit https://github.com/robotology/gazebo-yarp-plugins/commit/64199f6306c41ef36020454d52dfa14c72b24a8a addresses time handling

traversaro, 2019-04-21:
Great! Closing as merged in devel. It will part of the 3.2 release of gazebo-yarp-plugins, that will be synchronized with YARP 3.2 release.

closed 2019-04-21 by traversaro