robotology/gazebo-yarp-plugins #415: Make sure that we do not use Pose3::operator*

traversaro, 2019-02-28:
As it will change its semantics in ign-math5, see https://bitbucket.org/ignitionrobotics/ign-math/pull-requests/301/change-definition-of-pose3-operator-to-fix/diff .

traversaro, 2019-02-28:
Related issue: https://github.com/robotology-playground/sdf-modelica .

Yeshasvitvs, 2020-08-28:
@traversaro link in comment https://github.com/robotology/gazebo-yarp-plugins/issues/415#issue-415824954 is broken

traversaro, 2020-08-28:
> 
> 
> @traversaro link in comment [#415 (comment)](https://github.com/robotology/gazebo-yarp-plugins/issues/415#issue-415824954) is broken

Unfortunatly bitbucket now removed all the mercurial repository. There is a backup side, you can find link to it in ign-math's Changelog.