robotology/gazebo-yarp-plugins #417: Limitations of externalWrench plugin

Yeshasvitvs, 2019-03-27:
[externalWrench](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/externalwrench) plugin has a couple of shortcomings:

- There is no error message if the link is not discovered. It only acknowledges if the command format is correctly used i.e `Response: "[ACK] Correct command format"` is shown. The expected command format is `[link] [force] [torque] [duration]`. Any string used in place of `[link]` will give the correct command response instead of giving an error that the link is not discovered.

- Secondly, the `[link]` string is expected to be a full scoped link name of the model spawned in gazebo e.e `iCub::l_hand`. But if the model is spawned with a higher levels of scope like `iCub::iCub::l_hand` the external wrench plugin doesn't have any effect. This behaviour is checked on both @lrapetti machine and my machine. Some of the recent worlds in icub-gazebo-wholebody repo like [icub_standup_world](https://github.com/robotology/icub-gazebo-wholebody/blob/devel/worlds/icub_standup_world/icub_standup.world) uses [include](https://github.com/robotology/icub-gazebo-wholebody/blob/devel/worlds/icub_standup_world/icub_standup.world) tag for including the model in the world file. This is the reason for the addition scope of `iCub` resulting in the final link scope name as `iCub::iCub::l_hand`. So, it is better to have the command format as `[model] [link] [force] [torque] [duration]` and the user has to specify the first model name in the scope e.g `iCub` and the link name `l_hand`. This kinda functionality is already present in the [linkattacher](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/linkattacher) plugin. 

- Finally, the plugin can be used to genereate only one instance of the external wrench and currently it is not possible to use multiple external wrenches. Previously a PR https://github.com/robotology/gazebo-yarp-plugins/pull/293 was opened to add the functionality of multiple external wrenches. We can update the [externalWrench](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/externalwrench) to have this additional feature. 

@traversaro Please let me know your comments

traversaro, 2019-03-27:
I agree with all remarks.

Yeshasvitvs, 2019-03-27:
After a bit of thought, I realized the `[model]` is not necessary for the command as [externalWrench](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/externalwrench)  is actually a model plugin. So, we can simply find the correct link using `[link]` name provided in the comment.

Also, it is a bit tricky to set the error message through rpc reply but this can be shown on the gazebo running terminal. These changes are fixed in https://github.com/Yeshasvitvs/gazebo-yarp-plugins/commit/fb4b0e26d2907f8c56d320386819bfff8ef7db6d

Yeshasvitvs, 2019-03-28:
I added the multiple wrench functionality and tested with different models:

The model launched through a world that has robot link names spawned as `iCub::iCub::r_hand`

[image]

The model that is manually spawned in gazebo with the link names as `iCub::l_hand`

[image]

Users can stick with the old way using either a full scoped link name in their command or simply the link name. As long as the link is discoverable in the model the wrench is applied as per the command. If the link is not discovered in the model, the following error is shown on gazebo running terminal:

[image]

A big change now is that, the previous functinality is if a user gives two wrench commands consequtively and if the duration of the first one is not yet complete before the second command. The first one is dropped and the second wrench command is applied. @traversaro I am not certain if this kinda functionality should still remain. In that case we need to have an additional option to set the functionality of the externalWrench plugin to be set in a single wrench or multiple wrench mode. what do you think ?

Yeshasvitvs, 2019-03-28:
I added the option set the operation mode https://github.com/Yeshasvitvs/gazebo-yarp-plugins/commit/22c7f0cce2c91f519030e944af94fb4c46e67b37 to be either `single` or `multiple` wrench mode. The default mode is `single` as before. A small test showing the switch between single and multiple operation modes is shown below:

[image]

Yeshasvitvs, 2019-03-30:
I updated the logic to clear all the wrenches when the operation mode is set. Also, updated the rpc message handling. Users will see the message on the rpc terminal in case if the requested link is not present in the model. https://github.com/robotology/gazebo-yarp-plugins/pull/418/commits/f796908d68c22d046586b4248a9fc6bf79b1190d

[image]

Yeshasvitvs, 2019-04-03:
A PR https://github.com/robotology/gazebo-yarp-plugins/pull/418 is open addressing this issue.

closed 2019-04-03 by Yeshasvitvs