robotology/gazebo-yarp-plugins #420: Linking on gazebo_yarp_externalWrench plugin is broken on devel

traversaro, 2019-04-18:
In https://github.com/robotology/gazebo-yarp-plugins/pull/418, the `GazeboYarpPlugins::hasEnding` function contained in the `gazebo_yarp_lib_common` library is used in gazebo_yarp_externalWrench plugin is used, but the corresponding library is not linked in the `CMakeLists.txt`'s of the plugin. Depending on the os and build generator system this can create include or linking problems. 

@Yeshasvitvs can you fix this? Thanks!

Yeshasvitvs, 2019-04-18:
@traversaro fixed it in https://github.com/robotology/gazebo-yarp-plugins/pull/422

Yeshasvitvs, 2019-04-18:
@traversaro we still keep this issue open right ?

traversaro, 2019-04-18:
If the fix was merged in master or in devel, then we close it.

commit 2a5dc3ca (2019-04-18): Update CMakeLists with gazebo_yarp_lib_common

Fix #420
closed 2019-04-18 by traversaro