robotology/gazebo-yarp-plugins #421: Cleanup unused CMake checks

traversaro, 2019-04-18:
In the CMake's build scripts, there are a lot of checks of `CMAKE_VERSION` ( https://github.com/robotology/gazebo-yarp-plugins/search?p=3&q=CMAKE_VERSION&type=&utf8=%E2%9C%93  ), there are not useful as now we require CMake >= 3.5 . It would be a good idea to remove them to cleanup the code.

MiladShafiee, 2019-04-18:
@traversaro I'm interested to do this but I don't have any idea about that! :wink:

traversaro, 2019-04-18:
Great, let's discuss this f2f when you want. For a preliminary reading, I suggest to read the YARP's contributing guide, see https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md#workflow .

traversaro, 2019-04-24:
Fixed by @MiladShafiee in https://github.com/robotology/gazebo-yarp-plugins/pull/423, thanks!

closed 2019-04-24 by traversaro