robotology/gazebo-yarp-plugins #427: Add tests in Travis with Gazebo 10

traversaro, 2019-06-03:
With the current structure, this means adding gazebo10 support in https://github.com/robotology-playground/docker_images/blob/master/robotology-tdd/Dockerfile, or change the system to use some vanilla docker  image, and install gazebo & yarp autonomously, as done for example in https://github.com/robotology/gazebo-fmi/blob/master/.ci/install_debian_and_script.sh .

traversaro, 2019-07-03:
If we switch to use our own image, we can also go back to check compilation against Gazebo7, see https://github.com/robotology/gazebo-yarp-plugins/issues/433 for a related issue.

diegoferigo, 2019-07-03:
After more or less 2 years of usage, I came to the conclusion that using generic docker images for CI as the ones provided by [robotology/docker_images](https://github.com/robotology-playground/docker_images) is applicable only in particular cases. Often it is easier vendoring the CI docker image within the same project repository. The trade-off is that project maintainers have to edit individually all the images in case of an update that requires manual modification.

In the vendoring case, I found that a good solution to build images while minimizing manual effort is exploiting cron-based CI jobs. You can find an example [here](https://github.com/robotology/gym-ignition/blob/aa91b25a23c44305b9a62de74a3c0f7eeeed4006/.travis.yml#L58-L68).

traversaro, 2019-07-03:
I totally agree.