robotology/gazebo-yarp-plugins #428: Compilation problem in macOS

kouroshD, 2019-06-06:
~~~
[ 16%] Linking CXX shared library libgazebo_yarp_externalwrench.dylib
Undefined symbols for architecture x86_64:
  "gazebo::msgs::Set(gazebo::msgs::Pose*, ignition::math::Pose3<double> const&)", referenced from:
      gazebo::ApplyExternalWrench::UpdateChild() in ApplyExternalWrench.cc.o
  "gazebo::msgs::Set(gazebo::msgs::Color*, ignition::math::Color const&)", referenced from:
      gazebo::ApplyExternalWrench::UpdateChild() in ApplyExternalWrench.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [plugins/externalwrench/libgazebo_yarp_externalwrench.dylib] Error 1
make[1]: *** [plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
~~~

traversaro, 2019-06-06:
I suspect this is a problem related to the new homebrew version of Gazebo added in https://github.com/osrf/homebrew-simulation/pull/745 , but I am not sure why it was not detected by the robotology-superbuild nightly build, that unfortunately now is broken by https://github.com/robotology/robotology-superbuild/issues/208 .

lrapetti, 2019-09-30:
I am getting the same error after installing `gazebo9` on MacOS via homebrew.

CC @nunoguedelha

kouroshD, 2019-09-30:
I did
```
cd /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation
git checkout b2820c4fcab2126
brew uninstall gazebo9
brew install gazebo9
```
https://github.com/osrf/homebrew-simulation/commit/b2820c4fcab2126aaebf4ee5b0f8773fda94fd43

Since @nunoguedelha  does not have Mojave veriosn of MacOS, he did not face any problem.

I any case, after trying to build GazeboYarpPlugin, I still have the same error.

traversaro, 2020-06-26:
@lrapetti @kouroshD are you still experiencing this error?

lrapetti, 2020-06-26:
> @lrapetti @kouroshD are you still experiencing this error?

lately I was able to use Gazebo9 with gazebo-yarp-plugins so it should be fixed

traversaro, 2020-06-26:
Perfect, thanks!

closed 2020-06-26 by traversaro