robotology/gazebo-yarp-plugins #433: Repository does not compile with Gazebo7

gabrielenava, 2019-07-02:
I am trying to compile this repo in a machine that has Ubuntu 16.04 as operating system.

Furhtermore, I also need ROS. Looking at the [ROS installation](http://wiki.ros.org/ROS/Installation) page, it seems to me that the expected working version of ROS on Ubuntu 16.04 is `Kinetic`. When I installed it, together with ROS also Gazebo 7 and `libgazebo7_dev` have been installed. 

Now, I when compiling `gazebo-yarp-plugin` I got this error:

```
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc: In member function ‘void ExternalWrench::setVisual()’:
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:89:41: error: ‘class gazebo::physics::World’ has no member named ‘Name’
     this->node->Init(model->GetWorld()->Name());
                                         ^
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc: In member function ‘void ExternalWrench::applyWrench()’:
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:152:9: error: ‘Vector3d’ is not a member of ‘gazebo::math’
         math::Vector3d force (wrench.force[0], wrench.force[1], wrench.force[2]);
         ^
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:152:9: note: suggested alternatives:
In file included from /usr/include/ignition/math2/ignition/math/Box.hh:23:0,
                 from /usr/include/ignition/math2/ignition/math.hh:5,
                 from /usr/include/sdformat-4.4/sdf/Param.hh:35,
                 from /usr/include/sdformat-4.4/sdf/Element.hh:24,
                 from /usr/include/sdformat-4.4/sdf/sdf.hh:5,
                 from /usr/include/gazebo-7/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-7/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/include/gazebo/ExternalWrench.hh:18,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:8:
/usr/include/ignition/math2/ignition/math/Vector3.hh:748:29: note:   ‘ignition::math::Vector3d’
     typedef Vector3<double> Vector3d;
                             ^
In file included from /usr/include/gazebo-7/gazebo/msgs/pose.pb.h:27:0,
                 from /usr/include/gazebo-7/gazebo/msgs/any.pb.h:29,
                 from /usr/include/gazebo-7/gazebo/msgs/MessageTypes.hh:8,
                 from /usr/include/gazebo-7/gazebo/msgs/msgs.hh:32,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/include/gazebo/ExternalWrench.hh:18,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:8:
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:45:23: note:   ‘gazebo::msgs::Vector3d’
 class GZ_MSGS_VISIBLE Vector3d : public ::google::protobuf::Message {
                       ^
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:153:9: error: ‘Vector3d’ is not a member of ‘gazebo::math’
         math::Vector3d torque (wrench.torque[0], wrench.torque[1], wrench.torque[2]);
         ^
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:153:9: note: suggested alternatives:
In file included from /usr/include/ignition/math2/ignition/math/Box.hh:23:0,
                 from /usr/include/ignition/math2/ignition/math.hh:5,
                 from /usr/include/sdformat-4.4/sdf/Param.hh:35,
                 from /usr/include/sdformat-4.4/sdf/Element.hh:24,
                 from /usr/include/sdformat-4.4/sdf/sdf.hh:5,
                 from /usr/include/gazebo-7/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-7/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/include/gazebo/ExternalWrench.hh:18,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:8:
/usr/include/ignition/math2/ignition/math/Vector3.hh:748:29: note:   ‘ignition::math::Vector3d’
     typedef Vector3<double> Vector3d;
                             ^
In file included from /usr/include/gazebo-7/gazebo/msgs/pose.pb.h:27:0,
                 from /usr/include/gazebo-7/gazebo/msgs/any.pb.h:29,
                 from /usr/include/gazebo-7/gazebo/msgs/MessageTypes.hh:8,
                 from /usr/include/gazebo-7/gazebo/msgs/msgs.hh:32,
                 from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/include/gazebo/ExternalWrench.hh:18,
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:8:
/usr/include/gazebo-7/gazebo/msgs/vector3d.pb.h:45:23: note:   ‘gazebo::msgs::Vector3d’
 class GZ_MSGS_VISIBLE Vector3d : public ::google::protobuf::Message {
                       ^
[... 171 lines cut ...]
                 from /home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:8:
/usr/include/ignition/math2/ignition/math/Pose3.hh:407:27: note:   ‘ignition::math::Pose3d’
     typedef Pose3<double> Pose3d;
                           ^
/home/gnava/Software/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:168:45: error: ‘linkCoGPose’ was not declared in this scope
         msgs::Set(visualMsg.mutable_pose(), linkCoGPose.Ign());
                                             ^
plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/build.make:86: recipe for target 'plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/src/ExternalWrench.cc.o' failed
make[6]: *** [plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/src/ExternalWrench.cc.o] Error 1
CMakeFiles/Makefile2:646: recipe for target 'plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/all' failed
make[5]: *** [plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[4]: *** [all] Error 2
CMakeFiles/GazeboYARPPlugins.dir/build.make:107: recipe for target 'robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build' failed
make[3]: *** [robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build] Error 2
CMakeFiles/Makefile2:2488: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/all' failed
make[2]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
CMakeFiles/Makefile2:2500: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/rule' failed
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/rule] Error 2
Makefile:1096: recipe for target 'GazeboYARPPlugins' failed
make: *** [GazeboYARPPlugins] Error 2

```

To the best of my knowledge, it seems that something is missing in a Gazebo class: `‘Vector3d’ is not a member of ‘gazebo::math’` so maybe compatibility with Gazebo 7 is broken. Is the analysis correct?

traversaro, 2019-07-02:
I did not realized that https://github.com/robotology/gazebo-yarp-plugins/pull/376 remove Gazebo7 from Travis, my bad.

traversaro, 2019-07-02:
If you want to compile that plugin with Gazebo7, you can start changing the line that give error to follow Gazebo 7's API, for example for the first error in https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/externalwrench/src/ExternalWrench.cc#L89 : 
~~~
#if GAZEBO_MAJOR_VERSION > 7 
 this->node->Init(model->GetWorld()->Name());
#else 
 this->node->Init(model->GetWorld()->GetName());
#endif 
~~~
(see http://answers.gazebosim.org/question/16388/getname-has-not-been-declared-gazebo-9/) and so on and so forth. If you do not need the external wrench plugin, probably the easiest solution is to just disable its compilation on Gazebo7, so change https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/CMakeLists.txt#L4 to : 
~~~cmake
if (NOT GAZEBO_VERSION VERSION_LESS 8.0)
    add_subdirectory(externalwrench)
endif()
~~~
in both cases, it would be great if you could provide a PR. The recently released 3.2.0 release was advertised to have Gazebo 7 support, but apparently that turned out not to be true.

gabrielenava, 2019-07-02:
I do not need the plugin, therefore I chose the easiest solution (the second one). Now I am able to compile the repo. Here is the PR: https://github.com/robotology/gazebo-yarp-plugins/pull/434

diegoferigo, 2019-07-02:
Why not updating only Gazebo from the osrf repository while keeping ROS kinetic installed?

gabrielenava, 2019-07-02:
Originally I installed Gazebo 9 on the machine, but when I tried to install ROS I got a `broken package` error from `apt-get`. Reinstailing didn't fix the issue, the only thing that fixed it was removing Gazebo and install ROS first. I found this solution online: http://answers.gazebosim.org/question/15568/ros-kinetic-desktop-full-can-not-be-installed-after-installing-gazebo/ 

After reinstalling ROS, I discovered that Gazebo 7 and `libgazebo7-dev` have been installed too and I didn't try to upgrade Gazebo, fearing to break both Gazebo and ROS again and to have to reinstall all of them.

traversaro, 2019-07-03:
@gabrielenava Did you check http://gazebosim.org/tutorials/?tut=ros_wrapper_versions ?

gabrielenava, 2019-07-03:
@traversaro I had a look at it now, seems a good option. Anyway I don't think I need Gazebo > 7 or the `externalwrench` plugin for the moment. I will keep this solution in mind in case I need one of the two.

gabrielenava, 2019-07-03:
Fixed with [this commit](https://github.com/robotology/gazebo-yarp-plugins/commit/d748428fcb8b1f75a9981af2263dad0be2171833). Closing the issue.

closed 2019-07-03 by gabrielenava