robotology/gazebo-yarp-plugins #435: Gazebo crashes when removing iCubGazeboV2_5 in Ubuntu 18.04 with Gazebo 9.8.0

prashanthr05, 2019-07-08:
It crashes with the following fault,
``` sh
gzserver: symbol lookup error: /home/pramadoss/iit_ws/robotology-superbuild/build/install/lib/libgazebo_yarp_forcetorque.so: undefined symbol: _ZN4yarp3dev10PolyDriver9closeMainEv
``` 

I also see the error reflected in `libgazebo_yarp_distributedinertials.so`.

I think it has something to do with `ConnectionPtr`.

Possible related issue: 
#282 
#409

prashanthr05, 2019-07-08:
The demangler website is down, so I could not demangle the error. It seems to me something is wrong in the closing of the PolyDriver method, where we usually only release the shared pointer of the Update Event's connection pointer.

traversaro, 2019-07-25:
`yarp::dev::PolyDriver::closeMain` was removed in YARP 3.2.0 : https://github.com/robotology/yarp/commit/0967d0d5bd0b926d1c20d6a45df03922cb590f2d . I suspect that you updated YARP, but you did not recompiled gazebo-yarp-plugins after updating YARP. Can you try to recompile both YARP and gyp and test again? Thanks!

traversaro, 2023-05-23:
We dropped apt support for Ubuntu 18.04 in https://github.com/robotology/gazebo-yarp-plugins/pull/651 , so I think we can close this issue.

closed 2023-05-23 by traversaro