robotology/gazebo-yarp-plugins #436: Different behaviour of the devices when using Gazebo worlds and using multiple models

lrapetti, 2019-07-12:
I have been working with `Gazebo` `worlds` such as those in https://github.com/robotology/icub-gazebo-wholebody/tree/devel/worlds, and I have notice some unexpected behaviour, and some inconsistency with the models loaded from the `Gazebo` interface. Those problems seem to be due to the name of the ports opened by the `gazebo-yarp-plugins` devices.

The first thing to notice, it is that in order to work with multiple models it is required to comment out the `GazeboYarpPluginsRobotName` from the `gazebo_icub_robotname.ini` file (as described for example in https://github.com/robotology/icub-gazebo-wholebody/blob/devel/worlds/two_icubs_standup_world/README.md) in order to avoid having the same name used for all the ports.

If not specified in the `.ini`, the `GazeboYarpPluginsRobotName` is read directly from the name given to the model in Gazebo. In case the models are added manually trough Gazebo interface, their name would be assigned automatically as: `iCub`, `iCub_0`, `iCub_1`, ... ecc). In case of using a Gazebo world instead, the name of the model can be directly set from the configuration file, e.g.:
```
<model name="myModelName">
      <include>
        <pose>0 0 0.63 0 0 3.14</pose>
        <uri>model://iCubGazeboV2_5</uri>
      </include>
    </model>
```
will have `GazeboYarpPluginsRobotName`=`myModelName`.

Now let's see what happens to the Gazebo plugins ports name in the following four case (the model loaded in all the cases was: `iCubGazeboV2_5`):
- **A.** `gazebo_icub_robotname.ini` not commented, model via **interface **(`GazeboYarpPluginsRobotName`=`icubSim`)
- **B.** `gazebo_icub_robotname.ini` not commented, model via **world** (`GazeboYarpPluginsRobotName`=`icubSim`)
- **C.** one model added in the Gazebo **interface** (`GazeboYarpPluginsRobotName`=`iCub`)
- **D.** a second model is add in addiction to the first one in the **interface** (`iCub`, `iCub_0`)
- **E.** a single model is opened trough a Gazebo **world** (`iCub`)
- **F.** two models are opened trough a Gazebo **world** (`iCub`, `iCub_0`)

The following are the `yarp name list` for the 6 cases, in further comments I will describe the differences:
- **A.**
```
registration name /clock ip 10.240.2.14 port 10002 type tcp
registration name /clock/rpc ip 10.240.2.14 port 10003 type tcp
registration name /icub04 ip 10.240.2.14 port 10000 type tcp
registration name /icubSim/applyExternalWrench/rpc:i ip 10.240.2.14 port 10051 type tcp
registration name /icubSim/head/command:i ip 10.240.2.14 port 10032 type tcp
registration name /icubSim/head/rpc:i ip 10.240.2.14 port 10031 type tcp
registration name /icubSim/head/state:o ip 10.240.2.14 port 10033 type tcp
registration name /icubSim/head/stateExt:o ip 10.240.2.14 port 10034 type tcp
registration name /icubSim/inertial ip 10.240.2.14 port 10004 type tcp
registration name /icubSim/left_arm/analog:o ip 10.240.2.14 port 10014 type tcp
registration name /icubSim/left_arm/analog:o/rpc:i ip 10.240.2.14 port 10013 type tcp
registration name /icubSim/left_arm/command:i ip 10.240.2.14 port 10036 type tcp
registration name /icubSim/left_arm/rpc:i ip 10.240.2.14 port 10035 type tcp
registration name /icubSim/left_arm/state:o ip 10.240.2.14 port 10037 type tcp
registration name /icubSim/left_arm/stateExt:o ip 10.240.2.14 port 10038 type tcp
registration name /icubSim/left_foot/analog:o ip 10.240.2.14 port 10008 type tcp
registration name /icubSim/left_foot/analog:o/rpc:i ip 10.240.2.14 port 10007 type tcp
registration name /icubSim/left_hand/inertialMTB ip 10.240.2.14 port 10022 type tcp
registration name /icubSim/left_hand/inertialMTB/rpc:i ip 10.240.2.14 port 10021 type tcp
registration name /icubSim/left_leg/analog:o ip 10.240.2.14 port 10006 type tcp
registration name /icubSim/left_leg/analog:o/rpc:i ip 10.240.2.14 port 10005 type tcp
registration name /icubSim/left_leg/command:i ip 10.240.2.14 port 10048 type tcp
registration name /icubSim/left_leg/inertialMTB ip 10.240.2.14 port 10018 type tcp
registration name /icubSim/left_leg/inertialMTB/rpc:i ip 10.240.2.14 port 10017 type tcp
registration name /icubSim/left_leg/rpc:i ip 10.240.2.14 port 10047 type tcp
registration name /icubSim/left_leg/state:o ip 10.240.2.14 port 10049 type tcp
registration name /icubSim/left_leg/stateExt:o ip 10.240.2.14 port 10050 type tcp
registration name /icubSim/right_arm/analog:o ip 10.240.2.14 port 10016 type tcp
registration name /icubSim/right_arm/analog:o/rpc:i ip 10.240.2.14 port 10015 type tcp
registration name /icubSim/right_arm/command:i ip 10.240.2.14 port 10040 type tcp
registration name /icubSim/right_arm/rpc:i ip 10.240.2.14 port 10039 type tcp
registration name /icubSim/right_arm/state:o ip 10.240.2.14 port 10041 type tcp
registration name /icubSim/right_arm/stateExt:o ip 10.240.2.14 port 10042 type tcp
registration name /icubSim/right_foot/analog:o ip 10.240.2.14 port 10012 type tcp
registration name /icubSim/right_foot/analog:o/rpc:i ip 10.240.2.14 port 10011 type tcp
registration name /icubSim/right_hand/inertialMTB ip 10.240.2.14 port 10024 type tcp
registration name /icubSim/right_hand/inertialMTB/rpc:i ip 10.240.2.14 port 10023 type tcp
registration name /icubSim/right_leg/analog:o ip 10.240.2.14 port 10010 type tcp
registration name /icubSim/right_leg/analog:o/rpc:i ip 10.240.2.14 port 10009 type tcp
registration name /icubSim/right_leg/command:i ip 10.240.2.14 port 10044 type tcp
registration name /icubSim/right_leg/inertialMTB ip 10.240.2.14 port 10020 type tcp
registration name /icubSim/right_leg/inertialMTB/rpc:i ip 10.240.2.14 port 10019 type tcp
registration name /icubSim/right_leg/rpc:i ip 10.240.2.14 port 10043 type tcp
registration name /icubSim/right_leg/state:o ip 10.240.2.14 port 10045 type tcp
[... 335 lines cut ...]
registration name /iCub_0/right_leg/inertialMTB ip 10.240.2.14 port 10055 type tcp
registration name /iCub_0/right_leg/inertialMTB/rpc:i ip 10.240.2.14 port 10054 type tcp
registration name /iCub_0/right_leg/rpc:i ip 10.240.2.14 port 10078 type tcp
registration name /iCub_0/right_leg/state:o ip 10.240.2.14 port 10080 type tcp
registration name /iCub_0/right_leg/stateExt:o ip 10.240.2.14 port 10081 type tcp
registration name /iCub_0/torso/command:i ip 10.240.2.14 port 10063 type tcp
registration name /iCub_0/torso/inertialMTB ip 10.240.2.14 port 10061 type tcp
registration name /iCub_0/torso/inertialMTB/rpc:i ip 10.240.2.14 port 10060 type tcp
registration name /iCub_0/torso/rpc:i ip 10.240.2.14 port 10062 type tcp
registration name /iCub_0/torso/state:o ip 10.240.2.14 port 10064 type tcp
registration name /iCub_0/torso/stateExt:o ip 10.240.2.14 port 10065 type tcp
registration name /icub04 ip 10.240.2.14 port 10000 type tcp
registration name /l_arm_ft_sensor/left_arm/analog:o ip 10.240.2.14 port 10014 type tcp
registration name /l_arm_ft_sensor/left_arm/analog:o/rpc:i ip 10.240.2.14 port 10013 type tcp
registration name /l_foot_ft_sensor/left_foot/analog:o ip 10.240.2.14 port 10008 type tcp
registration name /l_foot_ft_sensor/left_foot/analog:o/rpc:i ip 10.240.2.14 port 10007 type tcp
registration name /l_leg_ft_sensor/left_leg/analog:o ip 10.240.2.14 port 10006 type tcp
registration name /l_leg_ft_sensor/left_leg/analog:o/rpc:i ip 10.240.2.14 port 10005 type tcp
registration name /r_arm_ft_sensor/right_arm/analog:o ip 10.240.2.14 port 10016 type tcp
registration name /r_arm_ft_sensor/right_arm/analog:o/rpc:i ip 10.240.2.14 port 10015 type tcp
registration name /r_foot_ft_sensor/right_foot/analog:o ip 10.240.2.14 port 10012 type tcp
registration name /r_foot_ft_sensor/right_foot/analog:o/rpc:i ip 10.240.2.14 port 10011 type tcp
registration name /r_leg_ft_sensor/right_leg/analog:o ip 10.240.2.14 port 10010 type tcp
registration name /r_leg_ft_sensor/right_leg/analog:o/rpc:i ip 10.240.2.14 port 10009 type tcp
```

lrapetti, 2019-07-12:
Here some relevant observation/problems I have noticed:
- case *A* and *B* opens the very same ports, so everything seems to work fine when the `gazeboYarpPluginsRobotName` is defined directly in the `.ini` file.
- The number of ports for the case *C* *E* are again the same as *A* and *B*, so in case of a single model all the ports are opened in all the cases.
- The ports opened in *C* seems the same as in *A* and *B* (apart the fact that are using `iCub` in place of `icubSim`) so it seems fine
- [ ] in *E*, the analog ports of the FT sensors are opened without using the `gazeboYarpPluginsRobotName` as prefix. They seems to be using the sensor name (e.g. see https://github.com/robotology/icub-models/blob/master/iCub/robots/iCubGazeboV2_5/model.urdf#L2485). This happens also in **F**, leading to the fact that those ports cannot be opened for the second model since they would have the same name, so the plugins runs only for the first model.
- [ ] using the world and two models (case *F*), we also have the problem with the inertial port which is opened only once as `/iCub/inertial` (problem described with more details in https://github.com/robotology/icub-gazebo-wholebody/issues/27). The same problem doesn apply when we don't use the world, in that case we have both `/iCub/inertial` and `/iCub_0/inertial`.
- [ ] [`externalwrench`] plugin doesn't use the `gazeboYarpPluginsRobotName` if not defined in the .ini file. Hence, a part from case *A* and *B*, the port is opened as `//applyExternalWrench/rpc:i` for one model, while it fails for the second model

lrapetti, 2019-07-12:
cc @traversaro @Yeshasvitvs

traversaro, 2019-07-25:
Just to understand, the actual problems are the last three points, right?

traversaro, 2019-07-25:
Which version of Gazebo you used for those tests?

lrapetti, 2019-07-29:
> Just to understand, the actual problems are the last three points, right?

yes

> Which version of Gazebo you used for those tests?

I'm on `Gazebo 9.9`

> I suggest to open three separate issues for the last three points, as the probably affect three different plugins.

I will proceed opening the three issues

lrapetti, 2019-07-30:
> [ ] [`externalwrench`] plugin doesn't use the `gazeboYarpPluginsRobotName` if not defined in the .ini file. Hence, a part from case **A** and **B**, the port is opened as `//applyExternalWrench/rpc:i` for one model, while it fails for the second model

Opened issue: https://github.com/robotology/gazebo-yarp-plugins/issues/438

lrapetti, 2019-07-30:
> * [ ]  in **E**, the analog ports of the FT sensors are opened without using the `gazeboYarpPluginsRobotName` as prefix. They seems to be using the sensor name (e.g. see [...]

isse opened: https://github.com/robotology/gazebo-yarp-plugins/issues/439

lrapetti, 2019-07-30:
> [ ] using the world and two models (case **F**), we also have the problem with the inertial port which is opened only once as `/iCub/inertial` (problem described with more details in [robotology/icub-gazebo-wholebody#27](https://github.com/robotology/icub-gazebo-wholebody/issues/27)). The same problem doesn apply when we don't use the world, in that case we have both `/iCub/inertial` and `/iCub_0/inertial`.

For this point, an issue was already opened at [robotology/icub-gazebo-wholebody#27](https://github.com/robotology/icub-gazebo-wholebody/issues/27), I have moved it into this repo at https://github.com/robotology/gazebo-yarp-plugins/issues/440

lrapetti, 2019-07-30:
For the #438 the solution seems trivial (I described it in the issue), for the other two the solution is obscure to me at the moment.

traversaro, 2019-07-30:
I commented  the three issue with possible leads for the debug. Let me know if anything is not clear.

lrapetti, 2019-07-30:
Thanks @traversaro, I will go through them and try to debug.