robotology/gazebo-yarp-plugins #439: forcetorque wrong prefix when using Gazebo world

lrapetti, 2019-07-30:
As observed in https://github.com/robotology/gazebo-yarp-plugins/issues/436, the `forcetorque` sensor ports are opened with the sensor name instead of the robot name prefix when using a Gazebo world and with the `gazebo_icub_robotname.ini` commented (this is useful when working with multiple robots: https://github.com/robotology/icub-gazebo-wholebody/blob/devel/worlds/two_icubs_standup_world/README.md):
e.g.
- no **world** + `gazebo_icub_robotname.ini`
  ```
  /icubSim/left_foot/analog:o
  /icubSim/left_foot/analog:o/rpc:i
  ```
- **world** + `gazebo_icub_robotname.ini`
  ```
  /icubSim/left_foot/analog:o
  /icubSim/left_foot/analog:o/rpc:i
  ```
- no **world** + `gazebo_icub_robotname.ini` commented
  ```
  /iCub/left_foot/analog:o
  /iCub/left_foot/analog:o/rpc:i
  ```
- **world** + `gazebo_icub_robotname.ini` commented
  ```
  /l_foot_ft_sensor/left_foot/analog:o
  /l_foot_ft_sensor/left_foot/analog:o/rpc:i
  ```

This is a problem because:
- in case of multiple robots only one port is opened
- the devices that should attach to those port should use different port name in case a worl is used or not

From my understanding the name is read from the Gazebo `_sensor` name in https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/forcetorque/src/ForceTorque.cc#L75, which would explain why the name of the sensor is used in the case **world** + `gazebo_icub_robotname.ini` commented, since it is using `l_foot_ft_sensor` that is the name of the sensor as defined in the model (see https://github.com/robotology/icub-models/blob/master/iCub/robots/iCubGazeboV2_5/model.urdf#L2437). But it is not clear to me where/why should be used the `gazeboYarpPluginsRobotName` instead, what is the difference when using the world, and so how to fix it.

traversaro, 2019-07-30:
> From my understanding the name is read from the Gazebo `_sensor` name in https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/forcetorque/src/ForceTorque.cc#L75, which would explain [...]

Do not get confused but all the different quantities called "name". : )
The port in which data of the sensor is published is opened by the **wrapper** device, that is not aware of being used inside gazebo at all, and  is  opened in https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/forcetorque/src/ForceTorque.cc#L85 . The `gazeboYarpPluginsRobotName` variable used by those conf files, if not explicitly defined in the `.ini` file should be obtained in https://github.com/robotology/gazebo-yarp-plugins/blob/06255f1a1b30580b74e237449ba2be79836d2a62/libraries/singleton/src/ConfHelpers.cc#L125 . I guess there  is something different in the sensor name when the model is loaded via the world vs when is spawned, this is probably a Gazebo problem. Not sure if https://bitbucket.org/osrf/gazebo/issues/2228/inconsistent-behavior-in-scoped-name could be related.

traversaro, 2019-07-30:
[image]

This always make me feel like a bad person. :D

lrapetti, 2019-07-30:
> [image]
> 
> This always make me feel like a bad person (we can have a f2f meeting if you want). :D

Thanks! No, I got the point.
It is just that I already went through this 'names' trap, but I got tricked again :D.