robotology/gazebo-yarp-plugins #440: Inertial port is not opened when using two robots via Gazebo world

lrapetti, 2019-07-30:
@lrapetti commented on [Thu Mar 14 2019](https://github.com/robotology/icub-gazebo-wholebody/issues/27)

I found out that when using two `icub` models in a single world (such as explained in [`two_icubs_standup_world`](https://github.com/robotology/icub-gazebo-wholebody/tree/devel/worlds/two_icubs_standup_world)), in the second model (`iCub_0`) the `inertial` port can not be opened (while all the other ports are opended correctly for both the models with the proper prefix `iCub/` or `iCub_0`).

Here is the error I see in the **terminal**:
```
$ gazebo -slibgazebo_yarp_clock.so two_icubs_standup_world/two_icub_standup.world
objc[62903]: Class OgreConfigWindowDelegate is implemented in both /usr/local/opt/ogre1.9/lib/libOgreMain.1.9.0.dylib (0x108bed9d8) and /usr/local/opt/ogre/lib/libOgreMain.1.7.4.dylib (0x11404ee30). One of the two will be used. Which one is undefined.
[DEBUG]GazeboYarpClock loaded. Clock port will be  /clock 
yarp: Port /clock active at tcp://10.240.2.33:10002/
yarp: Port /clock/rpc active at tcp://10.240.2.33:10003/
[INFO]created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /iCub/inertial active at tcp://10.240.2.33:10004/
[INFO]Server Inertial : no ROS initialization required 
[INFO]Starting server Inertial thread
[DEBUG]Writing an Inertial measurement.
[INFO]created wrapper <inertial>. See C++ class yarp::dev::ServerInertial for documentation.
[INFO]created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /iCub/inertial failed to activate at tcp://10.240.2.33:10004/ (address conflict)
[ERROR]ServerInertial, cannot open port  /iCub/inertial 
yarpdev: ***ERROR*** driver <inertial> was found but could not open
[INFO]Closing Server Inertial...
...
```
**observation:** it seems to me that it is opening a the port for the first model `iCub/inertial`, and then for the second model it tries again to open the `inertial` port with the same prefix and it fails with `address conflict`. Instead, the port for the second model should be open with the different prefix: `iCub_0/inertial`.

and here is the list of the **ports** opened for the two robots:
- `iCub` (35 ports)
```
registration name /iCub/head/command:i ip 10.240.2.33 port 10024 type tcp
registration name /iCub/head/rpc:i ip 10.240.2.33 port 10023 type tcp
registration name /iCub/head/state:o ip 10.240.2.33 port 10025 type tcp
registration name /iCub/head/stateExt:o ip 10.240.2.33 port 10026 type tcp
registration name /iCub/inertial ip 10.240.2.33 port 10004 type tcp
registration name /iCub/left_arm/command:i ip 10.240.2.33 port 10028 type tcp
registration name /iCub/left_arm/rpc:i ip 10.240.2.33 port 10027 type tcp
registration name /iCub/left_arm/state:o ip 10.240.2.33 port 10029 type tcp
registration name /iCub/left_arm/stateExt:o ip 10.240.2.33 port 10030 type tcp
registration name /iCub/left_hand/inertialMTB ip 10.240.2.33 port 10210 type tcp
registration name /iCub/left_hand/inertialMTB/rpc:i ip 10.240.2.33 port 10209 type tcp
registration name /iCub/left_leg/command:i ip 10.240.2.33 port 10036 type tcp
registration name /iCub/left_leg/inertialMTB ip 10.240.2.33 port 10206 type tcp
registration name /iCub/left_leg/inertialMTB/rpc:i ip 10.240.2.33 port 10205 type tcp
registration name /iCub/left_leg/rpc:i ip 10.240.2.33 port 10035 type tcp
registration name /iCub/left_leg/state:o ip 10.240.2.33 port 10037 type tcp
registration name /iCub/left_leg/stateExt:o ip 10.240.2.33 port 10038 type tcp
registration name /iCub/right_arm/command:i ip 10.240.2.33 port 10032 type tcp
registration name /iCub/right_arm/rpc:i ip 10.240.2.33 port 10031 type tcp
registration name /iCub/right_arm/state:o ip 10.240.2.33 port 10033 type tcp
registration name /iCub/right_arm/stateExt:o ip 10.240.2.33 port 10034 type tcp
registration name /iCub/right_hand/inertialMTB ip 10.240.2.33 port 10212 type tcp
registration name /iCub/right_hand/inertialMTB/rpc:i ip 10.240.2.33 port 10211 type tcp
registration name /iCub/right_leg/command:i ip 10.240.2.33 port 10040 type tcp
registration name /iCub/right_leg/inertialMTB ip 10.240.2.33 port 10208 type tcp
registration name /iCub/right_leg/inertialMTB/rpc:i ip 10.240.2.33 port 10207 type tcp
registration name /iCub/right_leg/rpc:i ip 10.240.2.33 port 10039 type tcp
registration name /iCub/right_leg/state:o ip 10.240.2.33 port 10041 type tcp
registration name /iCub/right_leg/stateExt:o ip 10.240.2.33 port 10042 type tcp
registration name /iCub/torso/command:i ip 10.240.2.33 port 10020 type tcp
registration name /iCub/torso/inertialMTB ip 10.240.2.33 port 10214 type tcp
registration name /iCub/torso/inertialMTB/rpc:i ip 10.240.2.33 port 10213 type tcp
registration name /iCub/torso/rpc:i ip 10.240.2.33 port 10019 type tcp
registration name /iCub/torso/state:o ip 10.240.2.33 port 10021 type tcp
registration name /iCub/torso/stateExt:o ip 10.240.2.33 port 10022 type tcp
```
- `iCub_0` (34 ports)
```
registration name /iCub_0/head/command:i ip 10.240.2.33 port 10050 type tcp
[... 97 lines cut ...]
registration name /iCub2/right_arm/command:i ip 10.240.2.33 port 10075 type tcp
registration name /iCub2/right_arm/rpc:i ip 10.240.2.33 port 10074 type tcp
registration name /iCub2/right_arm/state:o ip 10.240.2.33 port 10076 type tcp
registration name /iCub2/right_arm/stateExt:o ip 10.240.2.33 port 10077 type tcp
registration name /iCub2/right_hand/inertialMTB ip 10.240.2.33 port 10059 type tcp
registration name /iCub2/right_hand/inertialMTB/rpc:i ip 10.240.2.33 port 10058 type tcp
registration name /iCub2/right_leg/command:i ip 10.240.2.33 port 10079 type tcp
registration name /iCub2/right_leg/inertialMTB ip 10.240.2.33 port 10055 type tcp
registration name /iCub2/right_leg/inertialMTB/rpc:i ip 10.240.2.33 port 10054 type tcp
registration name /iCub2/right_leg/rpc:i ip 10.240.2.33 port 10078 type tcp
registration name /iCub2/right_leg/state:o ip 10.240.2.33 port 10080 type tcp
registration name /iCub2/right_leg/stateExt:o ip 10.240.2.33 port 10081 type tcp
registration name /iCub2/torso/command:i ip 10.240.2.33 port 10063 type tcp
registration name /iCub2/torso/inertialMTB ip 10.240.2.33 port 10061 type tcp
registration name /iCub2/torso/inertialMTB/rpc:i ip 10.240.2.33 port 10060 type tcp
registration name /iCub2/torso/rpc:i ip 10.240.2.33 port 10062 type tcp
registration name /iCub2/torso/state:o ip 10.240.2.33 port 10064 type tcp
registration name /iCub2/torso/stateExt:o ip 10.240.2.33 port 10065 type tcp
```

---

@lrapetti commented on [Fri Jul 12 2019](https://github.com/robotology/icub-gazebo-wholebody/issues/27#issuecomment-510927264)

The problem seems to apply only when the models are opened trough gazebo world (see https://github.com/robotology/gazebo-yarp-plugins/issues/436#issue-467462300 and https://github.com/robotology/gazebo-yarp-plugins/issues/436#issuecomment-510926834)

traversaro, 2019-07-30:
In the IMU case, the name of  the port opened is given by https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/libYARP_dev/src/devices/ServerInertial/ServerInertial.cpp#L316 , so it should used the one specified in https://github.com/robotology/icub-model-generator/blob/master/simmechanics/data/icub2_5/conf/gazebo_icub_inertial.ini#L3 . Again, I suspect that something strange is happening in https://github.com/robotology/gazebo-yarp-plugins/blob/06255f1a1b30580b74e237449ba2be79836d2a62/libraries/singleton/src/ConfHelpers.cc#L126, so it may be worth to add a few debug prints or checkpoints there.

EhsanRanjbari, 2024-01-12:
I more or less have a similar issue. In fact, if I run Gezebo, spawn a model and then close it after a while. Then sometimes, when I want to run the Gazebo again with the same model I get this error after spawning the model:

```bash
[ERROR] |yarp.os.Port|/syntroSim/left_arm/rpc:i| Port /syntroSim/left_arm/rpc:i failed to activate at tcp://10.240.2.41:10002/ (address conflict)
```
The solution so far is just to restart the machine. I use robotology-superbuild on Windows 11.