robotology/gazebo-yarp-plugins #447: Compilation of externalwrench failing on MacOS with Gazebo9

lrapetti, 2019-10-07:
I am trying to compile `gazebo-yarp-plugins` but it is failing with the following error:
```
[ 38%] Linking CXX shared library libgazebo_yarp_externalwrench.dylib
Undefined symbols for architecture x86_64:
  "gazebo::msgs::Set(gazebo::msgs::Pose*, ignition::math::Pose3<double> const&)", referenced from:
      ExternalWrench::applyWrench() in ExternalWrench.cc.o
  "gazebo::msgs::Set(gazebo::msgs::Color*, ignition::math::Color const&)", referenced from:
      ExternalWrench::applyWrench() in ExternalWrench.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [plugins/externalwrench/libgazebo_yarp_externalwrench.dylib] Error 1
make[4]: *** [plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build] Error 2
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
make: *** [all] Error 2
```

I am currently on `MacOS10.14.6`, and I have installed `Gazebo9` via `homebrew`.

From what I see, it may be similar to the problem with gazebo7 fixed by @gabrielenava with https://github.com/robotology/gazebo-yarp-plugins/pull/434.

lrapetti, 2019-10-07:
I also have ROS installed on my laptop, but I checked the cmake configuration and it seems to be pointing to the right files:
```
 gazebo_DIR                       /usr/local/lib/cmake/gazebo                                                                                                                                                               
 gazebo_proto_msgs_lib            /usr/local/Cellar/gazebo9/9.11.0/lib/libgazebo_msgs.dylib 
```

CarlottaSartore, 2019-10-16:
I had had the same error after trying to install ROS on MACOS 10.14.6 and after a gazebo9 re-installation via homebrew.
I succeded in compiling GazeboYARPPlugins by applying the same changes done for gazebo7 as in the issue   https://github.com/robotology/gazebo-yarp-plugins/pull/434

traversaro, 2019-10-17:
If that fixes the problem can you open a PR?  Thanks!

traversaro, 2019-10-17:
Sorry guys, the problem is indeed more fishy and can be properly solved. 
Could you print the output of the compilation done with `make VERBOSE=1`  (that should give the actual invocation command of the linker) and print the output of `nm -gU <library_path>/libgazebo_msgs.dylib` (see https://stackoverflow.com/questions/4506121/how-to-print-a-list-of-symbols-exported-from-a-dynamic-library), that should gives us all the symbols in the msgs library, that in theory should include the missing ones for which the linker is complaining. If you do not have clear what a linker or a symbol are, please read some related material: 
* https://github.com/green7ea/cpp-compilation
* https://gist.github.com/gubatron/32f82053596c24b6bec6
* https://gist.github.com/jvns/7894681

traversaro, 2019-10-17:
What is the difference w.r.t. to https://github.com/robotology/gazebo-yarp-plugins/issues/428 @lrapetti ? 
By the way, all those problems can probably be catched automatically if we enable some form CI on macOS, see https://github.com/robotology/gazebo-yarp-plugins/issues/444 .

lrapetti, 2019-10-17:
@traversaro Probably duplicated.

By the way @kouroshD, did you find any solution to the problem?

DanielePucci, 2019-10-26:
CC @kouroshD

kouroshD, 2019-10-28:
> @traversaro Probably duplicated.
> 
> By the way @kouroshD, did you find any solution to the problem?

No, not yet. The last update was the one I mentioned in the issue  #428 . I am not sure, if #434  is really a solution for the problem we are facing, it is just a workaround.  I will follow the suggestions given by @traversaro  this week.

kouroshD, 2019-11-05:
As @traversaro  suggested the path to the dynamic library is :
` /usr/local/Cellar/gazebo9/9.11.0_1/lib/libgazebo_msgs.dylib`
and a part of the output of the `nm` command is:
```
0000000000089088 T __ZN6gazebo4msgs3SetEPNS0_10QuaternionERKN8ignition4math2v410QuaternionIdEE
00000000000891fa T __ZN6gazebo4msgs3SetEPNS0_20SphericalCoordinatesERKNS_6common20SphericalCoordinatesE
00000000000890b1 T __ZN6gazebo4msgs3SetEPNS0_4PoseERKN8ignition4math2v45Pose3IdEE
00000000000891d7 T __ZN6gazebo4msgs3SetEPNS0_4TimeERKNS_6common4TimeE
0000000000089181 T __ZN6gazebo4msgs3SetEPNS0_5ColorERKN8ignition4math2v45ColorE
0000000000089134 T __ZN6gazebo4msgs3SetEPNS0_5ColorERKNS_6common5ColorE
00000000000894d4 T __ZN6gazebo4msgs3SetEPNS0_5ImageERKNS_6common5ImageE
000000000008935d T __ZN6gazebo4msgs3SetEPNS0_8InertialERKN8ignition4math2v411MassMatrix3IdEE
00000000000893a2 T __ZN6gazebo4msgs3SetEPNS0_8InertialERKN8ignition4math2v48InertialIdEE
000000000008906e T __ZN6gazebo4msgs3SetEPNS0_8Vector2dERKN8ignition4math2v47Vector2IdEE
000000000008904c T __ZN6gazebo4msgs3SetEPNS0_8Vector3dERKN8ignition4math2v47Vector3IdEE
0000000000089415 T __ZN6gazebo4msgs3SetEPNS0_9PlaneGeomERKN8ignition4math2v45PlaneIdEE
00000000000894ab T __ZN6gazebo4msgs3SetERNS_6common5ImageERKNS0_5ImageE
```

by [demangling the results of the symbol table](https://demangler.com/):
```
0000000000089088 T _gazebo::msgs::Set(gazebo::msgs::Quaternion*, ignition::math::v4::Quaternion<double> const&)
00000000000891fa T _gazebo::msgs::Set(gazebo::msgs::SphericalCoordinates*, gazebo::common::SphericalCoordinates const&)
00000000000890b1 T _gazebo::msgs::Set(gazebo::msgs::Pose*, ignition::math::v4::Pose3<double> const&)
00000000000891d7 T _gazebo::msgs::Set(gazebo::msgs::Time*, gazebo::common::Time const&)
0000000000089181 T _gazebo::msgs::Set(gazebo::msgs::Color*, ignition::math::v4::Color const&)
0000000000089134 T _gazebo::msgs::Set(gazebo::msgs::Color*, gazebo::common::Color const&)
00000000000894d4 T _gazebo::msgs::Set(gazebo::msgs::Image*, gazebo::common::Image const&)
000000000008935d T _gazebo::msgs::Set(gazebo::msgs::Inertial*, ignition::math::v4::MassMatrix3<double> const&)
00000000000893a2 T _gazebo::msgs::Set(gazebo::msgs::Inertial*, ignition::math::v4::Inertial<double> const&)
000000000008906e T _gazebo::msgs::Set(gazebo::msgs::Vector2d*, ignition::math::v4::Vector2<double> const&)
000000000008904c T _gazebo::msgs::Set(gazebo::msgs::Vector3d*, ignition::math::v4::Vector3<double> const&)
0000000000089415 T _gazebo::msgs::Set(gazebo::msgs::PlaneGeom*, ignition::math::v4::Plane<double> const&)
00000000000894ab T _gazebo::msgs::Set(gazebo::common::Image&, gazebo::msgs::Image const&)
```
and as we can see it can find the declaration of the `set` method.

@traversaro

traversaro, 2019-11-06:
> and as we can see it can find the declaration of the set method.

The linker is searching for `ignition::math::Pose3<double> const&`, while the linker offers  ` ignition::math::v4::Pose3<double> const&`, and due to the `v4::` inline namespace those are two different symbols. I suspect there is a mismatch between the latest ignition-math4 bottle available in homebrew, and the one used to compile the gazebo9 bottle. There is any reason why you are using `gazebo9` instead of the latest `gazebo10` ? ROS compatibility?

traversaro, 2019-11-06:
Note that if you recompiled from source gazebo9 instead of using the bottle, I would expect everything to work. However, I think the issue is in the homebrew-compiled binaries, so it could make sense to open an issue on  this  in https://github.com/osrf/homebrew-simulation .

kouroshD, 2019-11-06:
> > and as we can see it can find the declaration of the set method.
> 
> The linker is searching for `ignition::math::Pose3<double> const&`, while the library contains ` ignition::math::v4::Pose3<double> const&`, and due to the `v4::` inline namespace those are two different symbols. I suspect there is a mismatch between the latest ignition-math4 bottle available in homebrew, and the one used to compile the gazebo9 bottle. There is any reason why you are using `gazebo9` instead of the latest `gazebo10` ? ROS compatibility?

There is no specific reason, I will update my gazebo to gazebo10 and I will update here about the issue.
I will open an issue as you suggested. I wanted to install gazebo from gazebo superbuild yesterday. Since there were a number of errors and dependencies to fix, I stopped it, and came back again to homebrew installation.

traversaro, 2019-11-06:
> I wanted to install gazebo from gazebo superbuild yesterday. 

Yes, that superbuild is not updated at all, don't trust it at all.

kouroshD, 2019-11-06:
> The linker is searching for ignition::math::Pose3<double> const&, while the library contains ignition::math::v4::Pose3<double> const&, and due to the v4:: inline namespace those are two different [...]

I tried with gazebo10, the result is the same.
If what you said is right, then why @nunoguedelha  does not face any problem (the difference is in our macOS version) and only the users of macOS Mojave with the latest updates has this problem? 
For example, before I update my macOS version some time back to the latest updates of Mojave (I had Mojave even before), I did not have this problem. Same for all the other mac users having this problem.

CarlottaSartore, 2019-11-06:
> only the users of macOS Mojave with the latest updates has this problem?

I am using `Mojave Version 10.14.6` and `Gazebo9 ` (osrf/simulation/gazebo9: stable 9.11.0). 
The first time when I was compiling I had the same issue and I add the modification of  https://github.com/robotology/gazebo-yarp-plugins/pull/434
After changing gazebo plugins branch to `devel`, I removed the comments and I was still able to compile everything and use Gazebo9 and compiling `GazeboYarpPlugins`. 
Therefore, If I am not missing something,  I do not think that the one to blame is the OS version.

traversaro, 2019-11-06:
> If what you said is right, then why @nunoguedelha does not face any problem (the difference is in our macOS version) and only the users of macOS Mojave with the latest updates has this problem?

I guess each bottle is os-specific, so if this bug is in the Mojave bottle is different from @nunoguedelha's bottle.

traversaro, 2019-11-06:
> After changing gazebo plugins branch to devel, I removed the comments and I was still able to compile everything and use Gazebo9 and compiling GazeboYarpPlugins.

Did you run CMake again? Are you sure that the externalwrench plugin is actually compiling?

traversaro, 2019-11-06:
> After changing gazebo plugins branch to devel, I removed the comments and I was still able to compile everything and use Gazebo9 and compiling GazeboYarpPlugins.

It would be useful if you could check if the symbols in your `libgazebo_msgs.dylib`, to check if you have the `v4` in the symbols name or not.

traversaro, 2019-11-06:
Now that I think a bit about it, the strange things is not that the gazebo have the `v4` in their symbols, but rather that the `libgazebo_yarp_externalwrench.dylib` library does not have it. Can you check what is contained in the `Color.hh` of ignition math available in your system?

traversaro, 2019-11-06:
By the way, did you already tried with a clean build of gazebo-yarp-plugins?

kouroshD, 2019-11-06:
> By the way, did you already tried with a clean build of gazebo-yarp-plugins?

If you are with me, yes! It was cleaned when I moved to gazebo10.

lrapetti, 2019-11-06:
> Now that I think a bit about it, the strange things is not that the gazebo have the `v4` in their symbols, but rather that the `libgazebo_yarp_externalwrench.dylib` library does not have it. Can you check what is contained in the `Color.hh` of ignition math available in your system?

The beginning of the file for `ignition-math4` is the following
```
#ifndef IGNITION_MATH_COLOR_HH_
#define IGNITION_MATH_COLOR_HH_

#include <iostream>

#include <ignition/math/Helpers.hh>
#include <ignition/math/Vector3.hh>
#include <ignition/math/config.hh>

namespace ignition
{
  namespace math
  {
    inline namespace IGNITION_MATH_VERSION_NAMESPACE
    {
    /// \class Color Color.hh ignition/math/Color.hh
    /// \brief Defines a color using a red (R), green (G), blue (B), and alpha
    /// (A) component. Each color component is in the range [0..1].
    class IGNITION_MATH_VISIBLE Color
    {
```

In my case I have in my system also `ignition-math3`, where instead is the following
```
#ifndef IGNITION_MATH_COLOR_HH_
#define IGNITION_MATH_COLOR_HH_

#include <iostream>

#include <ignition/math/Helpers.hh>
#include <ignition/math/Vector3.hh>

namespace ignition
{
  namespace math
  {
    /// \class Color Color.hh ignition/math/Color.hh
    /// \brief Defines a color using a red (R), green (G), blue (B), and alpha
    /// (A) component. Each color component is in the range [0..1].
    class IGNITION_VISIBLE Color
    {
```

kouroshD, 2019-11-06:
I could find three versions of ignition math: 
```
/usr/local/Cellar/ignition-math2/2.9.0/include/ignition/math2/ignition/math/Color.hh
/usr/local/Cellar/ignition-math3/3.3.0/include/ignition/math3/ignition/math/Color.hh
/usr/local/Cellar/ignition-math4/4.0.0/include/ignition/math4/ignition/math/Color.hh
```
in all three versions the namespace is as following: `ignition::math::Color`.

CarlottaSartore, 2019-11-06:
I have just done again cmake, `GazeboYarpPlugIn-Clean` and rebuild it.
In the `robotology-superbuild⁩ ▸ ⁨build⁩ ▸ ⁨install⁩ ▸ ⁨lib`⁩ I have got the `libgazebo_yarp_externalwrench.dylb.` 
I am compiling  with `xCode`

I have got two versions of ignition-math:  `ignition-math3` and  i`gnition-math4`.

Color.hh of ignition-math3

```
#ifndef IGNITION_MATH_COLOR_HH_
#define IGNITION_MATH_COLOR_HH_

#include <iostream>

#include <ignition/math/Helpers.hh>
#include <ignition/math/Vector3.hh>

namespace ignition
{
  namespace math
  {
    /// \class Color Color.hh ignition/math/Color.hh
    /// \brief Defines a color using a red (R), green (G), blue (B), and alpha
    /// (A) component. Each color component is in the range [0..1].
    class IGNITION_VISIBLE Color
    {
```

Color.hh of ignition-math4

```
#ifndef IGNITION_MATH_COLOR_HH_
#define IGNITION_MATH_COLOR_HH_

#include <iostream>

#include <ignition/math/Helpers.hh>
#include <ignition/math/Vector3.hh>
#include <ignition/math/config.hh>

namespace ignition
{
  namespace math
  {
    inline namespace IGNITION_MATH_VERSION_NAMESPACE
    {
```

kouroshD, 2019-11-06:
@traversaro  a question, what is the result of this command in Ubuntu? `nm -gU <library_path>/libgazebo_msgs.dylib`

traversaro, 2019-11-06:
> in all three versions the namespace is as following: ignition::math::Color.

Your ignition-math4 `Color.hh` header does not have the `inline namespace IGNITION_MATH_VERSION_NAMESPACE` line? Then that could be the problem.

kouroshD, 2019-11-06:
> > in all three versions the namespace is as following: ignition::math::Color.
> 
> Your ignition-math4 `Color.hh` header does not have the `inline namespace IGNITION_MATH_VERSION_NAMESPACE` line? Then that could be the problem.

my fault! there is that line as written by @lrapetti  @CarlottaSartore ?

traversaro, 2019-11-06:
> @traversaro a question, what is the result of this command in Ubuntu? `nm -gU <library_path>/libgazebo_msgs.dylib`

On Linux, `nm` is a bit different and has different options, but here is the output:
~~~
straversaro@iiticublap103:~/Downloads/debug-cmake-issue$ nm --demangle --dynamic /usr/lib/x86_64-linux-gnu/libgazebo_msgs.so.10.1.0  | grep Set\(gaz
00000000001afa10 T gazebo::msgs::Set(gazebo::msgs::Quaternion*, ignition::math::v4::Quaternion<double> const&)
00000000001b6340 T gazebo::msgs::Set(gazebo::msgs::SphericalCoordinates*, gazebo::common::SphericalCoordinates const&)
00000000001afa50 T gazebo::msgs::Set(gazebo::msgs::Pose*, ignition::math::v4::Pose3<double> const&)
00000000001afb80 T gazebo::msgs::Set(gazebo::msgs::Time*, gazebo::common::Time const&)
00000000001afb20 T gazebo::msgs::Set(gazebo::msgs::Color*, ignition::math::v4::Color const&)
00000000001afdb0 T gazebo::msgs::Set(gazebo::msgs::Image*, gazebo::common::Image const&)
00000000001afba0 T gazebo::msgs::Set(gazebo::msgs::Inertial*, ignition::math::v4::MassMatrix3<double> const&)
00000000001afbf0 T gazebo::msgs::Set(gazebo::msgs::Inertial*, ignition::math::v4::Inertial<double> const&)
00000000001af9f0 T gazebo::msgs::Set(gazebo::msgs::Vector2d*, ignition::math::v4::Vector2<double> const&)
00000000001af9c0 T gazebo::msgs::Set(gazebo::msgs::Vector3d*, ignition::math::v4::Vector3<double> const&)
00000000001afc70 T gazebo::msgs::Set(gazebo::msgs::PlaneGeom*, ignition::math::v4::Plane<double> const&)
00000000001afd60 T gazebo::msgs::Set(gazebo::common::Image&, gazebo::msgs::Image const&)
~~~

traversaro, 2019-11-06:
> > > in all three versions the namespace is as following: ignition::math::Color.
> > 
> > 
> > Your ignition-math4 `Color.hh` header does not have the `inline namespace IGNITION_MATH_VERSION_NAMESPACE` line? Then that could be the problem.
> 
> my fault! there is that line as written by @lrapetti @CarlottaSartore .

Mhh, then the strange thing is why libgazebo_yarp_externalwrench.dylib is looking for the version without the `::v4`, if the `Color.hh` header is correct. Perhaps the compilation of `libgazebo_yarp_externalwrench.dylib` is using incorrectly the ignition-math3`s Color.hh instead of the correct one of ignition-math4?  We could inspect the actual command line command used to compile the libgazebo_yarp_externalwrench's compilation units, but just to be sure could you try to uninstall `igniton-math2` and `ignition-math3` for your system and try to compile gazebo-yarp-plugins from a clean build? Alternatively we can inspect your CMakeCache.txt for references to the include directory of ignition-math2 or ignition-math3 .

kouroshD, 2019-11-06:
with make VERBOSE=1, I can see I have both `/usr/local/lib/libignition-math3.3.3.0.dylib` and `/usr/local/lib/libignition-math4.4.0.0.dylib` for linking with the order mentioned here!

traversaro, 2019-11-06:
Mhh, this is interesting. However, what matters here are the include directories, that you can see by modifying the `ExternalWrench.cc` , and run again the compilation with `make VERBOSE=1`: in this way the `ExternalWrench.cc`  file will be compiled again, and we can see what are its include directories, that indeed define the symbols that it will use.

traversaro, 2019-11-06:
However, what happens if you uninstall ignition-math3 ?  You can't because it is some kind of  transitive dependency of gazebo10 in some way?

kouroshD, 2019-11-06:
No, I am uninstalling them:
```
 brew uninstall ignition-math2
Error: Refusing to uninstall /usr/local/Cellar/ignition-math2/2.9.0
because it is required by sdformat4, which is currently installed.
You can override this and force removal with:
  brew uninstall --ignore-dependencies ignition-math2
kdarvish@MacBook-Pro:~/Desktop/icub_ws/robotology-superbuild/build/robotology/GazeboYARPPlugins$ brew uninstall ignition-math3
Error: Refusing to uninstall /usr/local/Cellar/ignition-math3/3.3.0
because it is required by ignition-msgs0, ignition-transport3 and sdformat5, which are currently installed.
You can override this and force removal with:
  brew uninstall --ignore-dependencies ignition-math3
```
I checked their dependencies with gazebo10, and it seems there is not conflict!
```
brew info gazebo10
osrf/simulation/gazebo10: stable 10.1.0 (bottled), HEAD
Gazebo robot simulator
http://gazebosim.org
Conflicts with:
  gazebo2 (because Differing version of the same formula)
  gazebo3 (because Differing version of the same formula)
  gazebo4 (because Differing version of the same formula)
  gazebo5 (because Differing version of the same formula)
  gazebo6 (because Differing version of the same formula)
  gazebo7 (because Differing version of the same formula)
  gazebo8 (because Differing version of the same formula)
  gazebo9 (because Differing version of the same formula)
/usr/local/Cellar/gazebo10/10.1.0_5 (1,253 files, 125.8MB) *
  Poured from bottle on 2019-11-06 at 14:52:08
From: https://github.com/osrf/homebrew-simulation/blob/master/Formula/gazebo10.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: boost ✔, doxygen ✔, freeimage ✔, graphviz ✔, ignition-fuel-tools1 ✔, ignition-math4 ✔, ignition-msgs1 ✔, ignition-transport4 ✔, libtar ✔, ogre1.9 ✔, ossp-uuid ✔, protobuf ✔, protobuf-c ✔, qt ✔, qwt ✔, sdformat6 ✔, tbb ✔, tinyxml ✔, tinyxml2 ✔, zeromq ✔
Recommended: bullet ✔, dartsim ✔, ffmpeg ✔, gts ✔, simbody ✔
Optional: gdal ✘, player ✘
==> Options
--with-gdal
	Build with gdal support
--with-player
	Build with player support
--without-bullet
	Build without bullet support
--without-dartsim
	Build without dartsim support
--without-ffmpeg
	Build without ffmpeg support
--without-gts
	Build without gts support
--without-simbody
	Build without simbody support
--HEAD
	Install HEAD version
```

traversaro, 2019-11-06:
> No, I am uninstalling them:

It seems that the uninstallation is failing from the message that you pasted.

kouroshD, 2019-11-06:
No, I just put them here as a reference for the check I did for the dependencies. So, gazebo10 does not have dependency to gnition-math3 and gnition-math2. i remove them by following commnads:
```
brew uninstall --ignore-dependencies ignition-math3
brew uninstall --ignore-dependencies ignition-math2
```

kouroshD, 2019-11-06:
It works ! I could Build :)

kouroshD, 2019-11-06:
Thanks a lot @traversaro .

traversaro, 2019-11-06:
So, as a recap: 
* @lrapetti with ignition-math3 and ignition-math4 fails to compile 
* @CarlottaSartore with  ignition-math3 and ignition-math4 is able  to compile 
* @kouroshD with ignition-math2, 3 and 4 was not able to compile, but with just ignition-math4 works fine?

kouroshD, 2019-11-06:
but the question is why at least @lrapetti  and @CarlottaSartore have ignition-math3 ? Their laptop is new and it should be clean when they installed gazebo.

kouroshD, 2019-11-06:
> So, as a recap:
> 
> * @lrapetti with ignition-math3 and ignition-math4 fails to compile
> * @CarlottaSartore with  ignition-math3 and ignition-math4 is able  to compile
> * @kouroshD with ignition-math2, 3 and 4 was not able to compile, but with just ignition-math4 works fine?

@CarlottaSartore  was not using master branch! but in any case, even in devel there should be similar problem!

CarlottaSartore, 2019-11-06:
> but the question is why at least @lrapetti and @CarlottaSartore have ignition-math3 ? Their laptop was clean and it should be clean when they installed gazebo.

I think the presence of ignition-math3 comes from having tried to install `ROS`

lrapetti, 2019-11-06:
> I think the presence of ignition-math3 comes from having tried to install `ROS`

same for me

kouroshD, 2019-11-07:
To add, If we try to install gazebo, `brew install gazebo`, it installs gazebo8 which has the dependency to ` ignition-math3`[here](https://github.com/osrf/homebrew-simulation/blob/master/Formula/gazebo8.rb#L18). Then if we uninstall gazebo `brew uninstall gazebo`, it does not uninstall all the dependencies. So we will end up having ` ignition-math3`!

kouroshD, 2019-11-07:
if you agree we close this issue. @traversaro

traversaro, 2019-11-07:
Before closing the issue, we should at least clearly document that a user must uninstall ignition-math3 or ignition-math2 to compile the gazebo-yarp-plugins with gazebo10 .

lrapetti, 2019-11-08:
> Before closing the issue, we should at least clearly document that a user must uninstall ignition-math3 or ignition-math2 to compile the gazebo-yarp-plugins with gazebo10 .

we may add [here](https://github.com/robotology/robotology-superbuild#system-dependencies-8) a note for MacOS users.

traversaro, 2019-11-08:
I would also add a note in gyp's own docs at https://github.com/robotology/gazebo-yarp-plugins/blob/master/doc/install.md#dependencies .

traversaro, 2019-11-08:
I was unable to reproduce the issue on a clean build on 10.14  in https://github.com/robotology/robotology-superbuild/pull/291 , so unless someone is able to reproduce the issue on a clean system, I would not update the docs and I would just close the issue.

kouroshD, 2019-11-11:
Since on a very clean system, there is not `ignition-math3`, this problem should not happen. Also, this issue is a documentation per se for the people who may have this problem in the future. I agree also to close this issue.

lrapetti, 2019-11-11:
`brew uninstall --ignore-dependencies ignition-math3` worked for fixing the compilation but there was still a runtime problem when using `gazebo-yarp-plugins`:
```
$ gazebo -slibgazebo_yarp_clock.so --verbose
Gazebo multi-robot simulator, version 9.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 9.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:187] Failed to load plugin /Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/libgazebo_yarp_clock.dylib: dlopen(/Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/libgazebo_yarp_clock.dylib, 9): Library not loaded: /usr/local/opt/ignition-math3/lib/libignition-math3.3.dylib
  Referenced from: /usr/local/opt/sdformat5/lib/libsdformat.5.dylib
  Reason: image not found
```

I verified on Kourosh laptop and he was having the same problem. It is due to the fact that `gazebo-yarp-plugins` use `sdformat` library, and the version `sdformat5` (requiring `ignition-math3`) was installed in the machine and used.

The problem is fixed by 
```
brew uninstal sdformat5
brew link sdformat6
```
and recompiling `gazebo-yarp-plugins`

traversaro, 2020-05-03:
Thanks @lrapetti and @kouroshD .

closed 2020-06-26 by traversaro