robotology/gazebo-yarp-plugins #460: Problem of setting control mode(stiff or compliant) during running walking controller

MiladShafiee, 2020-01-16:
I want to change the control mode of the some joints during walking from stiff to compliant  (or vice versa).
For debugging, I try to re-set at each 100 hundred cycle the mode of the joints to stiff mode in `updateModule()` function in the walking controller by following function:

```
    if(!m_InteractionInterface->setInteractionModes(m_JointModeStiffVectorDefult.data()))
    {
         yError() << "[RobotInterface::setIntractionModeToStiffMode] Error while setting the stiff interaction modes of the joints";
        return false;
    }
```
that `m_JointModeStiffVectorDefult` is:
```
        for (unsigned int i=0;i<m_actuatedDOFs;i++)
        {
             m_JointModeStiffVectorDefult.at(i)=yarp::dev::InteractionModeEnum::VOCAB_IM_STIFF;
        }
```

I run the walkingModule and this is the behavior!!!!!

[image]

traversaro, 2020-01-16:
Hi @MiladShafiee , 

>  each 100 hundred cycle

Each 100 hundred cycle of what? 

> I run the walkingModule and this is the behavior!

So, you basically already have the joints in stiff mode, and you get that strange behaviour when you try to "switch again" in stiff mode? Probably there is some wrong reset that is happening when the  stiff mode is set.

traversaro, 2020-01-16:
The problem is  in https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/plugins/controlboard/src/ControlBoardDriverInteractionMode.cpp#L43 . The PID are reset (including its integral term, that is reset to 0) and the setpoint is reset to the current position when changing the control mode, but they should change only if the specified interaction mode is different from the current one, otherwise the function should return without any reset. A Pull Request to address this is welcome.

traversaro, 2020-01-16:
Note that it may be possible that such behavior is present also in physical robots. Furthermore, changing the interaction mode may be an expensive operation in physical robots (the low-level controlboards need to change their  control law, etc etc), so in general it may not be ideal to set the interaction  mode every one second (you can add a check in your code to only set the interaction mode to stiff if it is not stiff at the moment).

MiladShafiee, 2020-01-16:
Hi @traversaro ,

 > > each 100 hundred cycle

> Each 100 hundred cycle of what?

Control Cycle of Walking. Actually I was setting at each control cycle and robot fell down continuously and I tried to set at 100 control cycle to have a delay and to understand the problem better. Now you can see in the simulation that has a discrete error by a delay.

MiladShafiee, 2020-01-16:
I ask some questions to understand better :smile: 

> The PID are reset (including its integral term, that is reset to 0)

Why is reset the integral to zero? I mean at first is it a PID controller and by re-setting the stiff mode does it change to PD controller? I was thinking it's a PD control.

> but they should change only if the specified interaction mode is different from the current one,

> Furthermore, changing the interaction mode may be an expensive operation in physical robots (the low-level controlboards need to change their control law, etc etc), so in general it may not be ideal to set the interaction mode every one second

Actually my goal is to have a compliance mode in torso active. Pushing the robot. Soing step adjustment. Going back to stiff mode in double support to remove the error of the torso orientation again setting the compliance mode at the start of single support. Repeating this in every step.

traversaro, 2020-01-16:
> > The PID are reset (including its integral term, that is reset to 0)
> 
> Why reset to stiff mode changes the integral to zero? I mean at first is it a PID controller and by re-setting the stiff mode does it change to PD controller? I was thinking it's a PD control.

The fact that the PID is reset when interaction mode is set to stiff is a bug.

traversaro, 2020-01-16:
> Actually my goal is to have a compliance mode in torso active. Pushing the robot. Soing step adjustment. Going back to stiff mode in double support to remove the error of the torso orientation again setting the compliance mode at the start of single support. Repeating this in every step.

To be completely honest, on the real iCub I am afraid this will never work due to the disturbances/discontinuities created at the low level by switching between a:
* nested position PD/torque loop (compliant mode) 
* direct position PID (stiff mode) 
clearly you are free to try, but my intuition is that it will be difficult to get it to work. However, this is quite unrelated to this issue that you opened in gazebo-yarp-plugins.

MiladShafiee, 2020-01-16:
>  A Pull Request to address this is welcome.

Would you please guide me in more detail how should we solve this? because I don't know exactly how is the pipeline of that functions.

traversaro, 2020-01-17:
> > A Pull Request to address this is welcome.
> 
> Would you please guide me in more detail how should we solve this? because I don't know exactly how is the pipeline of that functions.

I guess it should be sufficient to add a: 
~~~
// The joint is already in the specified interaction mode, return
// See https://github.com/robotology/gazebo-yarp-plugins/issues/460 for more details
if (m_interactionMode[j] == mode) {
    return;
}
~~~
in https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/plugins/controlboard/src/ControlBoardDriverInteractionMode.cpp#L43 . Once you verified that this change is sufficient, please open a PR following the instructions in https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork .

MiladShafiee, 2020-01-27:
```
// The joint is already in the specified interaction mode, return
// See https://github.com/robotology/gazebo-yarp-plugins/issues/460 for more details
if (m_interactionMode[j] == mode) {
    return true;
}
```
By adding the above code It has the previous behavior and when I use  `yInfo()<<"I'm Here";`  in setIntractionModes function to become sure that whether this part of code runs ro no, but it does not publish anything and is a bit strange!

traversaro, 2020-01-27:
Are you sure that Gazebo is actually loading the modified plugin? 
* Make sure that after the modifications in `gazebo-yarp-plugins` you install the plugins in the appropriate directory (if this requires sudo, probably something is wrong, **do not run commands with sudo without thinking about this**
* Make sure that the `GAZEBO_MODEL_PATH` env variable points to the correct directory (it should be set automatically by the `setup.sh` of the superbuild).

MiladShafiee, 2020-01-27:
Thanks. I built again all of them and also by closing the qtcreator and building in terminal, now it publishes the comments but the behavior is similar to before and  the error is as follows:

```
[INFO]I'm Here1111111111111111 
[INFO]I'm Here1111111111111111 
[INFO]I'm Here1111111111111111 
[INFO]I'm Here1111111111111111 
[ERROR]An hardware fault occurred on joint  2  torque too big! (  -9999  ) 
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
[ERROR]gazebo_yarp_controlboard: you tried to call setPosition  for a joint that is not in POSITION_DIRECT control mode. 
[ERROR]Error while trying to command a streaming position direct message on joint group
```

The `[INFO]I'm Here1111111111111111 ` shows that we go inside of ` if`, but the robot falls down like before.

MiladShafiee, 2020-01-27:
Actually  the robot falls down but the behavior is different:
 
[image]

traversaro, 2020-01-27:
I think  this problem is unrelated to the subject of the issue. Which position references are sending?

MiladShafiee, 2020-01-27:
> I think this problem is unrelated to the subject of the issue. Which position references are sending?

Actually I ran it now multiple times. The above behavior randomly happens! The position references is something that always we send to robot for walking. Probably running the `setIntractionMode` causes some delay that causes this problem.

And also I tried to set only one joint to compliant mode every one second, as soon as I run the walkingModule the robot falls like before, that shows we can not switch between compliance and stiff during walking.

traversaro, 2020-01-27:
Are you sure that the gravity compensation terms is correctly propagated to the low level Impedeance plugins? The behaviour seems such that the gravity is not correctly compensated.

MiladShafiee, 2020-01-28:
> Are you sure that the gravity compensation terms is correctly propagated to the low level Impedeance plugins?

I don't have any idea about what's happening behind of compliance mode. I just tried to tune the stiffness and damping idea in motorgui.
Today I ran around 100 simulations and the behavior of Gazebo sometimes is completely randomly!
I mean for the same gains of stiffness and damping, I see different behaviors sometimes.

Anyway the problem of above bug has been solved. May I open a pull request for that?

traversaro, 2020-01-28:
> Anyway the problem of above bug has been solved. May I open a pull request for that?

Yes, thanks!

traversaro, 2020-01-28:
> I don't have any idea about what's happening behind of compliance mode. I just tried to tune the stiffness and damping idea in motorgui.

In general, I suggest to first get an idea about how things are working, and then try to tune them. : )

MiladShafiee, 2020-01-28:
> In general, I suggest to first get an idea about how things are working, and then try to tune them. : )

Thanks :smile:  Actually I tried and I asked the others but I couldn't find someone that exactly knows it, so I just realized there is something like spring and damper behavior(a PD controller).

traversaro, 2020-01-28:
> 
> 
> > In general, I suggest to first get an idea about how things are working, and then try to tune them. : )
> 
> Thanks 😄 Actually I tried and I asked the others but I couldn't find someone that exactly knows it, so I just realized there is something like spring and damper behavior(a PD controller).

The  impedance controller should  be indeed a PD + gravity compensation loop that computes the set point for an inner torque controller (see http://wiki.icub.org/images/c/cf/ICub_Control_Modes_1_1.pdf).  In the simulation there is no inner torque controller, as the simulation takes directly the torque in the input. On the robot, the `wholeBodyDynamics` devices provides the gravity compensation torque. From the behavior of the simulation, I think that in simulation no gravity compensation is actually used in the impedance control, but we need to check the code for that.

traversaro, 2020-03-05:
Fixed in gazebo-yarp-plugins v3.3.1, see : https://github.com/robotology/gazebo-yarp-plugins/releases/tag/v3.3.1 .

closed 2020-03-05 by traversaro