robotology/gazebo-yarp-plugins #462: Add in the `externalwrench` plugin the option to specify the orientation frame "local"

HosameldinMohamed, 2020-01-28:
In the current implementation of the nice plugin `externalwrench`, the applied wrenches can only be set w.r.t the world reference frame.

See the response to the "help" command:

```
  Insert a command with the following format:
  [link] [force] [torque] [duration]
  e.g. chest 10 0 0 0 0 0 1
  [link]:     (string) Link ID of the robot as specified in robot's SDF
  [force]:    (double x, y, z) Force components in N w.r.t. world reference frame
  [torque]:   (double x, y, z) Torque components in N.m w.r.t world reference frame
  [duration]: (double) Duration of the applied force in seconds
  Note: The reference frame is the base/root robot frame with x pointing backwards and z upwards.
```

It would be nice to have an option to specify the wrenches orientations in the link frame. It could be useful for simulating, for example, the forces of jet engines attached to the robot.

HosameldinMohamed, 2020-01-28:
The interface can be similar to what had been done in https://github.com/robotology/gazebo-yarp-plugins/pull/418

Where the user can write a command:

```
orient_local
```

And the mode will change to "orient_local" where the applied forces will have the orientation of the link frame. Moreover, the user will specify the force component w.r.t the link frame.

Writing the command:

```
orient_global
```

Will switch to the already implemented settings.

- The default setting is `orient_global`

HosameldinMohamed, 2020-01-28:
I have made a draft of the feature, It's in a Fork and a branch https://github.com/HosameldinMohamed/gazebo-yarp-plugins/tree/feature/addOrientOption

if it's okay I can open a pull request.

CC @prashanthr05 @traversaro

HosameldinMohamed, 2020-01-28:
> I have made a draft of the feature, It's in a Fork and a branch https://github.com/HosameldinMohamed/gazebo-yarp-plugins/tree/feature/addOrientOption

The following GIF shows the current status in the forked repo (https://github.com/HosameldinMohamed/gazebo-yarp-plugins/commit/b674a933bc6d5950fa603269c475da25e7b60eb7) when switching from between the modes `orient_local` and `orient_global` modes:
[image]

This shows the current status

traversaro, 2020-01-28:
> 
> 
> I have made a draft of the feature, It's in a Fork and a branch https://github.com/HosameldinMohamed/gazebo-yarp-plugins/tree/feature/addOrientOption
> 
> if it's okay I can open a pull request.
> 
> CC @prashanthr05 @traversaro

Please open a PR, thanks!

HosameldinMohamed, 2020-01-28:
> Please open a PR, thanks!

Done [here](https://github.com/robotology/gazebo-yarp-plugins/pull/463).

prashanthr05, 2020-02-08:
> > I have made a draft of the feature, It's in a Fork and a branch https://github.com/HosameldinMohamed/gazebo-yarp-plugins/tree/feature/addOrientOption
> 
> The following GIF shows the current status in the forked repo ([HosameldinMohamed@b674a93](https://github.com/HosameldinMohamed/gazebo-yarp-plugins/commit/b674a933bc6d5950fa603269c475da25e7b60eb7)) when switching from between the modes `orient_local` and `orient_global` modes:
> [image]
> 
> This shows the current status

Doesn't the force on the left hand that is applied along the x, seem a bit skewed?

HosameldinMohamed, 2020-02-10:
> Doesn't the force on the left hand that is applied along the x, seem a bit skewed in mode orient_local?

@prashanthr05, Here's a zoomed screenshot where the link frame is shown

[image]

It seems that it's aligned with the x-axis.