robotology/gazebo-yarp-plugins #465: iCub: iPositionControl: relativeMove execute correctly but not move the joints

torstenfol, 2020-03-04:
Hey,
I am working with the iCub in Chemnitz. While working with the iCub in the gazebo simulator a come across a problem with the motor control:
I have a problem in using the relativeMove() methods of the IPositionControl interface.
While with the positionMove the joints can be controlled correctly, the joints did not move with the use of relativeMove. This problem does not occur with the iCub-simulator, there the relativeMove are executed correctly.

used Software:
Gazebo V11
Yarp V3.3.101
Yarp-Gazebo-Plugins V3.3.0

Best,
Torsten Follak

traversaro, 2020-03-04:
Hi @torstenfol, thanks for reporting the issue. Can you check if the modification in https://github.com/robotology/gazebo-yarp-plugins/pull/466 fixes the problem for you? Thanks!

torstenfol, 2020-03-04:
@traversaro thank you, I will test it and report the result.

torstenfol, 2020-03-04:
It solved the problem and it works now correctly! Thank you!

closed 2020-03-05 by traversaro

traversaro, 2020-03-05:
Fixed in gazebo-yarp-plugins v3.3.1, see : https://github.com/robotology/gazebo-yarp-plugins/releases/tag/v3.3.1 .