robotology/gazebo-yarp-plugins #473: iCub eyes version/vergence and finger adduction/abduction coupling handlers

xEnVrE, 2020-03-27:
I am trying to understand the code of the coupling handlers for the eyes and the finger adduction/abduction.

#### Eyes

`decouplePos` should move from the physical joint space (i.e. `left` and `right` eyes) to the controlled DoFs space (i.e.`eyes-version` and `eyes-vergence`).

https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L77-L84

Then, shouldn't it be something like this?

```
// version = (left + right)  / 2.0
// vergence = left - right
 double left = current_pos[m_coupledJoints[0]]; 
 double right =  current_pos[m_coupledJoints[1]]; 
 current_pos[m_coupledJoints[0]] = (left + right) / 2.0; 
 current_pos[m_coupledJoints[1]] = left - right; 
```

`decoupleRefPos` should move from the controlled DoFs space (i.e.`eyes-version` and `eyes-vergence`) to the physical joint space (i.e. `left` and `right` eyes)

https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L110-L117

Then, shouldn't it be something like this?

```
// left   = version + vergence / 2.0
// right = version - vergence / 2.0
out[m_coupledJoints[0]] = pos_ref[m_coupledJoints[0]] + pos_ref[m_coupledJoints[1]]/2; 
out[m_coupledJoints[1]] = pos_ref[m_coupledJoints[0]] - pos_ref[m_coupledJoints[1]]/2; 
```

#### Finger adduction/abduction

`decouplePos` should move from the physical joint space (i.e. the four joints associated to index, middle, ring and little) to the controlled DoFs space (i.e. the iCub `{l, r}_hand_finger`, i.e. joint no. 7). 

https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L427-L432

Then, shouldn't it be something like this? This because the range spanned by the unique DoF `{l,r}_hand_finger` is three times the range spanned by the link associated to the little finger (or, alternatively, that of the index or that of the little).

```
current_pos[m_coupledJoints[0]] = current_pos[m_coupledJoints[3]] * 3;
```

`decoupleRefPos` should move from the controlled DoFs space (i.e. the unique DoF `{l,r}_hand_finger`) to the physical joint space (the four joints associated to index, middle, ring and little)

https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L455-L464

Then, shouldn't it be something like this? This because the best we can do is to associate one third of the range spanned by `{l,r}_hand_finger` to each joint (`out[m_coupledJoints[1]]` is set to the `0.0` as the link associated to the middle finger does not move on the real robot).

```
out[m_coupledJoints[0]] = -pos_ref[m_coupledJoints[0]]/3; 
out[m_coupledJoints[1]] = 0.0; 
out[m_coupledJoints[2]] = pos_ref[m_coupledJoints[0]]/3; 
out[m_coupledJoints[3]] = pos_ref[m_coupledJoints[0]]/3; 
```

xEnVrE, 2020-03-27:
cc @traversaro

traversaro, 2020-03-28:
Thanks a lot @xEnVrE for the clear explanation. 

I wonder if @randaz81 or @aerydna remember anything related to this, however it may be tricky because the development dates back to  approximately four years ago. @pattacini may also want to weigh in on this, clearly with absolutely low priority when he has time given the current situation. 

In any case, I try to summarize what you report in a more compact way (correct me if I am wrong): 

##  Eyes

It seems that  gazebo-yarp-plugins' `EyesCouplingHandler` defines the version (in the sense that with definition the existing`decouplePos` and `decoupleRefPos` are coherent) as : 
~~~
version = left + right
~~~
instead of the definition used for  example in https://github.com/robotology/QA/issues/158 and http://wiki.icub.org/wiki/Vergence,_Version_and_Disparity of: 
~~~
version = (left + right)/2
~~~

## Finger adduction/abduction

The  situation is less clear to me here. Probably it would be useful to have a  clear definition with names of the physical and controlled  joints, see https://github.com/robotology/icub-models/issues/30 . If you have a proposal feel free to drop it there.

traversaro, 2020-03-28:
cc @kouroshD

pattacini, 2020-03-28:
I can confirm that the law implemented in the gaze controller's code is:
```
version = (left + right)/2
```
as per our documentation.
I'm also reasonably sure that the FW copies that as well.

traversaro, 2020-03-28:
I dug a bit in old  issues, and I found the original issue related to eyes support in https://github.com/robotology/gazebo-yarp-plugins/issues/149#issue-47657172 . In that issue, apparently I am the one that reported the definition that is currently implemented, without the division by two: 
~~~
eyes_version    =  left_eye_pan + right_eye_pan
~~~
Even if I was  reporting the the Wiki page that contained the `version = (left + right)/2`, I apparently disregarded it on the basic of what was actually implemented in the firmware, i.e. quoting the original issue: 
> as you can check directly in the iCub firmware source code in firmware/motorControllerDsp56f807/position_decoupling.c, line 25 or in firmware/motorControllerDsp56f807/pwm_decoupling.c, line 83

Interestingly, the relative line of code in the firmware have not been modified in the meanwhile, see: 
* https://github.com/robotology/icub-firmware/blob/fcace9ed15ad99b3a8da656d1782ab3a19c7f7f9/motorControllerDsp56f807/common_source_code/position_decoupling.c#L25
* https://github.com/robotology/icub-firmware/blob/fcace9ed15ad99b3a8da656d1782ab3a19c7f7f9/motorControllerDsp56f807/common_source_code/pwm_decoupling.c#L82

In particular, the "version" numbers are documented in https://github.com/robotology/icub-firmware/blob/fcace9ed15ad99b3a8da656d1782ab3a19c7f7f9/motorControllerDsp56f807/4DC/source_code/include/options.h#L9 , and the  one related to the eyes (0x0215 and 0x0115) seems both to use the  formula `eyes_version    =  left_eye_pan + right_eye_pan` . So unless I  am missing something in the firmware (that is perfectly possible) it  seems that the  gazebo-yarp-plugins' EyesCouplingHandler is coherent with the iCub firmware behaviour. 

. 

PS: @xEnVrE given the complexity of the issue, probably  it could make sense to move the discussion on the finger abduction to a separate issue to simplify discussions.

xEnVrE, 2020-03-28:
> PS: @xEnVrE given the complexity of the issue, probably it could make sense to move the discussion on the finger abduction to a separate issue to simplify discussions.

Yep!

pattacini, 2020-03-28:
Eh! If this is true, that's a very good catch then @traversaro 
I've never browsed the FW myself but I clearly remember discussions that gravitated around our main Wiki resource touching on the requirements on the FW as well.

Anyway, let me tone down my assertion above, I can confirm that the gaze controller implements the law with the 2. Now wondering what could be happening with this mismatch between the high-level and the low-level policies. Perhaps the closed-loop is saving our day?

xEnVrE, 2020-03-28:
I updated the first comment of the issue, so that my doubts are explained more clearly.

traversaro, 2020-03-28:
I think I  also found the decoupling code also for ETH-based boards, but also on that case the coupling assumed is `eyes_version    =  left_eye_pan + right_eye_pan`, see: 
https://github.com/robotology/icub-firmware/blob/v1.15.0/emBODY/eBcode/arch-arm/embobj/plus/mc/Controller.c#L1127 .

traversaro, 2020-03-28:
Ok, last stage: I think the coupling information is now stored in configuration files, but even there the assumed version coupling seems to be `eyes_version = left_eye_pan + right_eye_pan` : 
https://github.com/robotology/robots-configuration/blob/v1.15.0/iCubGenova02/hardware/mechanicals/head-eb21-j2_5-mec.xml#L34

xEnVrE, 2020-07-24:
Can this be closed?

traversaro, 2020-07-24:
If they work fine, ok for me!

closed 2020-07-24 by xEnVrE