robotology/gazebo-yarp-plugins #474: iCub finger adduction/abduction coupling handlers

xEnVrE, 2020-03-28:
Similarly to #473, I am trying to understand the code of the coupling handlers for the finger adduction/abduction.

`decouplePos` should move from the physical joint space (i.e. the four joints associated to index, middle, ring and little) to the controlled DoFs space (i.e. the iCub `{l, r}_hand_finger`, i.e. joint no. 7). 

https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L427-L432

Then, shouldn't it be something like this? This because the range spanned by the unique DoF `{l,r}_hand_finger` is three times the range spanned by the link associated to the little finger (or, alternatively, that of the index or that of the little).

```
current_pos[m_coupledJoints[0]] = current_pos[m_coupledJoints[3]] * 3;
```

`decoupleRefPos` should move from the controlled DoFs space (i.e. the unique DoF `{l,r}_hand_finger`) to the physical joint space (the four joints associated to index, middle, ring and little)

https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L455-L464

Then, shouldn't it be something like this? This because the best we can do is to associate one third of the range spanned by `{l,r}_hand_finger` to each joint (`out[m_coupledJoints[1]]` is set to the `0.0` as the link associated to the middle finger does not move on the real robot).

```
out[m_coupledJoints[0]] = -pos_ref[m_coupledJoints[0]]/3; 
out[m_coupledJoints[1]] = 0.0; 
out[m_coupledJoints[2]] = pos_ref[m_coupledJoints[0]]/3; 
out[m_coupledJoints[3]] = pos_ref[m_coupledJoints[0]]/3; 
```

traversaro, 2020-03-28:
The  situation is not really clear to me here. Probably as a  start it would be useful to have a  clear definition/agreement with names of the physical and controlled  joints, see https://github.com/robotology/icub-models/issues/30 . cc @kouroshD

xEnVrE, 2020-07-24:
Can this be closed?

traversaro, 2020-07-24:
If it works fine, ok for me!

closed 2020-07-24 by xEnVrE