robotology/gazebo-yarp-plugins #476: Fix the discrepancy of the output IAnalogSensor device between real robot and yarp-gazebo-plugins

kouroshD, 2020-03-31:
Currently, the outputs of the IAnalogSensor Device from the real robot are the raw values, i.e., 0-255 values.  However, the output of simulated robot are in radians. This behavior has been observed for the values of the analog sensors of the hand joints.

CC @traversaro

traversaro, 2020-03-31:
Note that `IAnalogSensor` interface is actually exposed by several plugins. I think that @kouroshD refers to the `gazebo_yarp_maissensor` plugin, see https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/maissensor` .

xEnVrE, 2020-04-02:
> Currently, the outputs of the IAnalogSensor Device from the real robot are the raw values, i.e., 0-255 values.

Another difference we need to handle is that, on the real robot, `255` is what we get when the finger is fully open (and `0` when is fully closed), i.e. it decreases while the finger closes.

pattacini, 2020-07-15:
@traversaro this issue would be a blocker for our 2020.08 application demo.

traversaro, 2020-07-17:
Hi @kouroshD @pattacini @xEnVrE , 
  can you provide the (nominal) relation that should exist between the measurements of the model joints positions and the readings provided by the MAIS sensor to be aligned with the real robot? So we can try to understand how to implement this relation in the `gazebo_yarp_maissensor` plugin.

pattacini, 2020-07-17:
Hi @traversaro 

Relevant info is provided in this post: https://github.com/robotology/QA/issues/152#issuecomment-256921469.

#### TL;DR
- `255` ➡️ fully extended.
- `0` ➡️ fully bent.

The law is:
```c++
joints[i]=((1.0-std::min(1.0,std::max(0.0,analogs[2+i]/255.0)))*
                       (chain[i].getMax()-chain[i].getMin())+chain[i].getMin())*(180.0/M_PI);
```

traversaro, 2020-07-17:
Thanks. Given that the plugin is extremely specific (it is explicitly named `gazebo_yarp_maissensor`, after all) we can directly encode the inverse of the law presented by @pattacini directly in the code. Clearly it would be then of fundamental importance that the ordering of the joints passed to the plugin in https://github.com/robotology/icub-gazebo/blob/v1.16.0/icub/conf/gazebo_icub_right_hand_mais.ini#L10 and https://github.com/robotology/icub-gazebo/blob/v1.16.0/icub/conf/gazebo_icub_left_hand_mais.ini is exactly the same the one published by MAIS sensors and described in https://github.com/robotology/QA/issues/152#issuecomment-256921469 .

traversaro, 2020-07-17:
In particular, the place that needs to be modified is https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/maissensor/src/MaisSensorDriver.cpp#L211 , the joint hw limits can also be readed from Gazebo.

pattacini, 2020-07-23:
Fixed, isn't it?

traversaro, 2020-07-23:
Yes, fixed by https://github.com/robotology/gazebo-yarp-plugins/pull/502, thanks @xEnVrE .

PR #502 (2020-07-21): [plugins][maissensor] Fix discrepancy of MAIS analog sensor between simulation and real robot
Fixed mismatched behavior, with respect to the real robot, of MAIS analog sensors output

Output is constrained to be between 0 and 255, the former is reached when the joint is >= maximum value, the latter when the joint is <= its minimum value

Maximum and minimum values are obtained from Gazebo hw limits.

Fixes #476
closed 2020-07-23 by traversaro