robotology/gazebo-yarp-plugins #477: [plugins][basestate] Publishes orientation and angular velocity in radians while YARP convention over ports is degrees

prashanthr05, 2020-04-02:
Ignition API provides the Roll, Pitch Yaw angles in radians as can be seen [here](https://osrf-distributions.s3.amazonaws.com/ign-math/api/1.0.0/classignition_1_1math_1_1Quaternion.html).

In the basestate, there is no conversion from radians to degrees as it can be observed [here](https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/basestate/src/BaseStateDriver.cpp#L97). This is against the YARP convention of publishing angles in degrees.

Unfortunately, this is also not specified in the existing [documentation](https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/basestate/include/yarp/dev/BaseStateDriver.h#L52).

Will this be a potential issue? Or do the other plugins in the repository also publish angles in radians?

lrapetti, 2020-04-02:
> Will this be a potential issue? Or do the other plugins in the repository also publish angles in radians?

Other gazebo-yarp plugins are using radiants so in general it is not a problem as long as documented. However, in case this plugin is not used extensively (so that we avoid regression for many users), I would try to be aligned with imus convention in YARP  (not sure what is currently the convention)

lrapetti, 2020-04-02:
> 
> Just to understand, what other plugins use radians? The only use that I remember is the initial_configuration tag of `gazebo_yarp_controlboard`.

That's exactly the one I was thinking of. I guess in that case the choice was to be consistent with sdf convention.

prashanthr05, 2020-04-09:
> Unfortunately, this is also not specified in the existing [documentation](https://github.com/robotology/gazebo-yarp-plugins/blob/8c056e21c10bd231d1659612285386f12432626b/plugins/basestate/include/yarp/dev/BaseStateDriver.h#L52).

This was documented in #479. I will close this issue.

closed 2020-04-09 by prashanthr05