robotology/gazebo-yarp-plugins #487: Cannot access documentation

xEnVrE, 2020-05-14:
The documentation of `gazebo-yarp-plugins` is stored on `gitlab` but, at the moment - at least from my account - is not accessible from the provided link http://robotology.gitlab.io/docs/gazebo-yarp-plugins/master/

Specifically, I get a 404 error.

traversaro, 2020-05-14:
The generation on Gitlab has not been working for a long time, see https://github.com/robotology/gazebo-yarp-plugins/issues/448 . The quickest solution is probably to setup a GitHub Actions-based job to build the docs and upload it in CI, something similar to https://github.com/robotology/idyntree/pull/597 but for this repo.
At the moment I can't work on this, so volunteers are welcome.

rouzinho, 2020-09-09:
Hello,

Is it possible to access the documentation about the iCub and Gazebo at some places ? As you might know, the gitlab links are not leading anywhere. Some years ago I used to work on the iCub but I can see a lot of things changed. Because of covid, I thought it would be a good idea to work with the icub simulator again but I have a lot of trouble to use the icub inside gazebo (it took me a while already to use iCubSim again and run some modules). 
So I guess that if it's not possible to access the doc or the tutorials, I will try to find another simulated robot platform for my experiments. Which is too bad since I have a bunch of modules that are still running on iCubSim from back in the days...

Anyway, thanks in advance if you can help me with this.

traversaro, 2020-09-10:
Hi @rouzinho, as mentioned in the issue, volunteers to fix the documentation generation are welcome. However, the docs are in any case available at https://github.com/robotology/gazebo-yarp-plugins/tree/master/doc, and if you want to generate locally exactly the same docs that were pointed in the gitlab URL, you can do it yourself by installing doxygen and by running the `make dox` command in the `gazebo-yarp-plugins` build directory.

closed 2021-02-24 by traversaro