robotology/gazebo-yarp-plugins #491: Velocity and Mixed control of coupled joints uses the wrong feedback measurements

xEnVrE, 2020-05-23:
Sorry for the long issue but a bit of context is required: the `controlboard` plugin uses a set of methods called `decouple*` and `decoupleRef*` to handle coupled joints.:
- `decouple*` goes from actual physical joints to coupled joints (e.g. from eyes left/right pan to eyes version/vergence)
- `decoupleRef*` does the opposite

Typically, `decouple*` is used when we need to initialize the trajectory generators, while `decoupleRef*` is used to generate measurements for closed loop control at Gazebo level.

An example:

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L481-L492

`m_motPositions` store the values used to close the loop, i.e. those of `m_positions`, and, after that, `m_positions` is overwritten with values generated by `decouplePos` that are used in 

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriverPositionControl.cpp#L19

to initialize the trajectory generator when `positionMove` is called from the user side.

Finally, `m_motPositions` is used to close the loop in

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L573

## Issue with velocity control

In few words, the loop at Gazebo level is closed using references from the actual physical joints and measurements from the coupled-joints world. E.g., references of eyes left and right pan and measurements of eyes version and vergence. Of course, this is wrong.

### Detailed explanation

In case of velocity control, there is not a variable similar to `m_motPositions`, i.e., `m_motVelocities`, but only a variable called `m_velocities`. 

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L481-L492

`m_velocities` is replaced, in case of coupled joints, with measurements from the coupled-joints world, e.g. eyes version and vergence velocities, and, later,

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L583

is used to close the loop against `m_motReferenceVelocities` that, instead, is the reference for actual physical joints, e.g. eyes pan left and right. Indeed, `m_motReferenceVelocities`, in case of coupled joints, is set here

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L560

### Possible solutions:

- at the moment, it seems that the values generated by `decoupleVel(m_velocities)` are not used to generate trajectories because the `RampFilter` for velocity generation only needs the reference velocity 

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriverVelocityControl.cpp#L22

Then, we could simply remove the line 
```
m_coupling_handler[cpl_cnt]->decoupleVel(m_velocities);
```

- another, more clean solution, is to add a `m_motVelocities` variable and use that to close the loop

##  Issue with mixed mode control

As done for position control

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L573

we should use `m_motPositions` to close the loop. 

Instead, in case of `VOCAB_CM_MIXED`, 

https://github.com/robotology/gazebo-yarp-plugins/blob/6275d40549f5226e07847f0f198020416c37d3c6/plugins/controlboard/src/ControlBoardDriver.cpp#L594

`m_positions` is used.

I can open a PR to fix the issues.

cc @traversaro

traversaro, 2020-05-23:
The issue is clear to me, and the solution: 
> another, more clean solution, is to add a m_motVelocities variable and use that to close the loop
seems to be the correct one, thanks! If you could open a PR it would be great.

Even if it is a bugfix, probable to be conservative to merge in devel, what do you think? 
In particular I am not sure how this problem will affect all the other problems we experienced in the past with  coupled joints, what it is your opinion w.r.t. to this? 

Mentioning some users that could be interested on this:  
@ale-git @randaz81 @kouroshD

xEnVrE, 2020-05-23:
> seems to be the correct one, thanks! If you could open a PR it would be great.

Yep!

> probable to be conservative to merge in devel, what do you think?

Fine with me. 

> In particular I am not sure how this problem will affect all the other problems we experienced in the past with coupled joints, what it is your opinion w.r.t. to this?

These days I am working with the models of `icub-models`, specifically, `iCubGazeboV2_5`, in order to add support for fingers and eyes (unfortunately not automatically but adding the required links/joints by hand).

I tested the proposed modifications with the eyes and I was able to control them in velocity mode and also to use the gaze controller that control the eyes in mixed mode and everything seems fine. After a proper re-tuning of the velocity PIDs of the fingers, I could also test controlling the fingers in velocity mode and tells you if everything is working fine (as fingers are another place where we have coupled joints).

I did not perform tests on the old `icub-gazebo` models but consider that, at the moment, without these modifications velocity and mixed mode was not working for, e.g., the eyes due to the  mismatch between references and feedback.

Please consider that, while this bug fix is required per se because solves a wrong behavior, having coupled joints of iCub to work still requires other modifications in `.ini` files and, code-side, on the coupling handlers, that we start discussing in #473 and #474.

traversaro, 2020-05-23:
> Please consider that, while this bug fix is required per se because solves a wrong behavior, having coupled joints of iCub to work still requires other modifications in .ini files and, code-side, on the coupling handlers, that we start discussing in #472, #473 and #474.

Exactly, that's why I think it make sense to start doing this changes in `devel`, so we can work on those for a bit and then release a new version (the next new release of `gazebo-yarp-plugins` is scheduled for 2020/08).

xEnVrE, 2020-05-23:
The variable `m_velocities` is also used in many other control modes to close the loop (talking about the `COMPLIANT` ones). I am afraid I will not be able to test them and I would not change them in the PR to avoid pushing untested code. 

What do you think?

traversaro, 2020-05-23:
I think do whatever keep the code as simple as possible, as I do not think that anyone ever actually use the COMPLIANT flavors of position and mixed for coupled joints too much. If before they were using `m_velocities` but `m_velocities` was actually motor velocities, I would leave them to use the fixed version of `m_velocity` that seems to be the thing that make more sense.

PR #492 (2020-05-26): [plugins][controlboard] Fix wrong feedback measurements for coupled joints
Within this PR, several fixes for control modes in presence of coupled joints:

- add a variable `m_motVelocities` alongside `m_velocities`: `m_velocities` contains the feedback in the coupled-joints space, while `m_motVelocities` have to be used instead to close the loop as it contains the measurements in the implemented-joint space
- use `m_motVelocities` instead of `m_velocities` to close the loop
- use `m_motPositions` instead of `m_positions` to close the loop

**Note:** for position and mixed control modes, a vari [... cut]
closed 2020-05-27 by xEnVrE