robotology/gazebo-yarp-plugins #498: Cannot find "grabber" plugin (yarp + gazebo + iCub with cameras)

kirillin, 2020-07-05:
Hello! 

I run iCub robot with gazebo and want to use its cameras, but I get the errors. I used [robotology-superbuild](https://github.com/robotology/robotology-superbuild) and make it with default CMakeLists.txt. I have ubuntu 20.04 desktop. 

I run
`yarpserver
`
`gazebo icub_with_hands.world`

and while gazebo runs I get 

```
[DEBUG]GazeboYarpCamera Plugin: sensor scoped name is  default::iCub::left_eye_link::left_camera 
[ERROR]Cannot find "grabber" plugin (not built in, and no .ini file found for it)
Check that YARP_DATA_DIRS leads to at least one directory with plugins/grabber.ini or share/yarp/plugins/grabber.ini in it
yarpdev: ***ERROR*** could not find device <grabber>
[ERROR]GazeboYarpCamera Plugin Load failed: error in opening yarp driver 
[ERROR]Unable to get the iFrameGrabberImage interface from the device 
[DEBUG]GazeboYarpCamera Plugin: sensor scoped name is  default::iCub::right_eye_link::right_camera 
[ERROR]Cannot find "grabber" plugin (not built in, and no .ini file found for it)
Check that YARP_DATA_DIRS leads to at least one directory with plugins/grabber.ini or share/yarp/plugins/grabber.ini in it
yarpdev: ***ERROR*** could not find device <grabber>
[ERROR]GazeboYarpCamera Plugin Load failed: error in opening yarp driver 
```

My YARP_DATA_DIRS is
`/home/kika/local/robotology-superbuild/build/install/share/yarp:/home/kika/local/robotology-superbuild/build/install/share/iCub:/home/kika/local/robotology-superbuild/build/install/share/ICUBcontrib`

My `yarpdev --list` output is 

```yarpdev --list
[INFO]Here are devices listed for your system:
[INFO]Device "AudioPlayerWrapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_AudioPlayerWrapper.so library), is a network wrapper.
[INFO]Device "AudioRecorderWrapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_AudioRecorderWrapper.so library), is a network wrapper.
[INFO]Device "JoypadControlClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_JoypadControlClient.so library), is a network wrapper.
[INFO]Device "JoypadControlServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_JoypadControlServer.so library), is a network wrapper.
[INFO]Device "RGBDSensorClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_RGBDSensorClient.so library), is a network wrapper.
[INFO]Device "RGBDSensorWrapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_RGBDSensorWrapper.so library), is a network wrapper.
[INFO]Device "Rangefinder2DClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_Rangefinder2DClient.so library).
[INFO]Device "Rangefinder2DWrapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_Rangefinder2DWrapper.so library).
[INFO]Device "SDLJoypad", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_SDLJoypad.so library), wrapped by "JoypadControlServer".
[INFO]Device "analogServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_analogServer.so library), is a network wrapper.
[INFO]Device "analogsensorclient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_analogsensorclient.so library).
[INFO]Device "batteryClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_batteryClient.so library).
[INFO]Device "batteryWrapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_batteryWrapper.so library), is a network wrapper.
[INFO]Device "controlboardremapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_controlboardremapper.so library), wrapped by "controlboardwrapper2".
[INFO]Device "controlboardwrapper2", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_controlboardwrapper2.so library), is a network wrapper.
[INFO]Device "fakebot", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_fakebot.so library).
[INFO]Device "grabberDual", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_grabberDual.so library), is a network wrapper.
[INFO]Device "group", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_group.so library).
[INFO]Device "inertial", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_inertial.so library), is a network wrapper.
[INFO]Device "localization2DClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_localization2DClient.so library).
[INFO]Device "localization2DServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_localization2DServer.so library).
[INFO]Device "map2DClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_map2DClient.so library).
[INFO]Device "map2DServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_map2DServer.so library).
[INFO]Device "multipleanalogsensorsclient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_multipleanalogsensorsclient.so library).
[INFO]Device "multipleanalogsensorsremapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_multipleanalogsensorsremapper.so library).
[INFO]Device "multipleanalogsensorsserver", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_multipleanalogsensorsserver.so library).
[INFO]Device "navigation2DClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_navigation2DClient.so library).
[INFO]Device "navigation2DServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_navigation2DServer.so library).
[INFO]Device "pipe", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_pipe.so library).
[INFO]Device "remote_controlboard", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_remote_controlboard.so library), wrapped by "controlboardwrapper2".
[INFO]Device "remote_grabber", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_remote_grabber.so library), wrapped by "grabber".
[INFO]Device "remotecontrolboardremapper", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_controlboardremapper.so library), wrapped by "controlboardwrapper2".
[INFO]Device "robotDescriptionClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_robotDescriptionClient.so library).
[INFO]Device "robotDescriptionServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_robotDescriptionServer.so library).
[INFO]Device "serial", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_serial.so library), is a network wrapper.
[INFO]Device "test_grabber", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_test_grabber.so library), wrapped by "grabberDual".
[INFO]Device "test_motor", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_test_motor.so library), wrapped by "controlboard".
[INFO]Device "transformClient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_transformClient.so library).
[INFO]Device "transformServer", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/yarp/yarp_transformServer.so library).
[... cut ...]
[INFO]Device "cartesiancontrollerserver", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/iCub/cartesiancontrollerserver.so library), wrapped by "cartesiancontrollerclient".
[INFO]Device "gazecontrollerclient", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/iCub/gazecontrollerclient.so library), is a network wrapper.
[INFO]Device "imuFilter", available on request (found in /home/kika/local/robotology-superbuild/build/install/lib/iCub/imuFilter.so library).
```

I can't find any references about 'grabber.ini' in old commits. Can anyone help me with it, please!) Thank you!

xEnVrE, 2020-07-05:
Hi @kirillin,

the issue is due to the fact that the `grabber` device is not compiled by default within `yarp`. Instead, as the output you posted confirms, the `grabberDual` device is available.

Can you try changing from `grabber` to `grabberDual` in files

https://github.com/robotology/icub-gazebo/blob/master/icub/conf/left_camera.ini

and

https://github.com/robotology/icub-gazebo/blob/master/icub/conf/right_camera.ini ?

Do not forget to `make install` icub-gazebo in case you installed it somewhere.

kirillin, 2020-07-05:
@xEnVrE, 

you are my hero, it works!!) Great thanks! Also, I compiled `grabber`, but logs say me it deprecated.

xEnVrE, 2020-07-05:
> but logs say me it deprecated.

That's right. I think that the configuration files should be updated.I opened an issue [here](https://github.com/robotology/icub-gazebo/issues/78) in order to change to `grabberDual` by default.

If the camera are now working, please feel free to close the issue.

traversaro, 2020-07-05:
Thanks @kirillin for reporting the issue and @xEnVrE for providing support! 
I think we need that for this `grabber`/`grabberDual` deprecation we also need to update the gazebo-yarp-plugins codebase in s ome points:
* https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/plugins/camera/src/Camera.cc#L51 
* https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/plugins/multicamera/src/MultiCamera.cc#L54
* https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/tutorial/model/camera/camera.ini#L1
* https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/tutorial/model/multicamera/multicamera.ini#L1

Given this, I would keep this issue open until we fix the use  of the deprecated `grabber` device.