robotology/gazebo-yarp-plugins #503: dart-external-odelcpsolver is not searched at CMake time

drdanz, 2020-08-03:
After a successful CMake run, the build failed on my system with this error:

```
/usr/bin/ld: cannot find -ldart-external-odelcpsolver
collect2: error: ld returned 1 exit status
make[2]: *** [libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/build.make:160: libraries/singleton/libgazebo_yarp_singleton.so] Error 1
make[2]: Leaving directory '<gazebo-yarp-plugins>/build'
make[1]: *** [CMakeFiles/Makefile2:724: libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/all] Error 2
make[1]: Leaving directory '<gazebo-yarp-plugins>/build'
make: *** [Makefile:130: all] Error 2
```

This is easily resolved on by installing `libdart-external-odelcpsolver-dev`, but CMake should have warned me that the library was missing.

I think that you can check this just by adding

```cmake
find_package(DART COMPONENTS external-odelcpsolver REQUIRED)
```

traversaro, 2020-08-03:
Which Gazebo version did you installed? On which system? Did you manually compiled Gazebo or are you using some binary packages?

drdanz, 2020-08-03:
libgazebo-dev 11.0.0+dfsg1-4build2 on ubuntu 20.10 groovy nightly build

traversaro, 2020-08-03:
At a first glance, it seems that the `libdart-external-odelcpsolver-dev` package is missing from the dependency of the package, see https://packages.ubuntu.com/groovy/libgazebo-dev . @j-rivero may be interested in this.

drdanz, 2020-08-03:
I think it might actually come from an even deeper issue:

The `/usr/share/dart/cmake/dart_dartTargets.cmake` file from the [`libdart-dev`](https://packages.ubuntu.com/groovy/libdart-dev) package contains:

```cmake
set_target_properties(dart PROPERTIES
  INTERFACE_COMPILE_FEATURES "cxx_std_14"
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"
  INTERFACE_LINK_LIBRARIES "dl;dart-external-odelcpsolver;Eigen3::Eigen;ccd;fcl;assimp;Boost::boost;Boost::system;Boost::filesystem;octomap;Boost::regex"
```

which means that any target linking `dart` will link `dart-external-odelcpsolver` as well.

drdanz, 2020-08-03:
Opened a bug upstream: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=967008

traversaro, 2020-09-15:
To clarify for user that may see this issue: the workaround is simply to install ` libdart-external-odelcpsolver-dev`, i.e.: 
~~~
sudo apt-get install libdart-external-odelcpsolver-dev
~~~