robotology/gazebo-yarp-plugins #504: Gazbo Yarp Plugins won't compile after update/upgrade of homebrew

nunoguedelha, 2020-08-11:
After updating **homebrew**, the build of **GazeboYARPPlugins** is failing with multiple errors like the following:
```
CompileC /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/Objects-normal/x86_64/ForceTorqueDriver.o plugins/forcetorque/src/ForceTorqueDriver.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -target x86_64-apple-macos10.15 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -stdlib=libc++ -Wno-trigraphs -fpascal-strings -O3 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Release\" -Dgazebo_yarp_forcetorque_EXPORTS -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -DBOOST_TEST_DYN_LINK -DBOOST_ALL_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -Wno-sign-conversion -Wno-infinite-recursion -Wno-move -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -Wno-range-loop-analysis -Wno-semicolon-before-method-body -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/Release/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/forcetorque/include/gazebo -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/forcetorque/include/yarp/dev -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/forcetorque/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/common/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton/include -isystem /usr/local/include -isystem /Users/nunoguedelha/tools/gazebo/include -isystem /Users/nunoguedelha/tools/gazebo/include/gazebo-11 -isystem /usr/local/Cellar/bullet/2.89/include/bullet -isystem /usr/local/Cellar/bullet/2.89/include -isystem /usr/local/include/simbody -isystem /usr/local/Cellar/sdformat9/9.2.0_3/include/sdformat-9.2 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -isystem /Users/nunoguedelha/dev/robotology-superbuild/build/install/include -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/fcl/0.6.1/include -isystem /usr/local/include/sdformat-9.2/sdf/.. -isystem /usr/local/include/ignition/cmake2 -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/DerivedSources-normal/x86_64 -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/DerivedSources/x86_64 -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/DerivedSources -F/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/Release -std=c++11 -DNDEBUG -fPIC -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=gnu++17 -MMD -MT dependencies -MF /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/Objects-normal/x86_64/ForceTorqueDriver.d --serialize-diagnostics /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/Objects-normal/x86_64/ForceTorqueDriver.dia -c /Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/forcetorque/src/ForceTorqueDriver.cpp -o /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/forcetorque/GazeboYARPPlugins.build/Release/gazebo_yarp_forcetorque.build/Objects-normal/x86_64/ForceTorqueDriver.o
/Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/forcetorque/src/ForceTorqueDriver.cpp:72:137: error: use of undeclared identifier '_1'
    this->m_updateConnection = gazebo::event::Events::ConnectWorldUpdateBegin(boost::bind(&GazeboYarpForceTorqueDriver::onUpdate, this, _1));
```

Related  issue https://github.com/freeorion/freeorion/pull/3123.

This is due to the fact that:
```
  "The practice of declaring the Bind placeholders (_1, _2, ...) "
  "in the global namespace is deprecated. Please use "
  "<boost/bind/bind.hpp> + using namespace boost::placeholders, "
  "or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior."
```

Which I think was introduced in BOOST 1.73.

I'm on an old version of **GazeboYARPPlugins** (v3.3.0), but I didn't see any fixregarding this up to the latest commit on `devel` nor `master`.

@traversaro did I miss something?

nunoguedelha, 2020-08-11:
@traversaro ,  if you confirm this hasn't been fixed, I'll do the proper PR. For now I pushed the fix is in branch `fix/boost-placeholders-errors`. on top of tag `v3.3.0`.

traversaro, 2020-08-11:
I did not experienced this problem on Windows even with a recent Boost, however it may be possible that on macOs the situation is different. If you could open a PR against master it would be great, thanks!

nunoguedelha, 2020-08-11:
done. @traversaro

PR #505 (2020-08-11): Declared properly the Bind placeholders (_1, _2, ...) with the boost::placeholders namespace
This fixes #504 .

More precisely...
The practice of declaring the Bind placeholders (_1, _2, ...) in the global
namespace is deprecated. Use <boost/bind/bind.hpp> + using namespace boost::placeholders,
or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
(Refer to Boost 1.73 release notes https://www.boost.org/users/history/version_1_73_0.html)
closed 2020-08-11 by traversaro