robotology/gazebo-yarp-plugins #516: Use of RampFilter in velocity control

traversaro, 2020-10-01:
The implementation of the velocity control in `gazebo-yarp-plugins` contains a RampFilter, that limits the reference acceleration. The max acceleration can be set with the `IVelocityControl::setRefAccelerations` , and the default maximum acceleration is set to 10 deg/s^2 for revolute joints and 10 m/s^2 for prismatic joints (https://github.com/robotology/gazebo-yarp-plugins/blob/52491d69365a32ad930f2b858b9f3a43e7596510/plugins/controlboard/src/ControlBoardDriver.cpp#L115). 

However, this logic created problem when porting some VVV exercises from iCub_SIM to Gazebo, as no such logic is present was present on the iCub_SIM. If the logic is not present at all in any YARP-related robot, then I think it make sense to align the Gazebo implementation with the real existing YARP devices, and remove this RampFilter from the `gazebo_yarp_controlboard` implementation. 

@randaz81 do you happen to rember if you added the RampFilter on  `gazebo-yarp-plugins` for some specific reasons? 

@ale-git Do you know if there is any ramp filter of these kind in the implementation of the velocity control of iCub or R1?

traversaro, 2020-10-01:
cc @xEnVrE @GiulioRomualdi  that worked with velocity control in the past.

traversaro, 2020-10-01:
Related commit: https://github.com/vvv-school/tutorial_joint-interface/commit/1a3c9370fb338c4c3e0e5891146e16002f81016b .

pattacini, 2020-10-02:
Updates on this.

Actually, after a better and deeper look, I can confirm that the Cartesian Controller and the Gaze Controller do make use of `setRefAcceleration(s)` methods to remove the constraint on the maximum acceleration when configured in velocity mode.

Relevant snippets:
- https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/cartesianController/ServerCartesianController.cpp#L2179-L2183
- https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/cartesianController/ServerCartesianController.cpp#L2179-L2183