robotology/gazebo-yarp-plugins #522: Some joints goes in HF is the max step size is decrease in Gazebo

GiulioRomualdi, 2020-11-11:
If I decrease the `max step size` in Gazebo from `0.001` to `0.0001` torso and shoulder often go in hardware fault. As far as I understood the hardware fault is manually set if
https://github.com/robotology/gazebo-yarp-plugins/blob/52491d69365a32ad930f2b858b9f3a43e7596510/plugins/controlboard/src/ControlBoardDriver.cpp#L568-L575

I completely understand that this is done to emulate the real robot model but it would be nice if this check can be easily disabled from the configuration file. It will help a lot while running simulations 

cc @prashanthr05 @traversaro and @S-Dafarra

traversaro, 2020-11-11:
I think you can increase the `max_torques` parameter in the conf file, see https://github.com/robotology/gazebo-yarp-plugins/blob/52491d69365a32ad930f2b858b9f3a43e7596510/plugins/controlboard/src/ControlBoardDriver.cpp#L1322  .

traversaro, 2020-11-11:
I am also open to increase the default, remove the default (i.e. requiring the parameter to be set in the .ini conf file) or do whatever it make sense to do.

traversaro, 2020-11-11:
> If I decrease the max step size in Gazebo from 0.001 to 0.0001 torso and shoulder often go in hardware fault. 

By the way, it would be interesting to understand why this happens, by decreasing  the timestamp the numerical errors should decrease.

GiulioRomualdi, 2021-09-03:
Another possibility is to add a flag to disable the check.