robotology/gazebo-yarp-plugins #525: gazebo_imu: ADDITIONAL_WRAPPER group should be not mandatory

Nicogene, 2020-12-07:
The device `gazebo_imu` can be opened and attached to the "old" wrapper (see http://yarp.it/git-master/classServerInertial.html) and to the "new" one (see http://yarp.it/git-master/classMultipleAnalogSensorsServer.html) with the `WRAPPER` and `ADDITIONAL_WRAPPER` configuration logic.

Unfortunately, both wrappers are mandatory in the configuration phase, since not specifying the `ADDITIONAL_WRAPPER` you get this error:
```
[ERROR] GazeboYarpIMU : [ADDITIONAL_WRAPPER] group not found in config file
```

I think it could be better if this would be an optional parameter

traversaro, 2021-02-11:
Related to https://github.com/robotology/gazebo-yarp-plugins/issues/534 . Since https://github.com/robotology/gazebo-yarp-plugins/pull/532, no network wrapper should be mandatory.

traversaro, 2022-06-19:
Fixed by https://github.com/robotology/gazebo-yarp-plugins/pull/586 .

closed 2022-06-19 by traversaro