robotology/gazebo-yarp-plugins #526: Gazebo stucks when YARP_CLOCK is set

randaz81, 2021-01-11:
If I set the env variable:
`YARP_CLOCK=/clock`
and the I start gazebo with:
`gazebo -s libgazebo_yarp_clock.so`
then the system immediately freezes (the gazbeo gui does not appear).

So, currently, the only way to work with simulated time is to have all process with YARP_CLOCK=/clock, with the only exception of gazebo that must have this env variable unset.
Thus the environment variable YARP_CLOCK cannot be put into a bash_rc file.

traversaro, 2021-01-11:
Back in time (2015/2016) this used not to be the case, but at some point it started happening, probably due to some `yarp::os::Time::delay` that is being called in some callback of the Gazebo's Physics update, that is causing a deadlock, as `yarp::os::Time::delay` will be waiting for the Gazebo time to pass to return, but the Gazebo's Physics time will not pass if the physics update does not complete because it is blocked by `yarp::os::Time::delay`. 

Is this happening in the case of empty world?

randaz81, 2021-01-12:
Yes, it freezes just with: 
`gazebo -s libgazebo_yarp_clock.so`
So the world is empty and the only plugin loaded is the clock

traversaro, 2021-01-12:
Then it is possible that there is something going on with the Network constructors or the checkNetwork method?

drdanz, 2021-01-13:
I don't remember changing anything...
A bisect is probably the best way to find where this issue was introduced

traversaro, 2021-01-13:
> 
> 
> I don't remember changing anything...
> A bisect is probably the best way to find where this issue was introduced

I am afraid that bisecting will not help as gazebo-yarp-plugins will not compile against of a YARP older then 2018/2019 . Probably it is easier to check where the program is blocked after is launched via `gazebo -s libgazebo_yarp_clock.so` .

xEnVrE, 2021-01-13:
To add some context, when launching Gazebo with that configuration on the terminal one reads

```
[INFO] |yarp.os.Port| Port /iiticublap202/gzserver/44642/clock:i active at tcp://192.168.1.5:10005/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[ERROR] |yarp.os.NetworkClock| Cannot find time port "/clock" or a time topic "/clock@"
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
(...)
```

It seems that this is happening in 

https://github.com/robotology/yarp/blob/7e0f54d7bb4621739b8a4dc43025f5c2c87d179c/src/libYARP_os/src/yarp/os/Time.cpp#L223

and is part of the initialization of the object `yarp::os::Network` in

https://github.com/robotology/gazebo-yarp-plugins/blob/6936859f7249d234e895a97556c884159cced643/plugins/clock/src/Clock.cc#L71

traversaro, 2021-01-13:
Thanks @xEnVrE , now I remember. Probably the regression was caused by https://github.com/robotology/yarp/pull/1277, that introduced the early clock initialization.

traversaro, 2021-01-13:
A possible solution is to switch to use the system clock after yarp::sig::Network was initialized, and switch back to use the default clock once `/clock`  has been created by calling `yarp::os::NetworkBase::yarpClockInit(YARP_CLOCK_DEFAULT)`.

traversaro, 2021-02-11:
A fix for this issue is provided in https://github.com/robotology/gazebo-yarp-plugins/pull/537 .

closed 2021-02-15 by traversaro