robotology/gazebo-yarp-plugins #534: Permit to launch gazebo_yarp_plugins without creating any network wrapper

traversaro, 2021-02-11:
After https://github.com/robotology/gazebo-yarp-plugins/pull/532 is merged, we now have a `gazebo_yarp_robotinterface` plugin, that can be used to instantiate any YARP device that are not directly instantiated by model or sensor plugins, include network server for any middleware. For this reason, it is now a legitimate use case to configure a gazebo yarp plugin to just create one single YARP device, without the corresponding wrapper, so all the gazebo_yarp_plugins should support this.

Definition of done: all the plugins can be launched without any wrapper.

Nicogene, 2021-11-19:
Thanks to https://github.com/robotology/gazebo-yarp-plugins/pull/584 and https://github.com/robotology/gazebo-yarp-plugins/pull/583 now it is possible to run `gazebo_imu` and `gazebo_forcetorque` without wrappers

traversaro, 2022-03-31:
Fixed by: 
* https://github.com/robotology/gazebo-yarp-plugins/pull/583
* https://github.com/robotology/gazebo-yarp-plugins/pull/584
* https://github.com/robotology/gazebo-yarp-plugins/pull/586

closed 2022-03-31 by traversaro