robotology/gazebo-yarp-plugins #535: Ensure that all device created by plugins can specify their YARP device instance name to be used by gazebo_yarp_robotinterface

traversaro, 2021-02-11:
https://github.com/robotology/gazebo-yarp-plugins/pull/532 added a `gazebo_yarp_robotinterface`. However, for YARP devices to be available to be connected to yarp device instantiated by the robotinterface plugin, it is necessary that the plugin support specifying the "YARP device instance name" of the created YARP devices.

The plugin that support doing that are listed in https://github.com/robotology/gazebo-yarp-plugins/pull/532/files#diff-016af8ed33c29358106ec0e48301e53602fc63ce5be3b985e45c73675367e49eR70, but this at the moment does not include useful plugin such as `gazebo_yarp_imu` or `gazebo_imu_forcetorque`.

Definition of done: all the plugin can specify the the YARP device instance name of the YARP device that they create, so that this device can be attached to devices instantiate by `gazebo_yarp_robotinterface` .

Nicogene, 2021-11-19:
Thanks to https://github.com/robotology/gazebo-yarp-plugins/pull/584 and https://github.com/robotology/gazebo-yarp-plugins/pull/583 the support has been extended also to `gazebo_imu` and `gazebo_forcetorque`

traversaro, 2021-11-19:
Thanks @Nicogene and @GrmanRodriguez ! Probably it is a bit helpful to have a recap on the devices migrated and the one still to migrate, I have added one to the top comment.