robotology/gazebo-yarp-plugins #545: If I delete a model and I reinsert it, Gazebo hangs indefinitely

S-Dafarra, 2021-03-25:
I am using Gazebo 11.3 with the latest master of this repo. Whenever I load a model and I delete it, then the gazebo real-time factor seems stuck and I cannot add any model.

I am not sure if it is a problem of the plugins, the models, or Gazebo itself.

traversaro, 2021-03-25:
With which model are you experiencing this?

S-Dafarra, 2021-03-25:
> With which model are you experiencing this? On which OS?

On Ubuntu 20.04. Actually, I just run some more tests, and this seems to happen only with the ``iCubGazeboV3`` model.

S-Dafarra, 2021-03-25:
By looking at the terminal output, I also noticed the following error:
```
Node::Advertise(): Error advertising topic [/iCub/joint_cmd]. Did you forget to start the discovery service?
```

traversaro, 2021-03-25:
> 
> 
> By looking at the terminal output, I also noticed the following error:
> 
> ```
> Node::Advertise(): Error advertising topic [/iCub/joint_cmd]. Did you forget to start the discovery service?
> ```

Can you try to run the command `gz topic -l` to see the opened Gazebo topics for both iCub 2.5 and 3? Perhaps we can spot some differences that could point us in the right direction.

S-Dafarra, 2021-03-25:
> Can you try to run the command `gz topic -l` to see the opened Gazebo topics for both iCub 2.5 and 3? Perhaps we can spot some differences that could point us in the right direction.

I inserted ``iCubGazevoV2_5 (no hands)`` and obtained the following topics
```
sdafarra@iiticublap104:~$ gz topic -l
/gazebo/default/atmosphere
/gazebo/default/diagnostics
/gazebo/default/factory
/gazebo/default/factory/light
/gazebo/default/gui
/gazebo/default/gzclient_camera/cmd
/gazebo/default/iCub/chest/chest_ems_acc_eb1/imu
/gazebo/default/iCub/chest/chest_ems_acc_eb2/imu
/gazebo/default/iCub/chest/chest_ems_acc_eb3/imu
/gazebo/default/iCub/chest/chest_ems_acc_eb4/imu
/gazebo/default/iCub/chest/chest_ems_gyro_eb1/imu
/gazebo/default/iCub/chest/chest_ems_gyro_eb2/imu
/gazebo/default/iCub/chest/chest_ems_gyro_eb3/imu
/gazebo/default/iCub/chest/chest_ems_gyro_eb4/imu
/gazebo/default/iCub/chest/chest_mtb_acc_0b10/imu
/gazebo/default/iCub/chest/chest_mtb_acc_0b7/imu
/gazebo/default/iCub/chest/chest_mtb_acc_0b8/imu
/gazebo/default/iCub/chest/chest_mtb_acc_0b9/imu
/gazebo/default/iCub/chest/wrench
/gazebo/default/iCub/head/head_imu_acc_1x1/imu
/gazebo/default/iCub/head/wrench
/gazebo/default/iCub/l_ankle_1/wrench
/gazebo/default/iCub/l_ankle_2/l_foot_ft_acc_3b13/imu
/gazebo/default/iCub/l_ankle_2/l_foot_ft_gyro_3b13/imu
/gazebo/default/iCub/l_ankle_2/wrench
/gazebo/default/iCub/l_arm_ft_sensor/l_arm_ft_sensor/wrench
/gazebo/default/iCub/l_elbow_1/wrench
/gazebo/default/iCub/l_foot/l_foot_mtb_acc_10b12/imu
/gazebo/default/iCub/l_foot/l_foot_mtb_acc_10b13/imu
/gazebo/default/iCub/l_foot/wrench
/gazebo/default/iCub/l_foot_ft_sensor/l_foot_ft_sensor/wrench
/gazebo/default/iCub/l_forearm/l_forearm_mtb_acc_1b7/imu
/gazebo/default/iCub/l_forearm/l_forearm_mtb_acc_1b8/imu
/gazebo/default/iCub/l_forearm/l_forearm_mtb_acc_1b9/imu
/gazebo/default/iCub/l_forearm/wrench
/gazebo/default/iCub/l_hand/wrench
/gazebo/default/iCub/l_hip_1/wrench
/gazebo/default/iCub/l_hip_2/l_upper_leg_ft_acc_3b12/imu
/gazebo/default/iCub/l_hip_2/l_upper_leg_ft_gyro_3b12/imu
/gazebo/default/iCub/l_hip_2/wrench
/gazebo/default/iCub/l_hip_3/wrench
/gazebo/default/iCub/l_leg_ft_sensor/l_leg_ft_sensor/wrench
/gazebo/default/iCub/l_lower_leg/l_lower_leg_ems_acc_eb7/imu
/gazebo/default/iCub/l_lower_leg/l_lower_leg_ems_gyro_eb7/imu
/gazebo/default/iCub/l_lower_leg/l_lower_leg_mtb_acc_10b10/imu
/gazebo/default/iCub/l_lower_leg/l_lower_leg_mtb_acc_10b11/imu
/gazebo/default/iCub/l_lower_leg/l_lower_leg_mtb_acc_10b8/imu
/gazebo/default/iCub/l_lower_leg/l_lower_leg_mtb_acc_10b9/imu
/gazebo/default/iCub/l_lower_leg/wrench
/gazebo/default/iCub/l_shoulder_1/wrench
/gazebo/default/iCub/l_shoulder_2/wrench
/gazebo/default/iCub/l_shoulder_3/wrench
/gazebo/default/iCub/l_upper_arm/l_upper_arm_mtb_acc_1b10/imu
/gazebo/default/iCub/l_upper_arm/l_upper_arm_mtb_acc_1b11/imu
/gazebo/default/iCub/l_upper_arm/l_upper_arm_mtb_acc_1b12/imu
/gazebo/default/iCub/l_upper_arm/l_upper_arm_mtb_acc_1b13/imu
/gazebo/default/iCub/l_upper_arm/wrench
/gazebo/default/iCub/l_upper_leg/l_upper_leg_ems_acc_eb10/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_ems_acc_eb6/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_ems_gyro_eb10/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_ems_gyro_eb6/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_mtb_acc_10b1/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_mtb_acc_10b2/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_mtb_acc_10b3/imu
/gazebo/default/iCub/l_upper_leg/l_upper_leg_mtb_acc_10b4/imu
[... 317 lines cut ...]
Testing /icubSim/right_arm/state:o at tcp://192.168.1.84:10049/
* No response, removing port /icubSim/right_arm/state:o
Testing /icubSim/right_arm/stateExt:o at tcp://192.168.1.84:10050/
* No response, removing port /icubSim/right_arm/stateExt:o
Testing /icubSim/right_leg/command:i at tcp://192.168.1.84:10036/
* No response, removing port /icubSim/right_leg/command:i
Testing /icubSim/right_leg/rpc:i at tcp://192.168.1.84:10035/
* No response, removing port /icubSim/right_leg/rpc:i
Testing /icubSim/right_leg/state:o at tcp://192.168.1.84:10037/
* No response, removing port /icubSim/right_leg/state:o
Testing /icubSim/right_leg/stateExt:o at tcp://192.168.1.84:10038/
* No response, removing port /icubSim/right_leg/stateExt:o
Testing /icubSim/torso/command:i at tcp://192.168.1.84:10028/
* No response, removing port /icubSim/torso/command:i
Testing /icubSim/torso/rpc:i at tcp://192.168.1.84:10027/
* No response, removing port /icubSim/torso/rpc:i
Testing /icubSim/torso/state:o at tcp://192.168.1.84:10029/
* No response, removing port /icubSim/torso/state:o
Testing /icubSim/torso/stateExt:o at tcp://192.168.1.84:10030/
* No response, removing port /icubSim/torso/stateExt:o
Ignoring /slocale
Ignoring fallback
Giving name server a chance to do garbage collection.
Name server says: garbage collection done.
```

traversaro, 2021-03-25:
> 
> 
> By looking at the terminal output, I also noticed the following error:
> 
> ```
> Node::Advertise(): Error advertising topic [/iCub/joint_cmd]. Did you forget to start the discovery service?
> ```

To clarify, this is happening when Gazebo hangs indefinitely ?

S-Dafarra, 2021-03-25:
> > By looking at the terminal output, I also noticed the following error:
> > ```
> > Node::Advertise(): Error advertising topic [/iCub/joint_cmd]. Did you forget to start the discovery service?
> > ```
> 
> To clarify, this is happening when Gazebo hangs indefinitely ?

I payed more attention, and that happened when I clicked on it before deleting it. 

I am not sure it is related, but I just noticed that when I click on the ``iCubGazeboV3``model, a weird body appears on the left arm
[image]
This disappears as soon as I click in the gazebo world and insert it.

traversaro, 2021-03-25:
I searched for "Did you forget to start the discovery service?" in the Gazebo codebase, and nothing showed up, so I realized that I was barking to the wrong tree. That error is printed by Ignition Transport (see https://github.com/ignitionrobotics/ign-transport/search?q=Did+you+forget+to+start+the+discovery+service%3F). So @S-Dafarra , can you to a check similar to the one done via `gz topic -l`  by using `ign topic list` ?

S-Dafarra, 2021-03-25:
Gazebo starts:
```
sdafarra@iiticublap104:~$ ign topic --list
/iCub/joint_cmd
/introspection/eshozp/items_update
```
Insert ``iCubGazeboV2_5``
```
sdafarra@iiticublap104:~$ ign topic --list
/introspection/scgilr/items_update
sdafarra@iiticublap104:~$ 
```
Touch ``iCubGazeboV2_5``
```
sdafarra@iiticublap104:~$ ign topic --list
/iCub/joint_cmd
/introspection/scgilr/items_update
```
Delete ``iCubGazeboV2_5``
```
sdafarra@iiticublap104:~$ ign topic --list
/iCub/joint_cmd
/introspection/scgilr/items_update
```
Insert ``iCubGazeboV3``
```
sdafarra@iiticublap104:~$ ign topic --list
/default/iCub/chest/realsense_chest_depth/image
/default/iCub/chest/realsense_chest_rgb/image
/iCub/joint_cmd
/introspection/scgilr/items_update
```
Touched ``iCubGazeboV3``, appears the error above
```
sdafarra@iiticublap104:~$ ign topic --list
/default/iCub/chest/realsense_chest_depth/image
/default/iCub/chest/realsense_chest_rgb/image
/introspection/scgilr/items_update
```
Deleted ``iCubGazeboV3``
```
sdafarra@iiticublap104:~$ ign topic --list
/default/iCub/chest/realsense_chest_depth/image
/default/iCub/chest/realsense_chest_rgb/image
/introspection/scgilr/items_update
```

Gazebo is not stuck, but I cannot add any other model.

S-Dafarra, 2021-04-30:
I just noticed by looking at the Gazebo terminal, that when deleting the model, the ``ServerInertial`` is closed twice:
```
[INFO] |yarp.devices.ServerInertial| Closing Server Inertial...
[INFO] |yarp.devices.ServerInertial| Server Intertial thread finished
[INFO] |yarp.devices.ServerInertial| Closing Server Inertial...
[INFO] |yarp.devices.ServerInertial| Server Intertial thread finished
```
while with ``iCubGazeboV2_5``, that message is printed only once.

S-Dafarra, 2021-04-30:
>while with iCubGazeboV2_5, that message is printed only once.

This is because ``iCubGazeboV3`` has two IMUS, while ``iCubGazeboV2_5`` only one.

Actually, I just tried again, and the problem got solved. At this point, I think it either got solved on the Gazebo side (I am using version 11.5 now), or maybe some other changes in my system. I think we can close this.

closed 2021-04-30 by S-Dafarra

traversaro, 2021-04-30:
> At this point, I think it either got solved on the Gazebo side (I am using version 11.5 now), or maybe some other changes in my system. I think we can close this.

It could be related to https://github.com/osrf/gazebo/pull/2950 , but it is just a guess.