robotology/gazebo-yarp-plugins #546: Sometimes the iCubGazeboV3 model fails to load because of the RGBD sensor wrapper

S-Dafarra, 2021-03-26:
I just tried to insert the ``iCubGazeboV3`` model in Gazebo, but it fails to load with the error
```
[INFO] |yarp.devices.RGBDSensorWrapper| RGBDSensorWrapper: ROS configuration parameters are not set, skipping ROS topic initialization.
[INFO] |yarp.os.Port| Port /icubSim/depthCamera/rpc:i active at tcp://10.240.9.61:10005/
[INFO] |yarp.os.Port| Port /icubSim/depthCamera/rgbImage:o active at tcp://10.240.9.61:10006/
[INFO] |yarp.os.Port| Port /icubSim/depthCamera/depthImage:o active at tcp://10.240.9.61:10007/
[INFO] |yarp.dev.PolyDriver| Created wrapper <RGBDSensorWrapper>. See C++ class RGBDSensorWrapper for documentation.
[DEBUG] CC: m_driverParameters:	 (CAMERA_PARAM (focalLengthX 570.34222412109375) (focalLengthY 570.34222412109375) (tangentialPointX 0.0) (tangentialPointY 0.0)) (device gazebo_depthCamera) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName realsense_chest_depth) (name "/icubSim/depthCamera") (period 30) (sensorScopedName "default::iCub::chest::realsense_chest_depth")
[INFO] |yarp.dev.PolyDriver| Created device <gazebo_depthCamera>. See C++ class GazeboYarpDepthCameraDriver for documentation.
[INFO] |yarp.devices.RGBDSensorWrapper| RGBDSensorWrapper: Bad name!
```
Also, now Gazevo is stuck
[image]

This does not happen every time, it seems like a racing condition. It seems to happen more often when I launch Gazebo with the ``YARP_CLOCK`` env variable set.

S-Dafarra, 2021-03-26:
Here the result of ``gz topic -l``

```
sdafarra@iiticublap104:~$ gz topic -l
/gazebo/default/atmosphere
/gazebo/default/default/realsense_chest_depth/cmd
/gazebo/default/diagnostics
/gazebo/default/factory
/gazebo/default/factory/light
/gazebo/default/gui
/gazebo/default/gzclient_camera/cmd
/gazebo/default/iCub/chest/realsense_chest_depth/image
/gazebo/default/iCub/chest/realsense_chest_rgb/image
/gazebo/default/iCub/l_ankle_1/wrench
/gazebo/default/iCub/l_ankle_2/wrench
/gazebo/default/iCub/l_foot_front/wrench
/gazebo/default/iCub/l_foot_rear/wrench
/gazebo/default/iCub/l_hip_1/wrench
/gazebo/default/iCub/l_hip_2/wrench
/gazebo/default/iCub/l_hip_3/wrench
/gazebo/default/iCub/l_lower_leg/wrench
/gazebo/default/iCub/l_upper_leg/wrench
/gazebo/default/iCub/r_ankle_1/wrench
/gazebo/default/iCub/r_ankle_2/wrench
/gazebo/default/iCub/r_foot_front/wrench
/gazebo/default/iCub/r_foot_rear/wrench
/gazebo/default/iCub/r_hip_1/wrench
/gazebo/default/iCub/r_hip_2/wrench
/gazebo/default/iCub/r_hip_3/wrench
/gazebo/default/iCub/r_lower_leg/wrench
/gazebo/default/iCub/r_upper_leg/wrench
/gazebo/default/iCub/root_link/wrench
/gazebo/default/iCub/torso_1/wrench
/gazebo/default/iCub/torso_2/wrench
/gazebo/default/joint
/gazebo/default/light/modify
/gazebo/default/log/control
/gazebo/default/log/status
/gazebo/default/model/info
/gazebo/default/model/modify
/gazebo/default/physics
/gazebo/default/physics/contacts
/gazebo/default/playback_control
/gazebo/default/pose/info
/gazebo/default/pose/local/info
/gazebo/default/pose/modify
/gazebo/default/request
/gazebo/default/response
/gazebo/default/roads
/gazebo/default/scene
/gazebo/default/selection
/gazebo/default/sensor
/gazebo/default/skeleton_pose/info
/gazebo/default/sky
/gazebo/default/undo_redo
/gazebo/default/user_camera/joy_pose
/gazebo/default/user_camera/joy_twist
/gazebo/default/user_camera/pose
/gazebo/default/user_cmd
/gazebo/default/user_cmd_stats
/gazebo/default/visual
/gazebo/default/wind
/gazebo/default/world_control
/gazebo/default/world_stats
/gazebo/server/control
/gazebo/world/modify
```

S-Dafarra, 2021-03-26:
The error comes fromhttps://github.com/robotology/yarp/blob/9b52a71e176e23ebe42956cf5b23d8ae40b255f0/src/devices/RGBDSensorWrapper/RGBDSensorWrapper.cpp#L525. I printed ``device2attach[0]->key`` and that is ``dummy``.

traversaro, 2021-03-26:
> 
> 
> The error comes from https://github.com/robotology/yarp/blob/9b52a71e176e23ebe42956cf5b23d8ae40b255f0/src/devices/RGBDSensorWrapper/RGBDSensorWrapper.cpp#L525. I printed `device2attach[0]->key` and that is `dummy`.

To clarify, that message is printed everytime, or only when Gazebo is stuck?

S-Dafarra, 2021-03-26:
> > The error comes from https://github.com/robotology/yarp/blob/9b52a71e176e23ebe42956cf5b23d8ae40b255f0/src/devices/RGBDSensorWrapper/RGBDSensorWrapper.cpp#L525. I printed `device2attach[0]->key` and that is `dummy`.
> 
> To clarify, that message is printed everytime, or only when Gazebo is stuck?

Actually, I did not notice it before in the flow of messages, but it is printed all the times. Incidentally, this is always the last message when Gazebo gets stuck.