robotology/gazebo-yarp-plugins #549: can't install the gazebo-yarp plugin

patrickshas, 2021-04-23:
I am trying to install the plugin on a fresh install of ubuntu 20.04 with working ROS, gazebo, yarp and iCub main with no luck.
I am following the instillation manual and can't get past the:
cd build
cmake -DCMAKE_BUILD_TYPE="Release "../ -DCMAKE_INSTALL_PREFIX=/path/to/the/install/folder
cmake --config Release --build . --target install

I replaced the middle line with: 
cmake -DCMAKE_BUILD_TYPE="Release" .. -DCMAKE_INSTALL_PREFIX=/usr/local/src/robot/iCubContrib/ICUBcontrib_x86-64

Here is a screenshot of my console:

[image]

Kindly please tell me what I am doing wrong.
Thanks in advance

traversaro, 2021-04-23:
Where did you get the command:
~~~
cmake --config Release --build . --target install
~~~
from the docs?

In my case this is working fine:
~~~
cmake --build . --config Release --target install
~~~

traversaro, 2021-04-23:
(Basically, `--build .` should be the first argument of the cmake).

patrickshas, 2021-04-23:
thanks for this.
That command ( cmake --build . --config Release --target install) commences the build but only to 81%

[image]

traversaro, 2021-04-26:
In any case, there is any specific reason why you want to install gazebo-yarp-plugins repo on its own? If you are installing it for iCub use, following the instructions in https://icub-tech-iit.github.io/documentation/sw_installation/ you may find easier way of installing it.

traversaro, 2021-05-31:
Hi @patrickshas, were you able to solve your problems? If so, feel free to comment and close the issue, thanks!

closed 2021-06-07 by traversaro