robotology/gazebo-yarp-plugins #553: Compilation failing on MacOS 11.3

lrapetti, 2021-06-09:
I am getting the following error
```
Scanning dependencies of target gazebo_yarp_singleton
[  1%] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o
[  2%] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o
make[6]: *** No rule to make target `/usr/lib/libm.dylib', needed by `lib/libgazebo_yarp_singleton.dylib'.  Stop.
```

I am on `MacOS 11.3.1` and I have installed `gazebo11` via homebrew.

I have found a similar issue https://github.com/PX4/PX4-SITL_gazebo/issues/679, I have tried to reinstall the command-line tools as suggested there but it didn't work out. I am trying to investigate further if it is a problem of my local configuration

traversaro, 2021-06-09:
Probably something somewhere is doing `target_link_libraries(<> PUBLIC m)` because on Linux they want to link the C math library (https://stackoverflow.com/questions/34625627/how-to-link-to-the-c-math-library-with-cmake), but apparently this is not needed/not present on macOS 11.3. Thanks for the reporting the issue.

lrapetti, 2021-06-09:
> Probably something somewhere is doing `target_link_libraries(<> PUBLIC m)` because on Linux they want to link the C math library [...]

Thanks for the hint. I tried to search for it, but I didn't find it being explicitly linked. I have also searched among the [linked libraries of the singleton library](https://github.com/robotology/gazebo-yarp-plugins/blob/master/libraries/singleton/CMakeLists.txt#L17) (that is the one that seems to be failing), but I didn't find it either

traversaro, 2021-06-09:
> I have tried to reinstall the command-line tools as suggested there but it didn't work out. I am trying to investigate further if it is a problem of my local configuration

After installing the command-line tools did you recompiled from a clean build gazebo-yarp-plugins (see https://github.com/PX4/PX4-SITL_gazebo/issues/679#issuecomment-779406983). When it is the last time you updated the Homebrew packages?

traversaro, 2021-07-01:
@lrapetti any update on this issue?

lrapetti, 2021-07-06:
> @lrapetti any update on this issue?

Sorry for the delay. I found out that I still had gazebo8 on my system. I have uninstalled it and reinstalled gazebo11. Everything compiled fine, and the plugins seem to be working.

I guess we can close this issue.

closed 2021-07-06 by lrapetti