robotology/gazebo-yarp-plugins #585: Provide a way for plugins to explicitly disable implicitly created network wrappers server

traversaro, 2021-11-19:
See https://github.com/robotology/gazebo-yarp-plugins/issues/569 for a recap on this in situation. In a nutshell, now we have the `GAZEBO_YARP_PLUGINS_DISABLE_IMPLICIT_NETWORK_WRAPPERS` option that just disable all implicitly created network wrappers server, as we would expect that `gazebo-yarp-plugins` will work in the future. However users that are migrating to the new `nws`/`nwc` system, are actually requiring the `GAZEBO_YARP_PLUGINS_DISABLE_IMPLICIT_NETWORK_WRAPPERS` option to be `ON`, as they do not want to launch at all the implicit network wrappers.

However, for various reason having **models** that use `gazebo-yarp-plugins` compiled with a non-standard flag enabled is problematic for a various of reason:
* First of all, it prevents people to use the same gazebo-yarp-plugins to simulate both models that still needs to be migrated to use nws/nwc structure and models that already use the new nws/nwc structure but required `GAZEBO_YARP_PLUGINS_DISABLE_IMPLICIT_NETWORK_WRAPPERS` to be ON
* Secondly, it prevents users that get `gazebo-yarp-plugins` via binaries (for example conda binaries) to simulate the new models

To avoid this situation, I guess the easier way to avoid this problem is to provide users a way to explicity disable at runtime the instantiation of implicity created network wrappers servers, for example a `noImplicitNetworkWrapperServer` option. This option will be ignored if `GAZEBO_YARP_PLUGINS_DISABLE_IMPLICIT_NETWORK_WRAPPERS` option is `ON` and when it will be `ON` by default, but it will help a lot in this transition period to simulate with the same `gazebo-yarp-plugins` both new and old models.

Nicogene, 2021-11-22:
The thing that confused me a little bit about `GAZEBO_YARP_PLUGINS_DISABLE_IMPLICIT_NETWORK_WRAPPERS` is that it should actually _disable implicit network wrappers_, like for example: 

https://github.com/robotology/gazebo-yarp-plugins/blob/880af12278ac8718ab91a247cfae898f94ecbd32/plugins/forcetorque/src/ForceTorque.cc#L57-L63

But it is doing more than that, it disable also the **explicit** wrapper instantiation from ini files see:

https://github.com/robotology/gazebo-yarp-plugins/blob/880af12278ac8718ab91a247cfae898f94ecbd32/plugins/controlboard/src/ControlBoard.cc#L103-L119

https://github.com/robotology/gazebo-yarp-plugins/blob/880af12278ac8718ab91a247cfae898f94ecbd32/plugins/depthCamera/src/DepthCamera.cc#L92-L108

https://github.com/robotology/gazebo-yarp-plugins/blob/880af12278ac8718ab91a247cfae898f94ecbd32/plugins/imu/src/IMU.cc#L86-L130

https://github.com/robotology/gazebo-yarp-plugins/blob/880af12278ac8718ab91a247cfae898f94ecbd32/plugins/forcetorque/src/ForceTorque.cc#L84-L92

I would re-enable the open of the wrappers, also when the implicit wrapper is disabled

traversaro, 2021-11-22:
I think this is the source of the confusion. What you are calling **explicit** network wrapper is what we are calling (me and @ste93) the **implicit** network wrapper (we are calling it implicit, as it is always create whenever the Gazebo plugin is created). However, we can clarify ourself on teams.

> I would re-enable the open of the wrappers, also when the implicit wrapper is disabled

But in this case we would not solve the problem listed in https://github.com/robotology/gazebo-yarp-plugins/issues/585, if I am not wrong.

Nicogene, 2021-11-22:
Yes what confuse me is the term "implicit" but for sure we can continue F2F or T2T.

> If I am not wrong, what you are calling implicit created device are only created if you are using the command line tool yarpdev, so they do not play a role here as they are never created by a Gazebo simulation.

Yes for me implicit is that, but it is just a matter of agree on the nomeclature, and maybe write it somewhere

ste93, 2021-11-22:
yep, I will be at cris wednesday if you want f2f, otherwise I'm always available on teams

traversaro, 2022-03-31:
Fixed by https://github.com/robotology/gazebo-yarp-plugins/pull/586 .

closed 2022-03-31 by traversaro