robotology/gazebo-yarp-plugins #592: Wrong data streamed by the accelerometer on the iCub3 head

isorrentino, 2021-12-03:
Here is what I see logging data received on the yarp port `/icubSim/head/inertials`.

[image]

I expect to see the gravity in the `z` direction instead of `0`.

I tried with two models: `iCubGazeboV3_fixed (no hands)` and `iCubGazeboV3 (no hands)`.

cc @prashanthr05 @traversaro

traversaro, 2021-12-03:
Can you post a few lines of the dump of `/icubSim/head/inertials` ?

isorrentino, 2021-12-03:
I dumped data by using the `YarpRobotLoggerDevice` in `BipedalLocomotionFramework` (https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/devices/YarpRobotLoggerDevice).
Here is the .mat file:
[yarprobot_dataset_2021_12_03_17_34_55.zip](https://github.com/robotology/gazebo-yarp-plugins/files/7651363/yarprobot_dataset_2021_12_03_17_34_55.zip)

The `mas remapper` is defined in https://github.com/ami-iit/bipedal-locomotion-framework/blob/master/devices/YarpRobotLoggerDevice/app/robots/iCubGazeboV3/interface/mas-remapper.xml and the `multipleanalogsensorsclient` is https://github.com/ami-iit/bipedal-locomotion-framework/blob/master/devices/YarpRobotLoggerDevice/app/robots/iCubGazeboV3/interface/head_imu.xml

traversaro, 2021-12-03:
Can you read/dump at least a few lines of the port that starts with `/icubSim/head/inertials` with `yarp read`?  So that we can at least understand at which point of the stack the problem is.

isorrentino, 2021-12-03:
Here is a dump of `yarp read ... /icubSim/head/inertials/measures:o`

[port.txt](https://github.com/robotology/gazebo-yarp-plugins/files/7652543/port.txt)

traversaro, 2022-05-03:
fyi @gabrielenava that also reported strage results from a `gazebo_yarp_imu` plugin recently.

gabrielenava, 2022-05-03:
@FabioBergonti suggested me to have a look at https://github.com/ami-iit/element_floating-base-estimation/issues/61#issuecomment-651695040

I am investigating the comment to understand if this is the problem I am facing

traversaro, 2022-05-03:
I have the impression that is related to orientation, while the problem here is on accelerometer and in your case is on angular velocity, but the problem can be related in an indirected way I guess.

prashanthr05, 2022-05-09:
The problem reported in the comment https://github.com/robotology/gazebo-yarp-plugins/issues/592#issue-1070771144 is due to a corner-case unhandled in  `YarpSensorBridge` while dealing with multiple analog sensors. 

Apparently, `YarpSensorBridge` use a single unordered_map [std::unordered_map<std::string, StampedYARPVector> wholeBodyInertialMeasures;](https://github.com/ami-iit/bipedal-locomotion-framework/blob/fa7f881dac170d3cee36cd82048aab08d22bbc16/src/RobotInterface/YarpImplementation/include/BipedalLocomotion/RobotInterface/YarpSensorBridgeImpl.h#L161) to store different sensor measures with unique names which is used when an user calls `getLinearAcceleromterMeasurement()`, `getGyroscopeMeasurement()` [here](https://github.com/ami-iit/bipedal-locomotion-framework/blob/fa7f881dac170d3cee36cd82048aab08d22bbc16/src/RobotInterface/YarpImplementation/src/YarpSensorBridge.cpp#L464-L534).

This problem occurs when we specify the same ids for different sensors as specified [here](https://github.com/ami-iit/bipedal-locomotion-framework/blob/fa7f881dac170d3cee36cd82048aab08d22bbc16/devices/YarpRobotLoggerDevice/app/robots/iCubGazeboV3/interface/mas-remapper.xml#L8-L16) which is the case only for simulation.

This should be fixed in `YarpSensorBridge` side by handling separate buffers for each sensor type in MAS inertials.

prashanthr05, 2022-05-09:
Nevertheless, what I see is the accelerometer measurements streamed by `gazebo_imu` do not seem to be rotated in the sensor frame, instead it's in world frame.

For example, the below plot is from the accelerometer on the root link of `iCubGazeboV2_5` (Xsens IMU), in which the IMU z-axis is tilted and pointing downwards, ideally on the robot we observe in stationary condition the value `[0.09; -5.72; -7.98]`, here it doesn't seem to be the case. 

[image]

isorrentino, 2024-01-17:
This problem has been fixed. Closing.

closed 2024-01-17 by isorrentino