robotology/gazebo-yarp-plugins #635: Problem with right wrist of iCub3 with torque control?

Woolfrey, 2022-09-06:
I've been having some issues when controlling the iCub3 in torque mode in Gazebo. I have a QP solver that I use to solve the joint accelerations/torques for a Cartesian impedance controller. When moving the hands to certain poses it frequently throws an error about constraint violations for the `r_wrist_yaw` joint.

Every loop in the control I compute the minimum and maximum joint accelerations allowed to avoid hitting joint limits (both position and velocity).  The screen grab below shows the joint state when the failure occurred:
- Minimum position, current position, maximum position
- Minimum velocity, current velocity, maximum velocity
- Minimum and maximum bounds on the joint acceleration (which is fed in to the QP solver):

[image]

`r_wrist_yaw` is on the lower position limit, but its speed is **below** the limit. As a consequence the lower acceleration bound of 400 is larger than the upper acceleration bound of 375. It shouldn't be possible for the velocity to be outside its limits since the QP solver should always return a feasible solution for the acceleration.

When reading the encoders, I can override the joint state to avoid this, but it still causes some instability which affects the whole body.

To anticipate some questions:

- This has **never** happened for the `l_wrist_yaw` joint, even when the hands are performing symmetric actions.
- I have never seen this behavior with other joints, even when deliberately driving them to the limits.
- I don't recall having this problem when running the robot in velocity control mode.
- I don't recall seeing this kind of behavior when using the iCub2 model (although I never used torque control mode for iCub2...).

Woolfrey, 2022-09-12:
Hi @traversaro, I'm a little busy trying to get a paper done for ICRA, but I will get back to you with information as soon as I can.

traversaro, 2022-09-12:
Sure, and good luck for the ICRA submission!

closed 2022-10-13 by traversaro

Woolfrey, 2022-10-19:
@traversaro, I think I've discovered another clue to the problem. I recently upgraded my home laptop to Ubuntu 22.04, and Gazebo simulations have been running extremely slow (real time factor can drop below 0.1). In these situations, the joint state I'm getting are frequently outside the joint limits, which causes my controller to throw a fit.

Conversely, I have rarely observed this behaviour with my work laptop (real time factor is usually above 0.5).

traversaro, 2022-10-19:
Interesting. What is the OS on your work laptop? Did you installed gazebo with apt or conda-forge?

Woolfrey, 2022-10-19:
Work laptop is running Ubuntu 20.04, but is more powerful than my personal laptop so I generally get better performance.

Personal laptop is a Surface Book 2. There are known issues dual-booting Ubuntu with Surface hardware. Ironically, upgrading to 22.04 has seemed to make things worse.

I installed Gazebo with apt.

I forgot to check which joint was outside the limits this time. It's usually the wrist joint since they seem to have limited range.