robotology/gazebo-yarp-plugins #643: CMake Error when linking `GazeboYARPPlugins` libraries

HosameldinMohamed, 2023-01-13:
When trying to build a plugin that uses some of the libraries of `gazebo-yarp-plugins`. Following http://robotology.github.io/gazebo-yarp-plugins/master/use_as_library.html

I get the following error when generating CMake:

```
 CMake Error in software/gazebo/plugins/vive-transform-plugin/CMakeLists.txt:
   Imported target "GazeboYARPPlugins::gazebo_yarp_lib_common" includes
   non-existent path

     "/home/hosameldin/code/robotology-superbuild/build/install/include/include"

   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

   * The path was deleted, renamed, or moved to another location.

   * An install or uninstall procedure did not complete successfully.

   * The installation package was faulty and references files it does not
   provide.
```

However, the build goes through with no errors.

I thought it was worth reporting it anyways.

HosameldinMohamed, 2023-01-16:
CC @traversaro

traversaro, 2023-01-16:
Indeed, there is a bug in the CMake script. If you are compiling gazebo-yarp-plugins from source, can you check if https://github.com/robotology/gazebo-yarp-plugins/pull/644 solves the problem? Thanks!

traversaro, 2023-01-16:
> I thought it was worth reporting it anyways.

Thanks!

closed 2023-01-29 by traversaro