robotology/gazebo-yarp-plugins #647: Coupling handlers for ergoCub hand MK3, 4 and 5 possibly incorrect

xEnVrE, 2023-03-17:
Historically, **as far as I know**, coupling handlers within the `controlboard` plugin were designed to operate as follows:

- the `decouple*` methods takes the state of physical joints simulated in Gazebo and reconstruct the state of the DoFs that the user can actually control YARP-side
- the `decoupleRef*` methods take the desired reference for the DoFs YARP-side and convert it to references, one for each physical joint, simulated in Gazebo

As an example, for the thumb distal joints of iCub2.x we get:

https://github.com/robotology/gazebo-yarp-plugins/blob/4b35f025c93aa51870c57e81d97fbcee273ed876/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L183-L188

i.e., the `l_thumb_distal` YARP-side is reconstructed as the sum of the two physical joints `l_hand_thumb_1_joint` and `l_hand_thumb_2_joint`, where I am using the names coming from the [iCub joints nomenclature](https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-joints/icub-joints/#left-arm) and those from the [URDF config files](https://github.com/robotology/icub-models/blob/master/iCub_manual/conf_manual/iCubGazeboV2_5_visuomanip/gazebo_icub_left_hand_thumb.ini#L4), respectively.

Then, for the second method we have:

https://github.com/robotology/gazebo-yarp-plugins/blob/4b35f025c93aa51870c57e81d97fbcee273ed876/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L210-L219

i.e.,

```console
l_thumb_0_joint = l_thumb_oppose
l_thumb_1_joint = l_thumb_proximal
l_thumb_2_joint = l_thumb_distal / 2
l_thumb_3_joint = l_thumb_distal / 2
```

Let's now have a look at the implementation of the coupling handler for the ergoCub hand MK5 (differently from iCub2.x we don't have a C++ handler for each finger, rather a single C++ handler for the whole hand).

I assume that, given the figure below - taken from  the [documentation](https://icub-tech-iit.github.io/documentation/hands/hands_mk5_coupling/#coupling-variables) - and **given my understanding** that

[image]

- the joint `q1` is what we can actually control (via q0?);
- the joint `q2` depends directly on `q1` given the leverism via the coupling law q2 = f(q1)

Given the above, a sketch of the `decouple*` method should be - here I will use names from the [URDF conf files](https://github.com/icub-tech-iit/ergocub-software/blob/8d680512237a9677482531be055b1247abaeca07/urdf/ergoCub/conf/gazebo_ergocub_left_hand.ini#L3) and from the [ergoCub joints nomenclature](https://github.com/icub-tech-iit/ergocub-software/blob/8d680512237a9677482531be055b1247abaeca07/urdf/ergoCub/conf/gazebo_ergocub_left_hand.ini#L6) and concentrate on the **thumb** joint
```console
(...)
def decouple*:
    thumb_oc = thumb_prox // as we are directly controlling only the first link of the thumb
(...)
```
While a sketch of the `decoupleRef*` method should be:
```console
(..)
def decoupleRef*:
    thumb_prox = thumb_oc // as the user provides a reference YARP-side only for the first link of the thumb
    thumb_dist = f(thumb_prox) // because q2, i.e. thumb_dist, depends on q1, i.e.thumb_prox, given the coupling law q2 = f(q1)
```

Instead, by looking at the code what we found is :

https://github.com/robotology/gazebo-yarp-plugins/blob/4b35f025c93aa51870c57e81d97fbcee273ed876/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L1327

equivalent to `thumb_oc = thumb_prox + thumb_dist`, and

https://github.com/robotology/gazebo-yarp-plugins/blob/4b35f025c93aa51870c57e81d97fbcee273ed876/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L1378-L1379

equivalent to 

```console
thumb_prox = f^{-1}(thumb_oc)` // where `f^{-1}` is the inverse of the law `f`, as per the code
thumb_dist = thumb_oc - thumb_prox // i.e. enforcing that thumb_oc  = thumb_dist + thumb_prox
```

As it can be seen, these are different from the laws that I have sketched above.

Moreover, the [configuration files of the ergoCub fingers in simulation](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/gazebo_ergocub_left_hand.ini) are reflecting the same problem.

The section of the limits should contain the limits for the actual physical joints simulated in Gazebo, however it only contains the limits for the coupled ones and the others are set to 0:

[image]

The limits for the coupled joints - unfortunately optional software-side - are missing:

[image]

Is the reasoning above correct? If not, is there anything I am missing which explains why the current implementation of the coupling handler is different from the one I expected?

Thank you and sorry for the long description - unfortunately coupling laws are always a bit painful to deal with

cc @Nicogene @mfussi66 @pattacini @traversaro

Nicogene, 2023-03-20:
cc @ale-git

ale-git, 2023-03-20:
Hi @xEnVrE, this assumption isn't correct:
 - the joint q1 is what we can actually control (via q0?);
 - the joint q2 depends directly on q1 given the leverism via the coupling law q2 = f(q1)
 
In fact, the YARP side controlled DOF is the angle of the fingertip with respect to the palm, i.e. q1+q2.
With this assumption, we see that the decoupleRef* function has to properly distribute the reference angle (the sum (q1+q2)_ref) between q1 and q2 in order to produce q1_ref and q2_ref for Gazebo, respecting the non linear mechanical coupling law.

xEnVrE, 2023-03-20:
I talked f2f with @ale-git which explained that the current implementation of the plugin is reflecting what the firmware of the hand **should have done** i.e., controlling the angle of the fingertip with respect to the palm.

However, the current firmware, for that I asked @sgiraz and @AntonioConsilvio, is implementing things differently, i.e. the user controls the angle of the first phalanx with respect to the palm. For that, I think that the implementation of the coupling handlers should be updated as I explained above. 

Can I proceed opening a PR for that? 

cc @traversaro @randaz81

xEnVrE, 2023-03-20:
> It would be great if we could also align the documentation that was added in icub-tech-iit/documentation#249, fyi @Nicogene

From my understanding, the documentation still holds. It is just how the coupling law is used in Gazebo that is making the difference at the moment.

pattacini, 2023-03-21:
>I agree on aligning the simulator with the robot, thanks a lot @xEnVrE 

I agree too.
Perhaps, it'd be worth waiting for @mfussi66 to chime in.

xEnVrE, 2023-03-21:
Meanwhile, I will add some other details / news.

- the coupling law provided in the documentation provides the **absolute** angle of the second phalanx with respect to the first one, hence extra care must be taken while using the formula in the Gazebo implementation - as Gazebo uses relative angles

- after implementing a possible new version of the Gazebo coupling and setting the correct limits, there is something wrong with respect to the real robot

| | |
| ------------- | ------------- |
|Simulation (new coupling) | [image] |
|Real (not actuated) | [image]|
| Simulation (old coupling)|  [image] |

With the new coupling the maximum range YARP-side is now correct, that is `90 deg`, while with the old one the maximum was `160` - as the user should have controlled the angle of the fingertip in that implementation.

Nonetheless, the angle that the second phalanx makes with respect to the first one is quite different from the reality. Here it is a video showing the comparison:

https://user-images.githubusercontent.com/6014499/226482063-897ebc37-4edf-4917-a97f-167eafc07693.mp4

For that I am now comparing the actual implementation of the coupling law in Gazebo with the formulae from the official documentation, to make sure that there are no misalignment.

pattacini, 2023-03-21:
> Of course I agree too, I'm getting up to speed and checking the needed modifications.

@xEnVrE has already a branch for that, I guess.

xEnVrE, 2023-03-21:
> @xEnVrE has already a branch for that, I guess.

Indeed, it can be found in https://github.com/xEnVrE/gazebo-yarp-plugins/commits/fix/mk5_coupling

However, it will not work properly if not coupled with the proper configuration file `ergocub-software`-side. At the moment I am working on a standalone **left** hand that can be found in https://github.com/xEnVrE/ergocub-software/tree/study/hand_variation/urdf/ergoCub/robots/handMK5GazeboV1.

Changes has been done directly on the URDF - instead on the simmechanics part - just because it is easier to make quick tests.

pattacini, 2023-03-21:
Thanks heaps @xEnVrE for your work 👍🏻 

@mfussi66 is at moment busy with testing the ergoCub wrist+hand we have in the PROTO lab before the delivery. We will come back to this asap then.

xEnVrE, 2023-03-22:
After discussing a bit with @ale-git in https://github.com/icub-tech-iit/documentation/issues/257, I have changed the implementation of the coupling handlers again.

Starting from the index, the match with the real finger seems much better now:

https://user-images.githubusercontent.com/6014499/227058007-edac7655-01d2-429d-b547-a7d2d960c1c3.mp4

# 

I will made comparisons with the other fingers as soon as the real hand is available again.

> It would be great if we could also align the documentation that was added in https://github.com/icub-tech-iit/documentation/pull/249

At this point, it seems to me that the documentation is to be updated according to the discussion made in https://github.com/icub-tech-iit/documentation/issues/257.

pattacini, 2023-03-23:
We're gonna organize a meeting for that asap 👍🏻

traversaro, 2023-04-21:
I guess https://github.com/icub-tech-iit/documentation/pull/267 is related, even if I am not 100% of the issue status at the moment.

xEnVrE, 2023-04-21:
> I guess https://github.com/icub-tech-iit/documentation/pull/267 is related

Yes it is @traversaro

xEnVrE, 2023-04-21:
> I am not 100% aware of the issue status at the moment

We were waiting to update the documentation before proceeding to update the coupling handlers. I will work on it starting from the next week.

traversaro, 2023-04-21:
Great, thanks!

closed 2023-05-23 by traversaro