robotology/gazebo-yarp-plugins #655: CMake Error when linking `GazeboYARPPlugins` libraries

FabioBergonti, 2023-07-25:
When trying to build a plugin that uses some of the libraries of gazebo-yarp-plugins. Following http://robotology.github.io/gazebo-yarp-plugins/master/use_as_library.html

I get the following error when generating CMake:

```
CMake Error in software/gazebo/plugins/jets-plugin/CMakeLists.txt:
  IMPORTED_IMPLIB not set for imported target  
  "GazeboYARPPlugins::gazebo_yarp_singleton" configuration "Debug". 

CMake Error in software/gazebo/plugins/jets-plugin/CMakeLists.txt: 
  IMPORTED_IMPLIB not set for imported target  
  "GazeboYARPPlugins::gazebo_yarp_singleton" configuration "Release". 
  
CMake Error in software/gazebo/plugins/jets-plugin/CMakeLists.txt: 
  IMPORTED_IMPLIB not set for imported target  
  "GazeboYARPPlugins::gazebo_yarp_singleton" configuration "MinSizeRel".  

CMake Error in software/gazebo/plugins/jets-plugin/CMakeLists.txt: 
  IMPORTED_IMPLIB not set for imported target  
  "GazeboYARPPlugins::gazebo_yarp_singleton" configuration "RelWithDebInfo".
```

I'm working with Windows and conda

cc @davidegorbani @HosameldinMohamed @traversaro @Nicogene

HosameldinMohamed, 2023-07-25:
> I'm working with Windows and conda

But even when we build `gazebo-yarp-plugin` from source in Windows, and try to link that installation, throws this error.

Nicogene, 2023-07-26:
I never used `gyp` as library, but I think it misses an header like this one:

- https://github.com/robotology/yarp/blob/master/src/libYARP_conf/src/yarp/conf/api.h.in

Do you confirm @traversaro ?

traversaro, 2023-07-26:
On most libraries we do not explicitly exports symbols, but we just set CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS option to ON, see the how-to-export-cpp-libraries repo. It is possible we are not doing this correctly in this repo.

traversaro, 2023-07-26:
Indeed, that option is set in https://github.com/robotology/gazebo-yarp-plugins/blob/master/CMakeLists.txt#L69 . Furthermore, if the singleton shared library is already working fine for plugins in the repo (i.e. all the required symbols are exported), it would be surprising if more symbols need to be exported for the use case of plugins contained in other repos.

To give more context on the issue, it is important to know that Windows and Unix-like operating systems are different on how they handle shared libraries: on Unix there is a single file (.so or .dylib) that is used both during linking (the process when libraries or executable are created from .o object files) and during loading (the process when an executable is launched and all the required shared libraries are loaded in memory. On the other hand, on Windows when you create a shared library two files should be created: an "import library" (.lib) that is used during linking, and a "dynamically loaded library (.dll) that is used during loading. In this case, it seems that for some reason the import library (implib) file for the singleton library is not correctly set in the CMake config files. Ti help us debugging, it would be helpful to report in the issue:
* The output of the installation step of gyp when build from source (to check if the .lib for the singleton library is actually installed or not)
* The content of the file named  GazeboYARPplugins-targets.cmake or similar, that shows us how the libraries are described in the CMake config files that are used when a library is found via find_package

FabioBergonti, 2023-07-28:
Hi @traversaro 
Here the information you requested

<details><summary>output of the installation step of gyp</summary>
<p>

```
MSBuild version 17.6.3+07e294721 for .NET Framework

  Checking Build System
  Building Custom Rule C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/plugins/multicamera/CMakeLists.txt
  MultiCameraPlugin.cpp
  Auto build dll exports
     Creating library C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yarp_MultiCameraPlugin.lib and object C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/ga
  zebo_yarp_MultiCameraPlugin.exp
  gazebo_yarp_MultiCameraPlugin.vcxproj -> C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\bin\Release\gazebo_yarp_MultiCameraPlugin.dll
  Building Custom Rule C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/libraries/singleton/CMakeLists.txt
  Handler.cc
  ConfHelpers.cc
  Generating Code...
  Auto build dll exports
     Creating library C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yarp_singleton.lib and object C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yar
  p_singleton.exp
  gazebo_yarp_singleton.vcxproj -> C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\bin\Release\gazebo_yarp_singleton.dll
  Building Custom Rule C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/plugins/basestate/CMakeLists.txt
  BaseState.cc
  BaseStateDriver.cpp
  Generating Code...
  Auto build dll exports
     Creating library C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yarp_basestate.lib and object C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yar
  p_basestate.exp
  gazebo_yarp_basestate.vcxproj -> C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\bin\Release\gazebo_yarp_basestate.dll
  Building Custom Rule C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/plugins/camera/CMakeLists.txt
  Camera.cc
C:/Users/fbergonti/AppData/Local/mambaforge/envs/test2_davide/Library/include\yarp/dev/FrameGrabberInterfaces.h(11): warning : "<yarp/dev/FrameGrabberInterfaces.h> file is deprecated" [C:\Users\fbergonti\Deskt
op\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
  CameraDriver.cpp
C:/Users/fbergonti/AppData/Local/mambaforge/envs/test2_davide/Library/include\yarp/dev/FrameGrabberInterfaces.h(11): warning : "<yarp/dev/FrameGrabberInterfaces.h> file is deprecated" [C:\Users\fbergonti\Deskt
op\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(76,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(77,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(78,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(79,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(80,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(81,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(82,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
precation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\build\plugins\camera\gazebo_yarp_camera.vcxproj]
C:\Users\fbergonti\Desktop\test\gazebo-yarp-plugins\plugins\camera\src\CameraDriver.cpp(83,5): warning C4996: 'sprintf': This function or variable may be unsafe. Consider using sprintf_s instead. To disable de
[... cut ...]
set_target_properties(GazeboYARPPlugins::gazebo_yarp_robotinterface PROPERTIES
  IMPORTED_LOCATION_RELEASE "${_IMPORT_PREFIX}/bin/gazebo_yarp_robotinterface.dll"
  )

list(APPEND _cmake_import_check_targets GazeboYARPPlugins::gazebo_yarp_robotinterface )
list(APPEND _cmake_import_check_files_for_GazeboYARPPlugins::gazebo_yarp_robotinterface "${_IMPORT_PREFIX}/bin/gazebo_yarp_robotinterface.dll" )

# Commands beyond this point should not need to know the version.
set(CMAKE_IMPORT_FILE_VERSION)

```

</p>
</details>

traversaro, 2023-07-28:
From the message 
     Creating library C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yarp_singleton.lib
 
It seems that the .lib library is correctly created, however to be sure you can double check that the file  C:/Users/fbergonti/Desktop/test/gazebo-yarp-plugins/build/lib/Release/gazebo_yarp_singleton.lib actually exists and it is not empty. Assuming this is the case, the problems seems that the import library is not installed.

traversaro, 2023-07-28:
As a solution, can you try to add  

        ARCHIVE DESTINATION "${CMAKE_INSTALL_LIBDIR}")

 in https://github.com/robotology/gazebo-yarp-plugins/blob/master/cmake/AddGazeboYarpPluginTarget.cmake#L80 ?

traversaro, 2023-07-31:
@FabioBergonti @HosameldinMohamed can you try the fix in https://github.com/robotology/gazebo-yarp-plugins/pull/657 ?

closed 2023-07-31 by traversaro

traversaro, 2023-07-31:
Released in 4.8.0 that will be available in a few minutes also on conda-forge (see https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/pull/10).