robotology/gazebo-yarp-plugins #661: Add Hand mk5.1 coupling handler

Nicogene, 2023-09-07:
In agreement w/ @xEnVrE we decided to refactor the Mk5 handler in order to read the coupling quantities from the configuration file. This will allow to have just one class that handle mk5.0, mk5.1 and mk5.x.

The hand mk5.1 is currently mounted on ergoCub 1.1.

The coupling laws can be browsed [here](https://icub-tech-iit.github.io/documentation/hands/hands_mk5_coupling/#coupling-laws)

cc @fiorisi @pattacini

xEnVrE, 2023-09-07:
The coupling law has been recently [changed](https://github.com/icub-tech-iit/documentation/pull/299/files#diff-6fd610a5590e997d59f7049f795b6b691e9da52a5018be14a8272684a2d8ac60R58) in the [documentation](https://icub-tech-iit.github.io/documentation/hands/hands_mk5_coupling/#coupling-laws), in that the first coupling formula now ends as

$-q_{2bias} - \pi$ 

instead of

$q_{2bias} - \pi$

(because it was more coherent with the drawings, and the tables were updated accordingly).

Should we take the opportunity to also update the code of the coupling law in

https://github.com/robotology/gazebo-yarp-plugins/blob/0fd9300b583267c7bff4fb2be7fe82700ddeb613/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L1302

as part of the resolution to this issue?

Thanks

cc @Nicogene @pattacini

Nicogene, 2023-09-11:
This PR adds the new configuration files needed by the coupling handler:

- https://github.com/icub-tech-iit/ergocub-software/pull/169

@xEnVrE is it what you mean? -> [`322b960` (#662)](https://github.com/robotology/gazebo-yarp-plugins/pull/662/commits/322b96002964a9912a709b53bbb6e3d42375e3f4)

xEnVrE, 2023-09-11:
@Nicogene yes it is! We should be careful not to merge PR #662 before the updated configuration files are provided within `ergocub-software`, btw.

Nicogene, 2023-09-13:
This activity can be considered completed.

@xEnVrE you can start using the mk5.1 hands. you can reopen it if you find any problem

xEnVrE, 2023-09-13:
We will let you know in case of issues @Nicogene! Thanks!

commit c494009b (2023-09-11): HandMk5CouplingHandler: refactor it in order to make it configurable

It fixes #661
commit 794a6722 (2023-09-11): HandMk5CouplingHandler: refactor it in order to make it configurable

It fixes #661
commit 196eef92 (2023-09-11): HandMk5CouplingHandler: refactor it in order to make it configurable

It fixes #661
PR #662 (2023-09-13): HandMk5CouplingHandler: refactor it in order to make it configurable
It fixes #661

This is a first draft, I thoght that the configuration file will become:
```ini
[COUPLING]
icub_hand_mk5 (0 1 2 3 4 5 6 7 8 9 10 11) (l_thumb_add l_thumb_oc l_index_add l_index_oc l_middle_oc l_ring_pinky_oc reserved reserved reserved reserved reserved reserved) (0.0 0.0 0.0 0.0 0.0 0.0) (90.0 82.1 15.0 90.0 90.0 90.0)

[COUPLING_PARAMS]
L0x     -0.00555 -0.00555 -0.00555 -0.00555 -0.00555
L0y      0.00285  0.00285  0.00285  0.00285  0.00285
q2bias     180.0    180.0    180.0    180.0    180.0
q1off       4.29  [... cut]
closed 2023-09-15 by Nicogene