robotology/gazebo-yarp-plugins #682: Unable to stream the base pose via gazebo

GiulioRomualdi, 2024-05-29:
I'm trying to add the base state plugin (the deprecated one with the analog sensor) by following this 

https://github.com/robotology/gazebo-yarp-plugins/blob/29d474fbc674ef0f402cc9b42056bab88f99eb88/plugins/basestate/include/yarp/dev/BaseStateDriver.h#L76-L113.

However, this is no longer documented. What is the correct way to stream the base state into a port? Which device should we use? I think a README is required in the folder of the device. If you provide me with the information, I can write it.

Additionally, the following XML configuration:
Moreover 
```xml
  *          <yarpConfigurationString>(disableImplicitNetworkWrapper) (yarpDeviceName basestate_device) (baseLink torso_link)</yarpConfigurationFile> 
```
is incorrect and should be:

```xml
  *          <yarpConfigurationString>(disableImplicitNetworkWrapper) (yarpDeviceName basestate_device) (baseLink torso_link)</yarpConfigurationString> 
```

cc @traversaro @xela-95

GiulioRomualdi, 2024-05-29:
For instance, something it is not clear to me is where to add the config file:

In the documentation, there is written (it seems it should be placed in `model://`)
```
 *          <yarpConfigurationFile>model://path-to-the-configuration-file</yarpConfigurationFile>
```

can I add it to the path of the world file? I would like to have the config file in a `config` folder next to `world` file
```
.
└── rl_ergoCubGazeboV1_1
    ├── config
    │   └── base_estimator.ini
    └── world
```

traversaro, 2024-05-29:
> is incorrect and should be:

Fix suggested in https://github.com/robotology/gazebo-yarp-plugins/pull/683 .

> However, this is no longer documented.

What is no longer documented?

If you are ok putting the wrapper in its own robotinterface and avoid using the implicit wrapper, you can see how to use the plugin in https://github.com/robotology/gazebo-yarp-plugins/tree/29d474fbc674ef0f402cc9b42056bab88f99eb88/tutorial/model/basestate .

traversaro, 2024-05-29:
> For instance, something it is not clear to me is where to add the config file:
>
> In the documentation, there is written (it seems it should be placed in `model://`)
>
> ```
>  * [...]

`yarpConfigurationFile` arguments are parsed via the `gazebo::common::SystemPaths::Instance()->FindFileURI()` (see https://github.com/robotology/gazebo-yarp-plugins/blob/29d474fbc674ef0f402cc9b42056bab88f99eb88/libraries/singleton/src/ConfHelpers.cc#L70). In a nutshell, they will expand the `model://path/to/a/file.txt` looking if for each `model_path` the paths in `GAZEBO_MODEL_PATH` there exist the file `<model_path>/path/to/a/file.txt` (you can find the logic in https://github.com/gazebosim/gazebo-classic/blob/gazebo11_11.14.0/gazebo/common/SystemPaths.cc#L315-L339).

So, if you want to find it, you need to put some of its parents path in `GAZEBO_MODEL_PATH`. However, that may be inconvenient. There is any reason why you do not want to use `yarpConfigurationString` instead?

GiulioRomualdi, 2024-05-29:
> However, that may be inconvenient. There is any reason why you do not want to use yarpConfigurationString instead?

You are right. This is how I added the device to the gazebo world

```xml
      <plugin name="basestate" filename="libgazebo_yarp_basestate.so">
        <yarpConfigurationString>(WRAPPER (name /ergocubSim/floating_base/state:o) (period 10) (device analogServer)) (DRIVER (device gazebo_basestate) (baseLink root_link)) </yarpConfigurationString>
      </plugin>
```

GiulioRomualdi, 2024-06-25:
Closing with this https://github.com/robotology/gazebo-yarp-plugins/issues/682#issuecomment-2138295910 solution

closed 2024-06-25 by GiulioRomualdi